libretro/Lakka #318: [request] Set core to default

lir448, 2016-03-14:
For example, Lakka has Nestopia core by default for the NES games. I can install other core to emulate NES games, but Nestopia selected when the games start from a playlist anyway. I collided with incorrect emulation games on Nestopia core and want to change core by default. If this option is already exists, I did not find this.

Kivutar, 2016-03-14:
Yeah, it's a bug : the playlist will keep memory of the latest core used to launch this game.

What you can do to workaround this problem : 

 1. Get another core
 2. Set this as default core for NES, in the Playlist Settings
 3. Purge your lpl file in the Playlist folder
 4 Rescan your NES roms

BTW, Nestopia is the fastest and most accurate nes core we have, when you provide the BIOS. If not, it is slow.

lir448, 2016-03-20:
I use a collection of no-intro roms without mods and others, and Nestopia corrupts image in some games and plays it faster than it should be sometimes. I've updated Lakka to version 20160319233020-r21378 and found a new settings for playlist. Now games are started with selected core from the playlist without rescanning than those games that have already been run with another core.

lir448, 2016-03-20:
Playlist consists of six lines on each game:

- Path to the file of the game
- name of the game
- DETECT
- DETECT
- CRC sum
- The name of the playlist

First the DETECT - path to the emulator library, and the second - the name of the emulator. For now,  if I chose the core in the playlist settings and launched the game from the playlist, DETECT will be replaced by this core. If I chose a different core in the playlist settings and launched the same game, the game starts with an old core due to the "playlist.lpl" file. I think that it is possible to remove the lines with core from the "playlist.lpl" files and open the game from this with the core of which is listed in the playlist settings only.

Kivutar, 2016-03-21:
Yes, I've been looking into fixing this when the playlist core assignations got implemented the first time. I can remember that I went in some issues. It requires code refactoring. Because this playlist code is also used for the history list.

Arnochocobo, 2016-12-13:
What you recommended doesn't really work. The first loaded rom will be using the core selected in the first step, but the second rom you load will revert to Nestopia.

Kivutar, 2017-01-15:
Yes, because the default core for the playlist is still set. We would have to unset it so RetroArch would let you choose the core to assign to the entry.

Anyway, this bug is a duplicate of https://github.com/libretro/Lakka/issues/273 so I'm closing it.

I have a clear idea on how to fix it, I just lack free time.

For now editing your playlist with a text editor will do the trick.

closed 2017-01-15 by Kivutar