libretro/Lakka #319: [DISCUSSION] Quit retroarch feature info... and more :D

lollo78, 2016-03-16:
@Kivutar:
Hi dear, please don't hate me! 
I'm "working" (big word: it's more correct to say "I scraped all your hard work" :smile:) on a "Lakka Addon" for Rpi2/3 Openelec (6.0.3). This born from a "wife necessity" that requires a daily system to watch movie, TV and listen music. So, my new Pi3 is now fully dedicated on Multimedia content with OpenenElec. :(
Of course, my little son (and I) loves Lakka and play retrogames, but I hate the dual boot solutions on a little SD Card (too lost space) or swap SD with different distros. So I decided to package a Kodi addon to quickly launch Retroarch from OpeneElec when wife/mum is busy :smile: 
I write a simple script (I'm not a coder) with the help of other similar scripts found online and I did a preconfigured retroarch.cfg to use retroarch with XMB. Seems that all works fine (apart for the confirmed retroarch bug :D).
Of course, this is only for personal use. But, if you want, I'll send it to you for your considerations, advices and improvements. And... always if you want... we could use it to create an offical Lakka Addon for Kodi with a regular repository. If you approve it, also other people could use the Lakka heart direclty inside Openelec. If you don't want, I'll continue only for personal use. :kissing: 

About the info request of the topic...
If I select "Quit Retroarch", my Pi3 restart Kodi (not all the system) but after some minutes (sometimes second) the system totally freezes. This freeze seems not happen if I choose "Reboot".
What is the exactly routine of "Quit Retroarch"?
The code that I use in the script to manage app start and app quit is:
```
# Stop KODI
systemctl --no-block stop kodi
killall kodi.bin

# Enable ALSA
modprobe snd_bcm2835

# Launch emulator
$ADDON_DIR/bin/retroarch --menu -c $RETROARCH_CONFIG_FILE

# Disable ALSA
rmmod snd_bcm2835

# Start KODI
systemctl start kodi
```

duduke, 2016-03-16:
I'm using NOOBS to reboot between OpenELEC and LAKKA.
1. Have an addon within KODI to reboot into LAKKA (modify autoboot on the NOOBS partition).
2. Have a LAKKA autostart script to change the autoboot parameter back to the OpenELEC partition.
3. Selecting reboot in LAKKA gets you back to OpenELEC.
4. Win!

lollo78, 2016-03-16:
Thank for your advice duduke.
Your solution is the best official way, but sincerely I don't like dual boot configuration due to the space lost on the microSD card and the "long" time to switch from a system to another. 
The great Lakka OS is based on OpenElec, this help to extract a Kodi addons without add other libraries. I checked: all the dependencies are satified (ldd) and the retroarch startup is very quick, so I decide to try this way.

I know that all the great work of kivutar is not only on the retroarch side of the Lakka project. A big work is done to optimize OpenElec enviroinment for retrogaming. 
(@Kivutar For this reason I hope you do not get angry with me :D)
But the binary and libretro core compiled with Lakka toolchain are more lighter than other solution (retropie, recalbox...) because he focalizes the attention only on the drivers, plugins etc really needed.

duduke, 2016-03-16:
As I use it on a regular basis, rebooting takes about 10 seconds (not far from just restarting KODI).
No wasted space, As you've mentioned OpenELEC and LAKKA are close relatives :), I'm using the same "data" partition for both.

In any case, I would think you'd find it easier to add LAKKA to base OpenELEC than KODI to base LAKKA.

lollo78, 2016-03-16:
Now I'm curios to try also your solution, using the same data partition for Lakka nad OpenElec. Please, could you share your scripts and some knowledge to perform this?

Kivutar, 2016-03-17:
Why not trying to compile RetroPlayer for ARM?

lollo78, 2016-03-17:
lrusak (Team-Kodi) compiles and releases RetroPlayer for Rpi2/3.
So, this project is active and done from people more experienced than me.
In my little, I would like to contribute with this simple addon, but I realized that you don't like my idea :D
But, please, could you help me (just for personal use) to understand wath is the issue of the OpenElec freeze that occur if I use the Quit Retroarc command into the XMB gui? really, all works fine inside the addon, this freeze is the only issue.
I have to create only a script to automatize the installation...

Kivutar, 2016-03-17:
Well, it's not that I don't like the idea. It's that it is additionnal work that I have no time to do.

I have no idea why it freezes so far. Have you asked on the OpenELEC forums?

lollo78, 2016-03-18:
Sure, about the script I'll ask to the OpenELEC forum. 
But I need your help to undestand what happens when I press "Quit Retroarch".
I tried to search the code into the Retroarch github... but I haven't found it. I lose myself among all the retroarch files and all the code strings... 
But I think that even if I found it, I would not understand a penis the same :+1: 

Maybe I wasn't clear with my request, sorry (speak in english is a pain for me).
I retry: what are the processes killed with "Quit Retroarch" command?

I ask this because I tried also via SSH to stop and start Kodi with the two commands inside my script:
`systemctl --no-block stop kodi`
and 
`systemctl restart kodi`
and all it's OK (no freeze after restart).

So, maybe there are other process killed by retroarch that require to be restarted in Openelec (or it freeze).
Please, if you tell me this... I'll go to stress the OpenELEC guys :D
:kissing:

Kivutar, 2016-03-18:
From what I know of RetroArch, the quit entry just terminate the process. It doesn't do anything fancy about external processes.

lollo78, 2016-03-18:
If you could check/speak about this also with @twinaphex @Alcaro or @fr500 I'll really appreciate it.
In the while, I go to be a pain in the butt for the OE team :smile:

Alcaro, 2016-03-18:
What do I have to do with this? I don't know anything about systemd, how RetroArch quits, or how it interacts with outside processes.

lollo78, 2016-03-18:
Ops sorry: wrong man
:smile
:kissing:

lollo78, 2016-03-22:
@kivutar: on which OpeneElec version is based Lakka 20160319231640-r21378?
6.0.3 or 6.95.1?

Kivutar, 2016-03-22:
A mix of both. We're a kind of fork now.

lollo78, 2016-03-22:
can I ask you why this mix?
Why not base all on the latest Openelec version?

About my Lakka addon... can I send it to you for your consideration?
Could you give me an email? Mine is always 
lollo78 -at - gmail . com
Thx

Kivutar, 2016-03-22:
Sure, OpenELEC droped 32bit and BIOS support in v6. We didn't want to lose those, so I forked. They also droped the live mode or at least, the auto resizing in live mode.

lollo78, 2016-03-22:
Thx. The library dipendecies used from retroarch and libretro cores are based on 6.0.3 or 6.95.1?

Kivutar, 2016-03-22:
A mix of both. For example, retroarch is linked against an up-to-date alsalib, but an out of date glibc. I can't remember all. OpenELEC uses gcc 5+, we're still using gcc 4.9.

lollo78, 2016-03-22:
what is the improvement to use gcc 4.9 instead 5.x?

Kivutar, 2016-03-22:
Some hardkernel kernels refuse to build with gcc5

lollo78, 2016-03-22:
I sent you the email ;)

Kivutar, 2016-05-26:
Can I close this?

lollo78, 2016-05-26:
of course

closed 2016-05-26 by lollo78