libretro/Lakka #334: [Feature Request] Default support for NESPAD by Xgaming?

Hedda, 2016-03-22:
Please consider adding default support for the "NESPAD", which is a new cross-platform gamepad in extended SNES/NES/N64 retro-style combo by Xgaming of X-Arcade fame.

http://shop.xgaming.com/products/nes-wireless-gamepad-ios-android-mac

I know that Shawn Walters <shawn@xgaming.com> from Xgaming donated an Tankstick (XArcade) to developers before and bet they donate NESPADs too if asked ;)

gouchi, 2016-03-22:
If this gamepad comes from [8bitdo](http://www.8bitdo.com/nes30pro), there are already [autoconfig profiles](https://github.com/libretro/retroarch-joypad-autoconfig/tree/master/udev).

Kivutar, 2016-03-22:
@Hedda it's not working? Are you using the latest firmware? You're using wired or bluetooth?

Redemp, 2016-03-30:
This is a rebranding of the 8bitdo Nes30 pro.

I wonder why they did that?

Kivutar, 2016-04-05:
No reply from @Hedda so I close this for now.

closed 2016-04-05 by Kivutar