libretro/Lakka #335: beetle-psx-libretro missing in last Lakka nightly

lollo78, 2016-03-22:
Hi, considering that psx_rearmed will not be supported anymore, could you please add by default the Mednafen beetle-psx-libretro core? Thx

Kivutar, 2016-03-22:
It's too slow on weak hardware. Who told you that pcsx_rearmed will not be supported anymore?

lollo78, 2016-03-22:
twinaphex here:
https://github.com/libretro/Lakka/issues/305
> About pcsx rearmed: I don't think we are going to be continuing development on that. There is a large amount of ASM code in there that really cannot be maintained properly and it would just result in continuous laborous maintenance, far better to focus our efforts on Rustation and Mednafen PSX. And for now, ARM isn't a priority there. So you will just have to live with the current state of pcsx rearmed, there is no active development going on it and it makes for a pretty bad foundation for future PSX development as well.

Kivutar, 2016-03-22:
Yes, that's what he is saying for more than one year. He did a lot of improvements on Beetle PSX, but it's still not fast enough compared to pcsx_rearmed. We still need to wait.

lollo78, 2016-03-22:
:(

Kivutar, 2016-03-22:
Don't worry, it's not like pcsx_rearmed was going to disapear :)

lollo78, 2016-03-22:
In the meantime I opened an issue here: 
https://github.com/libretro/pcsx_rearmed/issues/31
:smile:

Kivutar, 2016-03-22:
I close it for now since beetle-psx can't run yet on those arm boards.

closed 2016-03-22 by Kivutar