libretro/Lakka #336: [request] Make Y and X into turbo B and A for NES (libretro-fceumm)

lir448, 2016-03-22:
This is implemented in the fceumm core and works well in Retroarch for Windows but not working in Lakka. I found this topic with a description: https://github.com/libretro/libretro-fceumm/pull/30

Kivutar, 2016-03-22:
You're aware that we don't ship fceumm in Lakka except on the RPi1 image?

lir448, 2016-03-22:
Nestopia core was in the system only after the installation Lakka. I found that support of turbo button has been added to fceumm and installed it via Core Updater. I have not found support turbo buttons in fceumm for Lakka but found that it works better than Nestopia and left it as a main. But I found that the turbo buttons works in Retroarch for Windows. So I'm writing here to support this in Lakka.

Kivutar, 2016-03-22:
I don't know yet about the turbo bug. But you should get good results with nestopia IF you use the bios.

lir448, 2016-03-23:
This is not a bug. This is a handy feature that was in some gamepads on some NES consoles. It's just helped to not to break buttons A and B when they needed to press many times in a short time. For example, this is described here [Wiki - NES Advantage - Features](https://en.wikipedia.org/wiki/NES_Advantage) and this is shown on the [picture](http://www.8bitnes.com/wp-content/uploads/2013/02/-29221110075649260.jpg). Also I installed all the bios for all the systems and I do not know why my device with allwinner a10 works better with fceumm than with nestopia.

Fantoccini, 2016-04-02:
:+1: for this feature, which reminds me [Takahashi became famous for his fast trigger finger speed of 16 shots per second during the 1980s](https://en.wikipedia.org/wiki/Takahashi_Meijin)

Kivutar, 2016-04-05:
I downloaded the FCEUMM core from the buildbot and this turbo feature is working.

Let try this: delete the NES LPL file in your playlists, and rescan. Set FCEUMM as default core in Settings->Playlist.

I think that the problem is some playlist entries stayed associated with Nestopia.

lir448, 2016-04-18:
I installed Lakka again to another memory card and added the library from here: Online Updater -> Core Updater -> NES / Famicom (FCEUmm). After that I chose Load Content -> Select File And Detect Core -> Contra (USA).nes -> NES / Famicom (FCEUmm). But X and Y did not become Turrbo-B and Turbo-A buttons. Then I opened Quick Menu -> Core Input Options in Lakka and in Retroarch for Windows, and there is no turbo buttons in FCEUmm for Lakka =(
Screenshots from Lakka: http://i65.tinypic.com/mj7mgz.jpg and from Retroarch for Windows: http://i64.tinypic.com/29xg5qe.jpg

hizzlekizzle, 2016-04-18:
Looks like the fceumm build for armhf is from October of last year and the turbo stuff was added in November.

Kivutar, 2016-05-27:
Remember that we don't support the cores from the buildbot.

lir448, 2016-07-13:
I did not check the console for a long time. Yesterday, I formatted the card and installed the latest build (Lakka-a10.arm-devel-20160711235956-r21520-g056bd24). Lakka refused to be updated online from the menu, so I added core manually [from here](http://buildbot.libretro.com/nightly/linux/armhf/latest/). Turbo buttons work perfectly in NES now =) (but I did not understand why the core are not added online even on the version of 01-Jul-2016, others did not try to). If the information is not necessary, the ticket can be closed, thanks.

Kivutar, 2016-07-13:
Ok thanks!

closed 2016-07-13 by Kivutar