libretro/Lakka #337: Rasp Pi 2 - Box Art issue not displaying.

Strider78, 2016-03-23:
Hello Everyone,

First of all, I would like to give thanks to everybody in this community for all the hard and all the time spent in this nice project.

I'm a new user to Lakka and I recently bought a RPi2. 

I'm having certain issues on RetroArch (1.3.2) on RPi2 that seems not to be a problem on RetroArch(1.2.2) for Mac for instance the BoxArts not displaying properly on RPi2.

I took some screenshots to better try to explain the issue. 

On my Macbook pro I have RetroArch 1.2.2 installed, and everything works as it should. It displays the boxarts and play most of the games.

These 4 pictures bellow illustrates the RetroArch 1.2.2 on the Mac (BoxArts Displayed correctly)

[image]

[image]

[image]

[image]

The Issue - The pictures illustraded bellow its for the RPi2 BoxArt issue.

I have the boxart files copied through Samba into RPi2 BoxArt Directory.

[image]

The boxart files are name properly according to the rom scanning.

[image]

I have boxart directory set at:
[image]

I have the display boxart on (enabled)

[image]

But unfortunately nothing shows on RPi2

[image]
[image]

I also tried restarting the RPi2 to see if the changes would take place. Unfortunately nothing happened.

Can anyone help ? Anyone having any similar issues like mine?  I'm not too technical or comfortable yet on RPi2 but I will read and listen to you all.

Once again thank you!

Kivutar, 2016-03-23:
Hello, we changed something in that feature. Read there: http://www.lakka.tv/articles/2015/11/26/new-major-version-released/

Anyway, it's going to change again soon, since we are going to provide 3 types of images in packs.

Strider78, 2016-03-23:
I will wait for a new future release and reinstall from ground up.

Thank you so much for replying so soon.

Kivutar, 2016-03-23:
No problem. I close this for now.

closed 2016-03-23 by Kivutar