libretro/Lakka #338: LibreELEC

RobLoach, 2016-03-24:
Is Lakka forked from OpenELEC? If so, it may be good to note that OpenELEC has since been forked to become [LibreELEC](https://libreelec.tv/about/). Whether or not there is consideration to switch to LIbreELEC, it may to good to take a look if there's anything Lakka could benefit from it.

Kivutar, 2016-03-24:
Thanks! I take a look at that.

Kivutar, 2016-03-24:
Got in touch with them. They are frendly. It's basically all the team of OE minus sraue. There were even zalaare.

They don't plan to add back 32bit or BIOS support. They don't plan either to merge new hardware ports.

However they want more democracy in the development process. And more stability in the release mainaintance.

So, for now, we just need to keep an eye on their commit log.

chewitt, 2016-03-26:
@Kivutar - We absolutely do want to merge new hardware ports into the LibreELEC git repo, but current priority is getting official release builds started on existing (and well-known to the team) hardware. Once there's an operating rhythm going we'll start looking at new (to us) projects and start the discussions on how to do things - and we'll be wanting Lakka input to ensure we're good neighbours. As you said, we're wanting a more stable and predictable release so we'll do things in baby-steps, that's all :)

Kivutar, 2016-03-26:
Ah good to know. Let me know when you reach that point, and I will join the discussions on what port we should include, and send some PR.

Kivutar, 2016-05-14:
Porting Lakka to LibreELEC is not difficult. Just a matter of copying a few folders like

 * distributions/Lakka
 * packages/libretro
 * packages/lakka (mostly oem packages that could be moved to respective projects)
 * projects/*
 * our own smb.conf
 * patches to plymouth-lite
 * overlayfs patches for the old kernels
 * bluez patches
 * small changes to the kernel and uboot package
 * small changes to scripts/mkimage

The difficulties I see are:

 * compiling hardkernel kernels with gcc5
 * restauring i386 support
 * restauring BIOS support

Porting to LibreELEC would give us:

 * OdroidC2 support
 * Virtual image
 * The ability to upgrade from the img instead of the tar

chewitt, 2016-05-14:
I have basic i386 support in https://github.com/chewitt/LibreELEC.tv/tree/appletv-7.0 but for ATV and not 100% working yet so a few things will be missing for a Generic build. Probably not too much though. If I manage to spend less time on planes and more time at home in the second half of this month ATV will be finished off.

BIOS support is what? .. the .img we ship _should_ work for EFI and non-EFI systems. If it doesn't we'd like to understand the issues and work together to ensure it works for both cases. That's the biggest thing LE can offer; the ability to actively work _with_ the group of people creating it instead of just being on the receiving end of someone else's inconsiderate decisions. I am biased though :)

Kivutar, 2016-05-14:
OE had a booloader that was able to boot on BIOS and emulated BIOS. They dropped it around OEv6. At that time, they also dropped the auto resize of the second partition in the Live USB mode. OE also lost the ability to boot on my macbook. I'm going to try LE now in Live USB and see if things are different. Thanks for the information.

Kivutar, 2016-05-14:
So, I booted LE on my macbook. The USB key is recognized as an EFI boot. I don't get the syslinux menu allowing me to choose the live mode, it boots directly to the installer mode.

I was not able to modify the syslinux.cfg to force the live mode, the first partition is read only and I don't get why.

Anyway, I inspected the second partition, and didn't see the .please_resize_me, so I guess it's like OE, the Live mode is kind of *deprecated*?

Kivutar, 2016-05-14:
On my Intel NUC, I can select the two boot modes. I can hit tab to choose the live mode. But whatever mode I choose, I always end up with just a black screen.

chewitt, 2016-05-14:
Live mode was either removed or broken around OE 5.x for reasons I still don't understand, but at the time the folk who ruled github were taking a minimalist approach and frequently removed features they didn't personally use "to make the distro simpler" even if a pile of other people thought it was essential and a good idea. The whole LE install process from creating a USB to what it does when booted needs to be redesigned; both from a user-experience perspective and to consider the needs of some of our downstream community builds and derivative projects. Improving USB creation is probably the higher priority but there are multiple people (self included) who think booting from one USB to install to another USB is dumb so Live mode needs to be looked at. The #installer channel in Slack has gone slow.. but needs some encouragement to pick up the conversation again.

Kivutar, 2016-05-14:
I remember sraue telling me that resizing  the second partition in live mode was incompatible with UEFI and that's why they removed it. I don't see why it would be incompatible. But I know less than him on the subject so there is certainly a technical reason. I'd really like to know.

A quick test would be adding manually the .please_resize_me and see what happens.

All these things are still working on Lakka, because I forked around v5 and backported most of the evolutions except the bootloader stuff, to preserve those features.

Kivutar, 2016-10-17:
We've started the port. I close this.

closed 2016-10-17 by Kivutar