libretro/Lakka #339: Xbox 360 controller don't pair with official Dongle on PI3

keulu, 2016-03-24:
can't pair controllers to the dongle.

PS3 controller with cable work's fine, but xbox not

Lakka:/ # lsusb
Bus 001 Device 008: ID 054c:0268 Sony Corp. Batoh Device / PlayStation 3 Controller
Bus 001 Device 005: ID 045e:0719 Microsoft Corp. Xbox 360 Wireless Adapter
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

so my Xbox_360_Wireless_Receiver.cfg is good
input_vendor_id = 1118
input_product_id = 1817

Lakka recognize it and assigne port 1, 2, 3 and 4 (my ps3 controller is on 0)

but xbox controller can't pair.

Kivutar, 2016-03-26:
Ok, I just forgot some patches on the RPi builds.

keulu, 2016-03-26:
ha ha ^^ ok :)

And my ps3 controller don't work with pi3 integrated bluetooth. did you forgot a patch too or it's just not implemented for the moment?

Kivutar, 2016-03-26:
We still need to implement this. It should be done on two weeks

Kivutar, 2016-04-04:
It should be fixed now by the new update.

keulu, 2016-04-04:
thx guy. can i use the lakka updater ? or make a new fresh install ?

Kivutar, 2016-04-04:
Oh, sorry, I made a mistake. I fixed the bluetooth on RPi3, but completely forgot that the xpad is not using bluetooth anyway.

So please forget what I said. I will try to check the kernel patches.

Kivutar, 2016-04-04:
Ok, can you update by droping this http://static.kivutar.me/Lakka-RPi2.arm-devel-20160404140036-r21394-g25646a6.tar in your .update folder, and reboot please?

keulu, 2016-04-04:
sorry dude, same problem for both :  xbox dongle and bluetooth ps3 controller

I can give you some log if you tell me the command.

Kivutar, 2016-05-05:
Hello @keulu 
Can you give a try to the last release please?

OGWillikers, 2016-05-30:
Can you pair the controller to the dongle with a PC, then use it on the RPi?  Mine work just fine like that minus not showing what player they're connected as.  Haven't tried pairing connected to the RPi as I don't currently have a 360 to break the pairing.

Kivutar, 2016-10-04:
No news, I close this for now.

closed 2016-10-04 by Kivutar