libretro/Lakka #340: Installation failure on VIA EPIA 10000 Board

outcastrift, 2016-03-28:
Hello I've been trying now for two days pretty much continuously to get Lakka installed on a VIA board of mine.  The board is a i386 Pico-form factor motherboard. Due to its architecture I have been trying to install the latest Generic i386 build listed on your website. 

I will go ahead and outline the problem and what I have attempted to do to fix it to give you a idea where I stand. 

MAIN ISSUE : While the Syslinux prompt to select "Run Installer" and "Run Live" render just fine after selecting either option the device progresses to a black screen shortly followed by the monitor saying there is no longer any input.  

I cant remember if I started doing anything else at the moment.  Honestly been a long weekend trying to get this to run for a Arcade cabinet I am setting up .  I really dont want to have to recompile this and trying to tinker with the internals to get it to work.  So i came here to well you know Talk to the source  :p 

Thank you to anyone that helps out in advance.  God knows I hit a brick wall on it. 
NOTE: If i make a live drive on another PC and then attempt to boot it within the Via system the exact same things happens EXCEPT I retain the ability to SSH into the box. 

STEPS I HAVE TAKEN TO RESOLVE: 
1. Edit kernel boot line parameters to disable the LVDS connector and output via VGA as primary. 
Result: Failed No change. 
2. Install the Lakka distrbution image on a i386 old laptop board using VGA out and then transfer the drive to the VIA board to see if its a installer issue. 
Result: Failed No Change
3. Hijack syslinux and edit linux boot parameters directly. Specifically booting to tty=3 and setting retroarch=o.
Result: Failed , never saw the prompt. 
4. While in SSH attempt to find all DMS outputs available to the board for more accurate bootline arguments. Command ran was 
for p in /sys/class/drm/*/status; do con=${p%/status}; echo -n "${con#*/card?-}: "; cat $p; done
Result: While /sys/class/drm does exist the underlying folder structure that normally comes after that has been moved , so I gained nothing here. 
5. Edited various BIOS options to disable particular outputs and only leave the specific one i need on. 
Result : Failed No Change 
6. After creating a Live Bootable version on another PC, Editing the retroarch.cfg file to output to monitor 2 .
Result : Failed No Change 
7. Attempted to install a previous version of Lakka that was built with x-system in the kernel instead of DRM. 
Result: Failed No Change : However I may have gotten the wrong Image, I tried versions from Nov-2015 and May-2015 .

outcastrift, 2016-03-28:
Full LSPCI-V listed below for the Via board. 

00:00.0 Host bridge: VIA Technologies, Inc. CX700/VX700 Host Bridge (rev 03)
        Subsystem: VIA Technologies, Inc. CX700/VX700 Host Bridge
        Flags: bus master, medium devsel, latency 8
        Memory at d0000000 (32-bit, prefetchable) [size=128M]
        Capabilities: [80] AGP version 3.5
        Capabilities: [50] Power Management version 2
        Kernel driver in use: agpgart-via

00:00.1 Host bridge: VIA Technologies, Inc. CX700/VX700 Host Bridge
        Flags: bus master, medium devsel, latency 0

00:00.2 Host bridge: VIA Technologies, Inc. CX700/VX700 Host Bridge
        Flags: bus master, medium devsel, latency 0

00:00.3 Host bridge: VIA Technologies, Inc. CX700/VX700 Host Bridge
        Flags: bus master, medium devsel, latency 0

00:00.4 Host bridge: VIA Technologies, Inc. CX700/VX700 Host Bridge
        Flags: bus master, medium devsel, latency 0

00:00.7 Host bridge: VIA Technologies, Inc. CX700/VX700 Host Bridge
        Flags: bus master, medium devsel, latency 0

00:01.0 PCI bridge: VIA Technologies, Inc. VT8237/VX700 PCI Bridge (prog-if 00 [Normal decode])
        Flags: bus master, 66MHz, medium devsel, latency 0
        Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
        I/O behind bridge: 0000e000-0000efff
        Memory behind bridge: dd000000-deffffff
        Prefetchable memory behind bridge: a0000000-bfffffff
        Capabilities: [70] Power Management version 2

00:0f.0 IDE interface: VIA Technologies, Inc. VX800 Serial ATA and EIDE Controller (prog-if 8a [Master SecP PriP])
        Subsystem: VIA Technologies, Inc. VX800 Serial ATA and EIDE Controller
        Flags: bus master, medium devsel, latency 32
        [virtual] Memory at 000001f0 (32-bit, non-prefetchable) [size=8]
        [virtual] Memory at 000003f0 (type 3, non-prefetchable)
        [virtual] Memory at 00000170 (32-bit, non-prefetchable) [size=8]
        [virtual] Memory at 00000370 (type 3, non-prefetchable)
        I/O ports at ff00 [size=16]
        Capabilities: [b0] Power Management version 2
        Kernel driver in use: pata_via

00:10.0 USB controller: VIA Technologies, Inc. VT82xx/62xx UHCI USB 1.1 Controller (rev 90) (prog-if 00 [UHCI])
        Subsystem: VIA Technologies, Inc. VT82xx/62xx UHCI USB 1.1 Controller
        Flags: bus master, medium devsel, latency 32, IRQ 20
        I/O ports at fe00 [size=32]
        Capabilities: [80] Power Management version 2
        Kernel driver in use: uhci_hcd

00:10.1 USB controller: VIA Technologies, Inc. VT82xx/62xx UHCI USB 1.1 Controller (rev 90) (prog-if 00 [UHCI])
        Subsystem: VIA Technologies, Inc. VT82xx/62xx UHCI USB 1.1 Controller
        Flags: bus master, medium devsel, latency 32, IRQ 22
        I/O ports at fd00 [size=32]
        Capabilities: [80] Power Management version 2
        Kernel driver in use: uhci_hcd

00:10.2 USB controller: VIA Technologies, Inc. VT82xx/62xx UHCI USB 1.1 Controller (rev 90) (prog-if 00 [UHCI])
        Subsystem: VIA Technologies, Inc. VT82xx/62xx UHCI USB 1.1 Controller
        Flags: bus master, medium devsel, latency 32, IRQ 21
        I/O ports at fc00 [size=32]
        Capabilities: [80] Power Management version 2
        Kernel driver in use: uhci_hcd

00:10.4 USB controller: VIA Technologies, Inc. USB 2.0 (rev 90) (prog-if 20 [EHCI])
        Subsystem: VIA Technologies, Inc. USB 2.0
        Flags: bus master, medium devsel, latency 32, IRQ 23
        Memory at dffff000 (32-bit, non-prefetchable) [size=256]
        Capabilities: [80] Power Management version 2
[... 25 lines cut ...]
        Flags: bus master, 66MHz, medium devsel, latency 32, IRQ 10
        Memory at a0000000 (32-bit, prefetchable) [size=512M]
        Memory at dd000000 (32-bit, non-prefetchable) [size=16M]
        [virtual] Expansion ROM at de000000 [disabled] [size=64K]
        Capabilities: [60] Power Management version 2
        Capabilities: [70] AGP version 3.0

02:06.0 Ethernet controller: VIA Technologies, Inc. VT6105/VT6106S [Rhine-III] (rev 8b)
        Subsystem: VIA Technologies, Inc. VT6105/VT6106S [Rhine-III] (VT6106S [Rhine-III])
        Flags: bus master, stepping, medium devsel, latency 32, IRQ 18
        I/O ports at de00 [size=256]
        Memory at dfeff000 (32-bit, non-prefetchable) [size=256]
        Capabilities: [44] Power Management version 2
        Kernel driver in use: via-rhine

80:01.0 Audio device: VIA Technologies, Inc. VT8237A/VT8251 HDA Controller (rev 10)
        Subsystem: VIA Technologies, Inc. VT8237A/VT8251 HDA Controller
        Flags: bus master, fast devsel, latency 0, IRQ 24
        Memory at 9fffc000 (64-bit, non-prefetchable) [size=16K]
        Capabilities: [50] Power Management version 2
        Capabilities: [60] MSI: Enable+ Count=1/1 Maskable- 64bit+
        Capabilities: [70] Express Root Complex Integrated Endpoint, MSI 00
        Capabilities: [100] Virtual Channel
        Kernel driver in use: snd_hda_intel
        Kernel modules: snd_hda_intel

outcastrift, 2016-03-28:
Trying to install OpenElec on the box results in the same behavior, any other Distro works fine, Ubuntu, Mint, Puppy, etc. ..

outcastrift, 2016-03-28:
Tried to set vmalloc=192M
Result:Failure No Change

outcastrift, 2016-03-28:
Attempting to boot directly to tty and setting loglevel=7 just result in a extremely fast print of the tasks of loading Lakka and then a blinking prompt for roughly 1-2 seconds then the video cut off from the display.

gouchi, 2016-03-28:
We need to add support for VIA cards. The [OpenChrome](https://www.freedesktop.org/wiki/Openchrome) project is not a good shape but recently a new maintainer arrived to maintain and develop the OpenChrome driver.

You card should be `lspci -nn | grep VGA` (1106:3157). It seems both 2D and 3D are [supported](https://www.freedesktop.org/wiki/Openchrome/SupportedHardware).  

You can try to add to boot parameters **textconsole** to get access directly to a shell.

outcastrift, 2016-03-28:
Hey gouchi, thanks for the response.  I will definitely try the textmode parameter when I get back to the house today.  Been using Linux for years always on RHEL or Debian based distributions, never thought I would miss Grub lol. 

Either way can't really tell from your post but your saying the machine is or isn't going to be able to install Lakka ? Not sure if your saying the OpenChrome project is being worked and their might be support for the device in the future or that it should be working now.

gouchi, 2016-03-28:
We need to enable the support of VIA cards in Lakka.

Even if we enabled it, I am not sure it will work, we need to make some test. First, can your try to boot on a  Linux distribution and see how it works ?

outcastrift, 2016-03-28:
It works fine in all Debian based distros, puppy Linux, Kali Linux and even Windows the greedy resource monitor it is. I even installed retroarch under Ubuntu and it worked just a little slow I feel it would do fine without a full OS running behind it

outcastrift, 2016-03-28:
Talking to your phone to reply to a  GitHub issues if there's typos

outcastrift, 2016-03-28:
If you want to try and run a one time build with the via support added I can test for you if you give me series of regression tests you guys usually run .

gouchi, 2016-03-29:
So can you paste [somewhere](http://www.hastebin.com) the result of these commands with one of the distro you tried.

`lspci -knn | grep -A2 VGA` and `glxinfo`

We need to check which drivers you are using.

outcastrift, 2016-03-29:
Dang gotta hate these time shifts, I get to work and realized you sent me a message. Sure thing though I'll get that output to you today .

outcastrift, 2016-03-30:
lspci -knn 

00:00.0 Host bridge [0600]: VIA Technologies, Inc. CX700/VX700 Host Bridge [1106:0324] (rev 03)
        Subsystem: VIA Technologies, Inc. CX700/VX700 Host Bridge [1106:aa09]
        Kernel driver in use: agpgart-via
00:00.1 Host bridge [0600]: VIA Technologies, Inc. CX700/VX700 Host Bridge [1106:1324]
00:00.2 Host bridge [0600]: VIA Technologies, Inc. CX700/VX700 Host Bridge [1106:2324]
00:00.3 Host bridge [0600]: VIA Technologies, Inc. CX700/VX700 Host Bridge [1106:3324]
00:00.4 Host bridge [0600]: VIA Technologies, Inc. CX700/VX700 Host Bridge [1106:4324]
00:00.7 Host bridge [0600]: VIA Technologies, Inc. CX700/VX700 Host Bridge [1106:7324]
00:01.0 PCI bridge [0604]: VIA Technologies, Inc. VT8237/VX700 PCI Bridge [1106:b198]
00:0f.0 IDE interface [0101]: VIA Technologies, Inc. VX800 Serial ATA and EIDE Controller [1106:5324]
        Subsystem: VIA Technologies, Inc. VX800 Serial ATA and EIDE Controller [1106:aa09]
        Kernel driver in use: pata_via
00:10.0 USB controller [0c03]: VIA Technologies, Inc. VT82xx/62xx UHCI USB 1.1 Controller [1106:3038] (rev 90)
        Subsystem: VIA Technologies, Inc. VT82xx/62xx UHCI USB 1.1 Controller [1106:aa09]
        Kernel driver in use: uhci_hcd
00:10.1 USB controller [0c03]: VIA Technologies, Inc. VT82xx/62xx UHCI USB 1.1 Controller [1106:3038] (rev 90)
        Subsystem: VIA Technologies, Inc. VT82xx/62xx UHCI USB 1.1 Controller [1106:aa09]
        Kernel driver in use: uhci_hcd
00:10.2 USB controller [0c03]: VIA Technologies, Inc. VT82xx/62xx UHCI USB 1.1 Controller [1106:3038] (rev 90)
        Subsystem: VIA Technologies, Inc. VT82xx/62xx UHCI USB 1.1 Controller [1106:aa09]
        Kernel driver in use: uhci_hcd
00:10.4 USB controller [0c03]: VIA Technologies, Inc. USB 2.0 [1106:3104] (rev 90)
        Subsystem: VIA Technologies, Inc. USB 2.0 [1106:aa09]
        Kernel driver in use: ehci-pci
00:11.0 ISA bridge [0601]: VIA Technologies, Inc. CX700/VX700 PCI to ISA Bridge [1106:8324]
        Subsystem: VIA Technologies, Inc. CX700/VX700 PCI to ISA Bridge [1106:aa09]
00:11.7 Host bridge [0600]: VIA Technologies, Inc. CX700/VX700 Internal Module Bus [1106:324e]
        Subsystem: VIA Technologies, Inc. CX700/VX700 Internal Module Bus [1106:324e]
00:13.0 Host bridge [0600]: VIA Technologies, Inc. CX700/VX700 Host Bridge [1106:324b]
00:13.1 PCI bridge [0604]: VIA Technologies, Inc. CX700/VX700 PCI to PCI Bridge [1106:324a]
01:00.0 VGA compatible controller [0300]: VIA Technologies, Inc. CX700/VX700 [S3 UniChrome Pro] [1106:3157] (rev 03)
        Subsystem: VIA Technologies, Inc. CX700/VX700 [S3 UniChrome Pro] [1106:3157]
02:06.0 Ethernet controller [0200]: VIA Technologies, Inc. VT6105/VT6106S [Rhine-III] [1106:3106] (rev 8b)
        Subsystem: VIA Technologies, Inc. VT6105/VT6106S [Rhine-III] (VT6106S [Rhine-III]) [1106:0105]
        Kernel driver in use: via-rhine
80:01.0 Audio device [0403]: VIA Technologies, Inc. VT8237A/VT8251 HDA Controller [1106:3288] (rev 10)
        Subsystem: VIA Technologies, Inc. VT8237A/VT8251 HDA Controller [1106:3288]
        Kernel driver in use: snd_hda_intel
        Kernel modules: snd_hda_intel

outcastrift, 2016-03-30:
lspci -knn | grep -A2 VGA 

01:00.0 VGA compatible controller [0300]: VIA Technologies, Inc. CX700/VX700 [S3 UniChrome Pro] [1106:3157] (rev 03)
        Subsystem: VIA Technologies, Inc. CX700/VX700 [S3 UniChrome Pro] [1106:3157]
02:06.0 Ethernet controller [0200]: VIA Technologies, Inc. VT6105/VT6106S [Rhine-III] [1106:3106] (rev 8b)

outcastrift, 2016-03-30:
Glaxinfo from puppy linux 

name of display: :0.0
display: :0  screen: 0
direct rendering: Yes
server glx vendor string: SGI
server glx version string: 1.2
server glx extensions:
    GLX_ARB_multisample, GLX_EXT_import_context, GLX_EXT_texture_from_pixmap,
    GLX_EXT_visual_info, GLX_EXT_visual_rating, GLX_OML_swap_method,
    GLX_SGI_make_current_read, GLX_SGI_swap_control, GLX_SGIS_multisample,
    GLX_SGIX_fbconfig, GLX_SGIX_visual_select_group
client glx vendor string: SGI
client glx version string: 1.4
client glx extensions:
    GLX_ARB_get_proc_address, GLX_ARB_multisample, GLX_EXT_import_context,
    GLX_EXT_visual_info, GLX_EXT_visual_rating, GLX_MESA_allocate_memory,
    GLX_MESA_copy_sub_buffer, GLX_MESA_swap_control,
    GLX_MESA_swap_frame_usage, GLX_OML_swap_method, GLX_OML_sync_control,
    GLX_SGI_make_current_read, GLX_SGI_swap_control, GLX_SGI_video_sync,
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer,
    GLX_SGIX_visual_select_group, GLX_EXT_texture_from_pixmap
GLX version: 1.2
GLX extensions:
    GLX_ARB_get_proc_address, GLX_ARB_multisample, GLX_EXT_import_context,
    GLX_EXT_visual_info, GLX_EXT_visual_rating, GLX_MESA_swap_control,
    GLX_MESA_swap_frame_usage, GLX_OML_swap_method, GLX_SGI_make_current_read,
    GLX_SGI_swap_control, GLX_SGI_video_sync, GLX_SGIS_multisample,
    GLX_SGIX_fbconfig, GLX_SGIX_visual_select_group
OpenGL vendor string: VIA Technology
OpenGL renderer string: Mesa DRI UniChrome 20060710 x86/MMX/SSE2
OpenGL version string: 1.2 Mesa 6.5.2
OpenGL extensions:
    GL_ARB_imaging, GL_ARB_multisample, GL_ARB_multitexture,
    GL_ARB_point_parameters, GL_ARB_texture_env_add,
    GL_ARB_texture_env_combine, GL_ARB_texture_mirrored_repeat,
    GL_ARB_transpose_matrix, GL_ARB_window_pos, GL_EXT_abgr, GL_EXT_bgra,
    GL_EXT_blend_color, GL_EXT_blend_minmax, GL_EXT_blend_subtract,
    GL_EXT_clip_volume_hint, GL_EXT_compiled_vertex_array, GL_EXT_convolution,
    GL_EXT_copy_texture, GL_EXT_draw_range_elements, GL_EXT_fog_coord,
    GL_EXT_histogram, GL_EXT_packed_pixels, GL_EXT_point_parameters,
    GL_EXT_polygon_offset, GL_EXT_rescale_normal, GL_EXT_secondary_color,
    GL_EXT_separate_specular_color, GL_EXT_stencil_wrap, GL_EXT_subtexture,
    GL_EXT_texture, GL_EXT_texture3D, GL_EXT_texture_edge_clamp,
    GL_EXT_texture_env_add, GL_EXT_texture_env_combine,
    GL_EXT_texture_lod_bias, GL_EXT_texture_object, GL_EXT_vertex_array,
    GL_APPLE_packed_pixels, GL_IBM_rasterpos_clip,
    GL_IBM_texture_mirrored_repeat, GL_MESA_window_pos, GL_NV_blend_square,
    GL_NV_light_max_exponent, GL_NV_texgen_reflection, GL_OES_read_format,
    GL_SGI_color_matrix, GL_SGI_color_table, GL_SGIS_texture_edge_clamp,
    GL_SGIS_texture_lod

12 GLX Visuals
   visual  x  bf lv rg d st colorbuffer ax dp st accumbuffer  ms  cav
 id dep cl sp sz l  ci b ro  r  g  b  a bf th cl  r  g  b  a ns b eat
----------------------------------------------------------------------
0x22 16 tc  0 16  0 r  y  .  5  6  5  0  0 24  8  0  0  0  0  0 0 None
0x23 16 tc  0 16  0 r  .  .  5  6  5  0  0 24  8  0  0  0  0  0 0 None
0x24 16 tc  0 16  0 r  y  .  5  6  5  0  0 16  0  0  0  0  0  0 0 None
0x25 16 tc  0 16  0 r  .  .  5  6  5  0  0 16  0  0  0  0  0  0 0 None
0x26 16 tc  0 16  0 r  y  .  5  6  5  0  0  0  0  0  0  0  0  0 0 None
0x27 16 tc  0 16  0 r  .  .  5  6  5  0  0  0  0  0  0  0  0  0 0 None
0x28 16 tc  0 16  0 r  y  .  5  6  5  0  0 24  8 16 16 16  0  0 0 Slow
0x29 16 tc  0 16  0 r  .  .  5  6  5  0  0 24  8 16 16 16  0  0 0 Slow
0x2a 16 tc  0 16  0 r  y  .  5  6  5  0  0 16  0 16 16 16  0  0 0 Slow
0x2b 16 tc  0 16  0 r  .  .  5  6  5  0  0 16  0 16 16 16  0  0 0 Slow
0x2c 16 tc  0 16  0 r  y  .  5  6  5  0  0  0  0 16 16 16  0  0 0 Slow
0x2d 16 tc  0 16  0 r  .  .  5  6  5  0  0  0  0 16 16 16  0  0 0 Slow

12 GLXFBConfigs:
   visual  x  bf lv rg d st colorbuffer ax dp st accumbuffer  ms  cav
 id dep cl sp sz l  ci b ro  r  g  b  a bf th cl  r  g  b  a ns b eat
----------------------------------------------------------------------
0x22  0 tc  0 16  0 r  y  .  5  6  5  0  0 24  8  0  0  0  0  0 0 None
0x23  0 tc  0 16  0 r  .  .  5  6  5  0  0 24  8  0  0  0  0  0 0 None
0x24  0 tc  0 16  0 r  y  .  5  6  5  0  0 16  0  0  0  0  0  0 0 None
0x25  0 tc  0 16  0 r  .  .  5  6  5  0  0 16  0  0  0  0  0  0 0 None
0x26  0 tc  0 16  0 r  y  .  5  6  5  0  0  0  0  0  0  0  0  0 0 None
0x27  0 tc  0 16  0 r  .  .  5  6  5  0  0  0  0  0  0  0  0  0 0 None
0x28  0 tc  0 16  0 r  y  .  5  6  5  0  0 24  8 16 16 16  0  0 0 Slow
0x29  0 tc  0 16  0 r  .  .  5  6  5  0  0 24  8 16 16 16  0  0 0 Slow
0x2a  0 tc  0 16  0 r  y  .  5  6  5  0  0 16  0 16 16 16  0  0 0 Slow
0x2b  0 tc  0 16  0 r  .  .  5  6  5  0  0 16  0 16 16 16  0  0 0 Slow
0x2c  0 tc  0 16  0 r  y  .  5  6  5  0  0  0  0 16 16 16  0  0 0 Slow
0x2d  0 tc  0 16  0 r  .  .  5  6  5  0  0  0  0 16 16 16  0  0 0 Slow

gouchi, 2016-03-30:
So you are using openchrome driver. I need to check with @Kivutar if it is possible to add [openchrome driver](https://cgit.freedesktop.org/openchrome) to Lakka.

outcastrift, 2016-03-30:
That would be great my friend any possible ETA on the decision ?

outcastrift, 2016-04-01:
Any news on this ? I have 5 of these machines and some younger family members I'd like to build something for. This is my only blocker at the moment and I'd much rather have a simple and clean solution. 

Using MAME under Linux is fine I guess or retroarch but I'd rather have Lanka

outcastrift, 2016-04-01:
Lakka*

gouchi, 2016-04-01:
Sorry for late reply but as soon as Kivutar comes back, normally next week. We can decide if we can add it.

In the meantime, can you test RetroArch under your Linux ?

outcastrift, 2016-04-01:
I have been testing under ubuntu, only really have one issue to be honest and its configuring a trackball mouse to use in a mame game .  I need it for Golden Tee Fore one of the games its terrible to play without and I can't seem to figure out how to do it. If I run regular mame emulation it works just fine but under retroarch it just opens the game and doesn't accept that the mouse is even there. Although on the retroarch menu the mouse works fine.

outcastrift, 2016-04-01:
Just a really big pain configuring that, but I look forward to what he says

gouchi, 2016-04-04:
I made a mistake you are using an old mesa with Puppy Linux. And it is [unichrome](http://unichrome.sourceforge.net/) driver not openchrome driver.

I check more closely openchrome driver and they don't support mesa yet so no 3D acceleration.

For now, your solution is to use a linux distribution with old mesa version (until version 7) and install RetroArch on it.

I am sorry for the mistake.

outcastrift, 2016-04-04:
I'm confused it seems that doing some research on this end led me to seeing that the UniChrome does support 3d acceleration in some cases however if it errors then it defaults to software rendering.

outcastrift, 2016-04-04:
From the OpenChrome website i read this. 

Unichrome	CLE266	0x3122	OK	OK	OK	OK (up to 1024x1024)	absent	not implemented	OK
Unichrome	KM400, KN400, KM400A, ?P4M800	0x7205	OK	OK	OK	not implemented (up to 1024x1024)	absent	not implemented	OK
Unichrome Pro B	?K8M800	0x3108	OK	OK (locks up with textures)	OK	OK (up to 1024x1024)	absent	not implemented	OK
Unichrome Pro A	CN400, PM800, PN800, PM880	0x3118	OK	OK	OK	OK (up to 2048x2048)	not implemented	not implemented	OK
Unichrome Pro	CN700, VN800, ?P4M800Pro	0x3344	OK	OK	OK	OK (up to 1024x1024)	absent	not implemented	OK
Unichrome Pro	?P4M890	0x3343	OK	OK	OK	OK (up to 1024x1024)	absent	not implemented	OK
Unichrome Pro II	CX700	0x3157	OK	OK	OK	WIP (up to 2048x2048)	absent	not implemented	OK
Unichrome Pro II	CX700M(2)	0x3157	OK	OK	OK	WIP (up to 2048x2048)	not implemented	not implemented	OK

outcastrift, 2016-04-04:
seems lilke all the Unichromes do support 3D ?

gouchi, 2016-04-04:
The documentation has not been updated it not true anymore.

All [DRI1 drivers](https://cgit.freedesktop.org/mesa/mesa/commit/?id=e4344161bde2e24fcfba65d30d58f087bd8bf94d) has been removed since 2011.

outcastrift, 2016-04-04:
Hard to understand why they would remove support for something without replacing that support with something new. This isn't bloody edge stuff were talking here you know. ? 

So how am I functioning under Ubuntu 14.04 software rendering ?

gouchi, 2016-04-05:
You should launch **glxinfo**  and check the output
```
**OpenGL vendor string**: VIA Technology
**OpenGL renderer string**: Mesa DRI UniChrome 20060710 x86/MMX/SSE2
**OpenGL version string**: 1.2 Mesa 6.5.2
```

outcastrift, 2016-04-12:
Sorry boss been really busy at work.  Enterprise java will quite literally drive you insane.  I'll check it out as soon as I can.

Kivutar, 2016-10-05:
Any news on this?