libretro/Lakka #341: Support SNES controllers on GPIO

DerekRoth, 2016-03-30:
I have 2 SNES controller ports plugged in to the GPIO pins of my RPi2, through the Retropie GPIO adapter. On Retropie, this setup is supported through the [SNESDev driver](https://github.com/petrockblog/SNESDev-RPi). An improved version that is not tied to the Retropie GPIO adapter can be found on @axle-h's [fork](https://github.com/axle-h/SNESDev-RPi).

In its current version, this driver creates 2 joysticks devices at boot (even with no controllers plugged in). So it may be better to make this driver optional, to prevent having 2 useless devices for people who don't use this setup.

digitalLumberjack, 2016-03-30:
Hi, i would recommand to look the gamecon kernel driver too. 
https://github.com/recalbox/recalbox-os/wiki/Gamecon-controllers-%28EN%29
https://www.raspberrypi.org/forums/viewtopic.php?f=78&t=15787

gouchi, 2016-04-01:
About the gamecon kernel driver,  it seems the driver is only available in [deb package](https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/supplementary/gamecondriver.sh#L50) so it is not convenient for the integration to Lakka.

They should provide the source of their modifications in a repository.

duduke, 2016-04-03:
It's true that the gamecon driver is available as a deb package, but it is DKMS which means it actually contains the source code to be recompiled after a kernel change/upgrade.

I'm going to try to add it to LAKKA.

gouchi, 2016-04-04:
You are right so we can try to extract the source and provide a patch.

Kivutar, 2016-04-04:
@duduke there are some packages that are using the unpack callback to extract archive. Take a look at them, if you want to automate the extraction of this deb.

duduke, 2016-04-04:
@Kivutar, can you give me a short guide on how to add a kernel module to Lakka?

Kivutar, 2016-04-04:
Take a look at those packages https://github.com/libretro/Lakka/tree/lakka/packages/linux-drivers
Then, add it in this list https://github.com/libretro/Lakka/blob/lakka/distributions/Lakka/options#L79

duduke, 2016-04-10:
I'm getting a compile error http://pastebin.com/pSdSrfCd complaining that it cannot find asm/bitsperlong.h

DerekRoth, 2016-06-07:
Hi,
Is there anything new on this ? I can help testing if needed.

Kivutar, 2016-06-09:
@duduke from what I see on google, this is a cross compilation issue. Reading the makefile and checking the environment variables and the paths would help. Also, I think this is packaged in Recalbox already, so taking a look at their package may help too.

duduke, 2016-06-09:
Thanks, I'll take a look at the RecalBox package.

Kivutar, 2016-06-09:
Their documentation about gamecon https://github.com/recalbox/recalbox-os/wiki/Gamecon-controllers-(EN)

And their package here https://github.com/recalbox/recalbox-buildroot/tree/bf58a540895d2a30c5d242ce161c86d4c9274720/package/gamecon_gpio_rpi

See how they pass ARCH and CROSS_COMPILE to make and that should fix your path issues