libretro/Lakka #343: Just Updated Lakka (Reflash) - Can't get connection via SSH anymore (RPi2)

Strider78, 2016-04-04:
Hello! 
I'm trying to connect to Lakka via SSH after new fresh install (Apr-3rd-2016) but getting this message instead of password request to login. RPi2.

~ Home$ ssh root@10.0.1.13
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
@    WARNING: REMOTE HOST IDENTIFICATION HAS CHANGED!     @
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
IT IS POSSIBLE THAT SOMEONE IS DOING SOMETHING NASTY!
Someone could be eavesdropping on you right now (man-in-the-middle attack)!
It is also possible that a host key has just been changed.
The fingerprint for the ECDSA key sent by the remote host is
SHA256:J/CFelCYWY5xC3x2YEh4H7q7BHecyBfxkmr1q+S3pTw.
Please contact your system administrator.
Add correct host key in /Users/Home/.ssh/known_hosts to get rid of this message.
Offending ECDSA key in /Users/Home/.ssh/known_hosts:1
ECDSA host key for 10.0.1.13 has changed and you have requested strict checking.
Host key verification failed.

Anybody having same issue ? 
Thank you guys in advance!

Kivutar, 2016-04-04:
Edit your .ssh/known_hosts and remove the last lakka line.

Strider78, 2016-04-04:
You're the man! Thank you Sir! It worked!

Kivutar, 2016-04-04:
If anybody knows a way to configure SSH to ignore known_hosts on local networks, please post it there.

closed 2016-04-04 by Kivutar