libretro/Lakka #344: [enhancement] Playlist Scanning MAME/FBA

fluffymadness, 2016-04-07:
Since i was a bit annoyed that I didn't have proper names with arcaderoms when I created a playlist, I wrote an extended bash script of this one.
http://www.lakka.tv/doc/Playlists/
Basically my script fetches the proper names of arcade roms from a local mame dat file (database.xml)
I hope its useful for someone :)

You can download the database from here
MAME Dats 0.172
http://www.progettosnaps.net/dats/

I just renamed the dat file to database.xml....

the getName() function fetches the name from the database
the checkIfChange() function checks if there were any new roms since last time (its a simple count function, so adding one new rom creates the playlist completely new)

Basically I wrote a systemd file and referenced the script, so that It gets called when I boot lakka
Well, here we go
(This script is more than arcade scanning, basically this is the whole playlist generation I use now)

`#!/bin/bash
getName () {
  #  name=$(wget -q -O - "$@" http://www.mamedb.com/game/$1 | grep -o -P '(?<=Name:&nbsp</b>).*(?=&nbsp;<br/>)')
  sed -n '/<machine name="'$1'"/,/<\/description>/p' database.xml > test.txt
  local name=$(sed -n 's:.*<description>\(.*\)</description>.*:\1:p' test.txt)
  echo $name
}

#$1 check if there were new roms added since last time
###################
checkIfChange (){	
	local ischange=2
	#if scanner startet for the first time create and initialize file with zero
	if [ ! -f $1/oldcount ]; then
		echo 0 > $1/oldcount
	fi
	countold=$(cat $1/oldcount)
	countnew=$(ls $1 -l | grep -v ^l | wc -l)
	echo $countnew > $1/oldcount
	if [ "$countold" = "$countnew" ]; then
		ischange=0
	else
		ischange=1
	fi
	echo $ischange	
}

scan () {
for fullpath in $1; do
	echo $fullpath
	filename=$(basename "$fullpath")
	echo "${filename%.*}"
	echo $2
	echo $3
	echo "DETECT"
	echo $4
done
}

scanmame () {
for fullpath in $1; do
	echo $fullpath
	filename=$(basename "$fullpath")
	gamename=$(getName ${filename%.*})
	echo $gamename
	if [ -z "$gamename" ]; then
		echo "Not in db" >&2
	else
		echo $2
		echo $3
		echo "DETECT"
		echo $4
	fi	
done
}

change=$(checkIfChange "/storage/roms/arcade")
changetwo=$(checkIfChange "/storage/roms/mame")

if [ "$change" -eq "1" ] || [ "$changetwo" -eq "1" ]; then
	scanmame "/storage/roms/arcade/*.zip" "/tmp/cores/fba_libretro.so" "Arcade (FB Alpha - SVN)" "FB Alpha - Arcade Games.lpl" > "/storage/playlists/FB Alpha - Arcade Games.lpl"
	scanmame "/storage/roms/mame/*.zip" "/tmp/cores/mame_libretro.so" "MAME" "FB Alpha - Arcade Games.lpl" >> "/storage/playlists/FB Alpha - Arcade Games.lpl"
fi

scan "/storage/roms/PSX/*.PBP" "/tmp/cores/pcsx_rearmed_libretro.so" "PCSX Rearmed" "Sony - Playstation.lpl" > "/storage/playlists/Sony - PlayStation.lpl"
scan "/storage/roms/megadrive/*.*" "/tmp/cores/genesis_plus_gx_libretro.so" "Genesis Plus" "Sega - Mega Drive - Genesis.lpl" > "/storage/playlists/Sega - Mega Drive - Genesis.lpl"
scan "/storage/roms/pcengine/*.*" "/tmp/cores/mednafen_pce_fast_libretro.so" "Mednafen PCE" "NEC - PC Engine - TurboGrafx 16.lpl" > "/storage/playlists/NEC - PC Engine - TurboGrafx 16.lpl"
#scan "/storage/roms/psp/*.*" "/tmp/cores/ppsspp_libretro.so" "PPSSPP " "Sony - PlayStation Portable.lpl" > "/storage/playlists/Sony - PlayStation Portable.lpl"
scan "/storage/roms/pcenginecd/*.cue" "/tmp/cores/mednafen_pce_fast_libretro.so" "Mednafen PCE" "NEC - PC Engine CD - TurboGrafx-CD.lpl" > "/storage/playlists/NEC - PC Engine CD - TurboGrafx-CD.lpl"
scan "/storage/roms/megacd/*.bin" "/tmp/cores/genesis_plus_gx_libretro.so" "Genesis Plus" "Sega - Mega-CD - Sega CD.lpl" > "/storage/playlists/Sega - Mega-CD - Sega CD.lpl"
`

fluffymadness, 2016-04-07:
Another thing that this one does is creating a merged playliste of mame and fba arcade roms.
The roms that don't work on fba are in my mame folder.
The script just goes through the arcade and mame directories and puts all the roms with the corresponding core into the playlist

fluffymadness, 2016-04-07:
@Kivutar 
is there anything planned scanner-wise for the future?
Because the no-intro databases don't cover every system.
Important ones like psx for example are missing...and the whole arcaderoms thing is not very comfortable/userfriendly atm  ^^"

Kivutar, 2016-04-15:
I pushed something yesterday in retroarch. But now I need to populate the database with the CRC of the zips.

fluffymadness, 2016-04-19:
awesome, thx :)

closed 2016-04-19 by fluffymadness