libretro/Lakka #349: RPi3 - Clean Install - Black Screen

zettam, 2016-04-29:
I got stuck at black screen after a clean installation. Nothing, upon boot, only a black screen.

zettam, 2016-04-29:
just tried on a rpi 2 -- same thing. tried with two different sd cards.

Kivutar, 2016-04-29:
Maybe the image you downloaded was corrupted during download. Please download again. Those images have been tested my many people (including me) so I can tell that they work.

zettam, 2016-04-29:
testing now once again

zettam, 2016-04-29:
retropie was previously installed on the disk, however I use win32 disk imager, would this effect the installation of the image?

Kivutar, 2016-04-29:
It should not affect it. Disk imager will copy bit per bit the image, erasing the partiton table, making any previous data unaccessible.

zettam, 2016-04-29:
hm same thing happening. tried with a third sd card, with a brand new download and win32diskimager installation. Just pure black.
this is the image I'm using 
http://mirror.lakka.tv/nightly/RPi.arm/Lakka-RPi.arm-devel-20160319231255-r21378-g78af2e1.img.gz

Kivutar, 2016-04-29:
I saw another bug report here, about Lakka not supporting an SD card. It could definitely be that!

zettam, 2016-04-29:
Oh, what does it support then? How do I make this run on a raspberry pi?

Kivutar, 2016-04-29:
Well, it supports every single card I tried so far. But there are so much types that I can't tell you. I got one report about one SD card nor compatible. And another report about an USB key.

zettam, 2016-04-29:
Hmm I use a sandisk ultra 8gb

zettam, 2016-04-29:
which microsd would you recommend in this case ?

Kivutar, 2016-04-29:
It was an RPi 1 image (not compatible with RPi2 and 3)

closed 2016-04-29 by Kivutar