libretro/Lakka #350: Bluez5 issue Dualshock 3

fluffymadness, 2016-04-29:
After buying the second bluetooth stick....I'm still unable to pair dualshock 3's with bluez...
Basically when starting the bluetooth daemon I'm getting the following errors.

bluetoothd[3481]: Failed to obtain handles for "Service Changed" characteristic
bluetoothd[3481]: Failed to read advertising features: Unknown Command (0x01)
bluetoothd[3481]: hci0 Load Connection Parameters failed: Unknown Command (0x01)

Does anyone else get the same, but still manages to pair?
I guess I managed to buy the exact same bluetoothstick that I have, but another brand.
Ironically the same stick is working with rpi2 + openelec + bluez and dualshocks.....

Kivutar, 2016-05-01:
Ok, if it works for openelec, we should be able to fix your problem.

Tell me everything about this dongle. I will check if some drivers are missing.

fluffymadness, 2016-05-09:
Hi. Here is some information about my dongle
Bus 003 Device 004: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          224 Wireless
  bDeviceSubClass         1 Radio Frequency
  bDeviceProtocol         1 Bluetooth
  bMaxPacketSize0        64
  idVendor           0x0a12 Cambridge Silicon Radio, Ltd
  idProduct          0x0001 Bluetooth Dongle (HCI mode)
  bcdDevice           88.91
  iManufacturer           0
  iProduct                2 CSR8510 A10
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:

fluffymadness, 2016-05-09:
On my raspberry 2, where bluetooth is working with the same dongle, I'm on openelec 6 stable.

Kivutar, 2016-05-10:
I'm reading this thread http://openelec.tv/forum/104-bluetooth-remotes/67730-openelec-3-2-3-and-modmypi-nano-bluetooth-adapter?limitstart=0 They give some information that could help.

gouchi, 2016-05-11:
There is an [issue](https://bugzilla.kernel.org/show_bug.cgi?id=103451) that should have been fixed for this dongle in particular.

fluffymadness, 2016-05-13:
Thank you for your links. I did some reading and posted an issue on the odroid forum.
http://forum.odroid.com/viewtopic.php?f=97&t=21197&p=141021#p141021
Regarding gouchis link, I probably have some fake chipset or so.
The patches won't be that easy to apply I guess, since there are some methods missing that are called in the patch
e.g
static int btusb_setup_csr(struct hci_dev *hdev) 
..
https://github.com/hardkernel/linux/blob/odroidxu3-3.10.y/drivers/bluetooth/btusb.c

fluffymadness, 2016-05-13:
http://blog.ruecker.fi/2013/10/06/adventures-in-bluetooth-4-0-part-i/
From this blog I installed the bluesuite which is linked in the comment section.

With this tool I can edit some parameters of the bluetooth stick
set bcdDevice, bootMode...etc

Kivutar, 2016-05-14:
So, did you change the bootmode?

fluffymadness, 2016-05-14:
I tried...but I think it was already in hci mode..cause it didn't change anything and hcitool itself gives my correct output
Lakka:~ # hcitool dev
Devices:
        hci0    00:1A:7D:DA:71:11

The following commands also work
Lakka:~ # hciconfig hci0 up
Lakka:~ # hciconfig
hci0:   Type: BR/EDR  Bus: USB
        BD Address: 00:1A:7D:DA:71:11  ACL MTU: 310:10  SCO MTU: 64:8
        UP RUNNING PSCAN
        RX bytes:589 acl:0 sco:0 events:34 errors:0
        TX bytes:1171 acl:0 sco:0 commands:34 errors:0

and sixad/sixpair also work with this dongle.
I really hope that someday well get a mainline kernel on xu3/xu4 ^^

Kivutar, 2016-05-26:
I tried bluetooth on my XU3, and the bluetooth service was not started automatically on boot. I had to start it manually.

fluffymadness, 2016-05-27:
Isnt that default behavior, that the service is disabled after a fresh lakka install, and that it has to be enabled manually under the settings->services tab, to be on after every reboot?
Or maybe your csr dongle also isnt playing nice ^^"

Kivutar, 2016-05-27:
It's the default behaviour, but even after enabling it, it will still not start on boot.

fluffymadness, 2016-05-27:
And these weird errors, do you have them too, if yes, maybe thats why the daemon refuses to start automatically. Bluetooth "seems" to work after starting that daemon manually, since i can do everything in bluetoothctl besides successfully pairing ds3s...

wakkawakka9k, 2016-09-08:
Having this same issue with my CSR dongle on XU4. I'd be interested in trying to debug this if someone could point me in the right direction. Thanks!

wakkawakka9k, 2016-09-13:
I put together a patch for the XU3 kernel that includes several changes from the upstream kernel that appear to have addressed issues with these particular CSR dongles (I can post it here later, but I don't have it handy). It's worth noting that the RC1 for the XU4 kernel appears to include these changes: https://github.com/hardkernel/linux/tree/odroidxu4-v4.2-rc1

Unfortunately, I'm having issues compiling Lakka to verify if this fixes anything.

When attempting to flash my SD card with the compiled image, Lakka hangs at the splash screen. If I update the kernel of an existing install that works, Lakka runs but there's a lot of broken stuff. I'm probably just doing something dumb when attempting to compile Lakka.

Any suggestions/ideas? Thanks!

Kivutar, 2016-09-13:
Yes, join us on IRC and I will help you to compile it.

wakkawakka9k, 2016-09-13:
Great, thanks. I'll try to hop on IRC later today.

wakkawakka9k, 2016-09-15:
So, I managed to compile in my patch, but alas, it didn't fix the issue.

I bought another bluetooth dongle with a totally separate chipset (Broadcom instead of CSR), and I'm seeing the same issues. I'm thinking there's some probably some lower level issue with Bluez or the kernel that's causing these issues.

I'm happy to keep hunting but I'm not really sure where to look for better debug information. Any suggestions are welcome.

Kivutar, 2016-09-16:
I don't have the XU3 with me now to test it. Have you created a post on the XU forums?

wakkawakka9k, 2016-09-16:
Not yet, I'll post something over there.

Kivutar, 2016-12-06:
I can confirm this bug is still there. And a duplicate of https://github.com/libretro/Lakka/issues/490