libretro/Lakka #354: DOSBOX!

fswolf, 2016-05-02:
DOSBOX NEEDS A LANCHER ICON

Can not manualy run core!

Kivutar, 2016-06-02:
I added DOSBox to every image. That's a start.

gouchi, 2016-06-02:
Did you try to create [configuration file](http://www.dosbox.com/wiki/Dosbox.conf#Sections) and check [compatibility status](http://www.dosbox.com/comp_list.php?letter=a) ?

Monanghan, 2016-11-23:
Hi,
i have the same problem. I can´t start the dosbox core. Screen turns black an the menu appears. The core is empty.
When i load the dosbox core, down in the left corner is the name from DosBox Version and lots of numbers and at the and the word "dirty" is this normal? 
I already create a configuration file and made a playlist. Games don´t start. Screen goes black and i got the message "could not load file from playlist" 
I also can´t start a game when i load the core and try to start a game manually when i pick the "game.exe" file. 
I run the newest lakka version on a intel pc hardware. Everything else work fine.  I don´t know what is or i do wrong? 
sry for my english i´m german ;-)

gouchi, 2016-11-23:
You need to create a [playlist](http://www.lakka.tv/doc/DOSBox) for now.

Monanghan, 2016-11-23:
As i said, i already created a playlist...Games don´t start. I got the message "could not load file from playlist"
my playlist: for example
/storage/roms/dos/ChaosEngine/game.conf
The Chaos Engine
/tmp/cores/dosbox_libretro.so
DOS (DOSBOX) 
DETECT
DOS.lpl
 
the .conf file is in the folder and like in the documentation.

gouchi, 2016-11-23:
Sorry I read too quickly. Is it possible to get the content of your game.conf ?

Directory name is case sensitive, so in your case you should have a directory named ChaosEngine.

Monanghan, 2016-11-23:
Ok here is my full game.conf file...thx for your help

# This is the configurationfile for DOSBox 0.74. (Please use the latest version of DOSBox)
# Lines starting with a # are commentlines and are ignored by DOSBox.
# They are used to (briefly) document the effect of each option.

[sdl]
#       fullscreen: Start dosbox directly in fullscreen. (Press ALT-Enter to go back)
#       fulldouble: Use double buffering in fullscreen. It can reduce screen flickering, but it can also result in a slow DOSBox.
#   fullresolution: What resolution to use for fullscreen: original or fixed size (e.g. 1024x768).
#                     Using your monitor's native resolution with aspect=true might give the best results.
#                     If you end up with small window on a large screen, try an output different from surface.
# windowresolution: Scale the window to this size IF the output device supports hardware scaling.
#                     (output=surface does not!)
#           output: What video system to use for output.
#                   Possible values: surface, overlay, opengl, openglnb, ddraw.
#         autolock: Mouse will automatically lock, if you click on the screen. (Press CTRL-F10 to unlock)
#      sensitivity: Mouse sensitivity.
#      waitonerror: Wait before closing the console if dosbox has an error.
#         priority: Priority levels for dosbox. Second entry behind the comma is for when dosbox is not focused/minimized.
#                     pause is only valid for the second entry.
#                   Possible values: lowest, lower, normal, higher, highest, pause.
#       mapperfile: File used to load/save the key/event mappings from. Resetmapper only works with the defaul value.
#     usescancodes: Avoid usage of symkeys, might not work on all operating systems.

fullscreen=true
fulldouble=false
fullresolution=original
windowresolution=original
output=surface
autolock=true
sensitivity=100
waitonerror=true
priority=higher,normal
mapperfile=mapper-0.74.map
usescancodes=true

[dosbox]
# language: Select another language file.
#  machine: The type of machine tries to emulate.
#           Possible values: hercules, cga, tandy, pcjr, ega, vgaonly, svga_s3, svga_et3000, svga_et4000, svga_paradise, vesa_nolfb, vesa_oldvbe.
# captures: Directory where things like wave, midi, screenshot get captured.
#  memsize: Amount of memory DOSBox has in megabytes.
#             This value is best left at its default to avoid problems with some games,
#             though few games might require a higher value.
#             There is generally no speed advantage when raising this value.

language=
machine=svga_s3
captures=capture
memsize=32

[render]
# frameskip: How many frames DOSBox skips before drawing one.
#    aspect: Do aspect correction, if your output method doesn't support scaling this can slow things down!.
#    scaler: Scaler used to enlarge/enhance low resolution modes.
#              If 'forced' is appended, then the scaler will be used even if the result might not be desired.
#            Possible values: none, normal2x, normal3x, advmame2x, advmame3x, advinterp2x, advinterp3x, hq2x, hq3x, 2xsai, super2xsai, supereagle, tv2x, tv3x, rgb2x, rgb3x, scan2x, scan3x.

frameskip=0
aspect=false
scaler=normal2x

[cpu]
#      core: CPU Core used in emulation. auto will switch to dynamic if available and appropriate.
#            Possible values: auto, dynamic, normal, simple.
#   cputype: CPU Type used in emulation. auto is the fastest choice.
#            Possible values: auto, 386, 386_slow, 486_slow, pentium_slow, 386_prefetch.
#    cycles: Amount of instructions DOSBox tries to emulate each millisecond.
#            Setting this value too high results in sound dropouts and lags.
[... 150 lines cut ...]
serial2=dummy
serial3=disabled
serial4=disabled

[dos]
#            xms: Enable XMS support.
#            ems: Enable EMS support.
#            umb: Enable UMB support.
# keyboardlayout: Language code of the keyboard layout (or none).

xms=true
ems=true
umb=true
keyboardlayout=auto

[ipx]
# ipx: Enable ipx over UDP/IP emulation.

ipx=false

[autoexec]
@echo off
mount d "/storage/roms/dos/ChaosEngine"
d:
chaos.exe

gouchi, 2016-11-23:
Can you try first with with a minimal one and then add your configuration ?

[autoexec]
@echo off
mount d "/storage/roms/dos/ChaosEngine"
d:
chaos.exe

Monanghan, 2016-11-23:
i did it with a minimal one...same thing...screen turns black an the info appears "could not load file from playlist"

maybe the file lakka can´t load the DosBox core from my playlist ? 
When is load the dosbox core, thename is somethin ...DosBox 2014-5-5..long number...-dirty
looks a bit strange...
on my desktop pc if i run retroarch and try to load DosBox i have no problems and the core name is DosBox 0.74 e1ab4fc

thx for your help

gouchi, 2016-11-24:
I test it on latest nightly and it works.

Can you try to get [Lakka system logs](http://www.lakka.tv/doc/Troubleshooting-Lakka/) ?

And it launch it with
retroarch -v -L /tmp/cores/dosbox_libretro.so /storage/roms/dos/ChaosEngine/game.conf  >> log.txt 2>&1

Monanghan, 2016-11-24:
thx for your effort
ok i will try...
i hope i did it correct but here is the lakka system log
```

RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 
Built: Nov  6 2016
RetroArch [INFO] :: Version: 1.3.6
RetroArch [INFO] :: Git: e218f0f
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Loading default config.
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Resetting undo buffers.
RetroArch [INFO] :: Loading dynamic libretro core from: "/tmp/cores/dosbox_libretro.so"
RetroArch [INFO] :: [overrides] no core-specific overrides found at /storage/.config/retroarch/config/DOSBox/DOSBox.cfg.
RetroArch [INFO] :: [overrides] no game-specific overrides found at /storage/.config/retroarch/config/DOSBox/game.cfg.
RetroArch [INFO] :: Shaders: preset directory: /usr/share/common-shaders/presets
RetroArch [INFO] :: Shaders: no game-specific preset found at /usr/share/common-shaders/presets/DOSBox/game.cgp.
RetroArch [INFO] :: Shaders: no game-specific preset found at /usr/share/common-shaders/presets/DOSBox/game.glslp.
RetroArch [INFO] :: Shaders: no core-specific preset found at /usr/share/common-shaders/presets/DOSBox/DOSBox.cgp.
RetroArch [INFO] :: Shaders: no core-specific preset found at /usr/share/common-shaders/presets/DOSBox/DOSBox.glslp.
RetroArch [INFO] :: Environ SET_SUPPORT_NO_GAME: yes.
RetroArch [INFO] :: Environ SET_VARIABLES.
RetroArch [INFO] :: Environ SET_CONTROLLER_INFO.
RetroArch [INFO] :: Controller port: 1
RetroArch [INFO] ::    Gamepad (2buttons) (ID: 517)
RetroArch [INFO] ::    Gamepad (4buttons) (ID: 1285)
RetroArch [INFO] ::    Joystick (2axis/2buttons) (ID: 261)
RetroArch [INFO] ::    Joystick (4axis/4buttons) (ID: 1029)
RetroArch [INFO] ::    Joystick (2axis/2buttons + kbd arrows) (ID: 773)
RetroArch [INFO] ::    Joystick (4axis/4buttons + kbd arrows) (ID: 1541)
RetroArch [INFO] ::    Mapper (ID: 1797)
RetroArch [INFO] :: Controller port: 2
RetroArch [INFO] ::    Gamepad (2buttons) (ID: 517)
RetroArch [INFO] ::    Gamepad (4buttons) (ID: 1285)
RetroArch [INFO] ::    Joystick (2axis/2buttons) (ID: 261)
RetroArch [INFO] ::    Joystick (4axis/4buttons) (ID: 1029)
RetroArch [INFO] ::    Mapper (ID: 1797)
RetroArch [INFO] :: Environ SYSTEM_DIRECTORY: "/storage/system".
RetroArch [INFO] :: Environ CORE_ASSETS_DIRECTORY: "/storage/roms/downloads".
RetroArch [INFO] :: Remaps: remap directory: /storage/remappings
RetroArch [INFO] :: Remaps: no game-specific remap found at /storage/remappings/DOSBox/game.rmp.
RetroArch [INFO] :: Remaps: no core-specific remap found at /storage/remappings/DOSBox/DOSBox.rmp.
RetroArch [INFO] :: Environ GET_LOG_INTERFACE.
RetroArch [libretro INFO] :: Logger interface initialized... 
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: XRGB8888.
RetroArch [INFO] :: Content loading skipped. Implementation will load it on its own.
RetroArch [libretro INFO] :: No configuration found, using core options
RetroArch [INFO] :: Environ GET_VARIABLE dosbox_machine_type:
RetroArch [INFO] :: 	hercules
RetroArch [INFO] :: Environ GET_VARIABLE dosbox_cpu_cycles_0:
RetroArch [INFO] :: 	0
RetroArch [INFO] :: Environ GET_VARIABLE dosbox_cpu_cycles_1:
RetroArch [INFO] :: 	0
RetroArch [INFO] :: Environ GET_VARIABLE dosbox_cpu_cycles_2:
RetroArch [INFO] :: 	0
RetroArch [INFO] :: Environ GET_VARIABLE dosbox_cpu_cycles_3:
RetroArch [INFO] :: 	1
RetroArch [INFO] :: Environ GET_VARIABLE dosbox_mapper_y:
RetroArch [INFO] :: 	1
RetroArch [INFO] :: Environ GET_VARIABLE dosbox_mapper_x:
RetroArch [INFO] :: 	1
RetroArch [INFO] :: Environ GET_VARIABLE dosbox_mapper_b:
RetroArch [INFO] :: 	1
RetroArch [INFO] :: Environ GET_VARIABLE dosbox_mapper_a:
RetroArch [INFO] :: 	1
RetroArch [INFO] :: Environ GET_VARIABLE dosbox_mapper_l:
RetroArch [INFO] :: 	1
RetroArch [INFO] :: Environ GET_VARIABLE dosbox_mapper_r:
RetroArch [INFO] :: 	1
RetroArch [INFO] :: Environ GET_VARIABLE dosbox_mapper_up:
RetroArch [INFO] :: 	1
RetroArch [INFO] :: Environ GET_VARIABLE dosbox_mapper_down:
RetroArch [INFO] :: 	1
RetroArch [INFO] :: Environ GET_VARIABLE dosbox_mapper_left:
RetroArch [INFO] :: 	1
RetroArch [INFO] :: Environ GET_VARIABLE dosbox_mapper_right:
RetroArch [INFO] :: 	1
RetroArch [INFO] :: Environ GET_VARIABLE dosbox_mapper_select:
RetroArch [INFO] :: 	1
RetroArch [INFO] :: Environ GET_VARIABLE dosbox_mapper_start:
RetroArch [INFO] :: 	1
RetroArch [INFO] :: Environ SET_KEYBOARD_CALLBACK.
RetroArch [libretro INFO] :: Configuring port: 0
RetroArch [libretro INFO] :: Keyboard Mapper: 0
RetroArch [libretro INFO] :: Configuring port: 1
RetroArch [libretro INFO] :: Keyboard Mapper: 0
RetroArch [INFO] :: Environ SET_INPUT_DESCRIPTORS:
RetroArch [INFO] :: 	RetroPad, User 1, Button "L2" => "Left Click"
RetroArch [INFO] :: 	RetroPad, User 1, Button "R2" => "Right Click"
Segmentation fault
```

gouchi, 2016-11-25:
Are you sure you gave a configuration file in parameter ? Because
`RetroArch [libretro INFO] :: No configuration found, using core options`

Can you retry using the gui ? Launch it like
`systemctl stop retroarch.service ; retroarch -v` 

Thank you.

Monanghan, 2016-11-25:
ok hmm maybe i made a mistake, i tried it again. now i got an info about the .conf file. 
```
RetroArch [libretro INFO] :: Configuration found at /storage/roms/dos/ChaosEngine/game.conf, ignoring core options
Segmentation fault

```
and after that the log ends. what is the gui? i did the commands with the program "putty"

gouchi, 2016-11-25:
I mean launching it with 

` systemctl stop retroarch.service ; retroarch -v` by ssh.

Then into RetroArch, launch the game in the DOS playlist.

Monanghan, 2016-11-25:
ok  i hope i did it correctly...but unfortunately it´s not starting the game...message "could not load file from playlist" 
damn in most of the cases the problem sits infront of the monitor. but  i don´t know what i´m doing wrong. i checked the playlist and the game.conf the hole time. why isn´t it launching the game...argh 

thanks for your help. in the ssh i got a log...maybe there is some helpful information.

```
RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3
Built: Nov  6 2016
RetroArch [INFO] :: Version: 1.3.6
RetroArch [INFO] :: Git: e218f0f
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Loading default config.
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch                                                                                                                                                             .cfg".
RetroArch [INFO] :: Resetting undo buffers.
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29999.51 Hz.
RetroArch [INFO] :: [SoftFilter]: Found plug: 2xBR (2xbr).
RetroArch [INFO] :: [SoftFilter]: Found plug: 2xSaI (2xsai).
RetroArch [INFO] :: [SoftFilter]: Found plug: Blargg NTSC SNES (blargg_ntsc_snes                                                                                                                                                             ).
RetroArch [INFO] :: [SoftFilter]: Found plug: Darken (darken).
RetroArch [INFO] :: [SoftFilter]: Found plug: EPX (epx).
RetroArch [INFO] :: [SoftFilter]: Found plug: LQ2x (lq2x).
RetroArch [INFO] :: [SoftFilter]: Found plug: Phosphor2x (phosphor2x).
RetroArch [INFO] :: [SoftFilter]: Found plug: Scale2x (scale2x).
RetroArch [INFO] :: [SoftFilter]: Found plug: Super2xSaI (super2xsai).
RetroArch [INFO] :: [SoftFilter]: Found plug: SuperEagle (supereagle).
RetroArch [INFO] :: Using 1 threads for softfilter.
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bilinear.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/nearest.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/stock.glsl"
RetroArch [INFO] :: Video @ fullscreen
RetroArch [INFO] :: Starting threaded video driver ...
RetroArch [INFO] :: [DRM]: Found 3 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: no
RetroArch [INFO] :: [DRM]: Connector 0 has 0 modes.
RetroArch [INFO] :: [DRM]: Connector 1 connected: no
RetroArch [INFO] :: [DRM]: Connector 1 has 0 modes.
RetroArch [INFO] :: [DRM]: Connector 2 connected: yes
RetroArch [INFO] :: [DRM]: Connector 2 has 10 modes.
RetroArch [INFO] :: [DRM]: Connector 2 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Mode 0: (1024x768) 1024 x 768, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 1: (1024x768) 1024 x 768, 70 Hz
RetroArch [INFO] :: [DRM]: Mode 2: (1024x768) 1024 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 3: (800x600) 800 x 600, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 4: (800x600) 800 x 600, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 5: (800x600) 800 x 600, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 6: (640x480) 640 x 480, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 7: (640x480) 640 x 480, 73 Hz
RetroArch [INFO] :: [DRM]: Mode 8: (640x480) 640 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 9: (720x400) 720 x 400, 70 Hz
RetroArch [INFO] :: Found GL context: kms
RetroArch [INFO] :: Detecting screen resolution 1024x768.
RetroArch [INFO] :: [EGL]: EGL version: 1.4
RetroArch [INFO] :: [EGL]: Current context: 0xb3a068c0.
RetroArch [INFO] :: [KMS]: New FB: 1024x768 (stride: 4096).
RetroArch [INFO] :: [GL]: Vendor: nouveau, Renderer: Gallium 0.4 on NVA8.
RetroArch [INFO] :: [GL]: Version: 3.0 Mesa 11.2.0-rc1.
RetroArch [INFO] :: [GL]: Using ARB_sync to reduce latency.
RetroArch [INFO] :: GL: Using resolution 1024x768
RetroArch [INFO] :: [GL]: Default shader backend found: glsl.
RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
RetroArch [INFO] :: Checking GLSL shader support ...
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
[... 57 lines cut ...]
RetroArch [INFO] :: Loading dynamic libretro core from: "/tmp/cores/dosbox_libretro.so"
RetroArch [INFO] :: Environ SET_SUPPORT_NO_GAME: yes.
RetroArch [INFO] :: Environ SET_VARIABLES.
RetroArch [INFO] :: Environ SET_CONTROLLER_INFO.
RetroArch [INFO] :: Controller port: 1
RetroArch [INFO] ::    Gamepad (2buttons) (ID: 517)
RetroArch [INFO] ::    Gamepad (4buttons) (ID: 1285)
RetroArch [INFO] ::    Joystick (2axis/2buttons) (ID: 261)
RetroArch [INFO] ::    Joystick (4axis/4buttons) (ID: 1029)
RetroArch [INFO] ::    Joystick (2axis/2buttons + kbd arrows) (ID: 773)
RetroArch [INFO] ::    Joystick (4axis/4buttons + kbd arrows) (ID: 1541)
RetroArch [INFO] ::    Mapper (ID: 1797)
RetroArch [INFO] :: Controller port: 2
RetroArch [INFO] ::    Gamepad (2buttons) (ID: 517)
RetroArch [INFO] ::    Gamepad (4buttons) (ID: 1285)
RetroArch [INFO] ::    Joystick (2axis/2buttons) (ID: 261)
RetroArch [INFO] ::    Joystick (4axis/4buttons) (ID: 1029)
RetroArch [INFO] ::    Mapper (ID: 1797)
RetroArch [INFO] :: Environ SYSTEM_DIRECTORY: "/storage/system".
RetroArch [INFO] :: Environ CORE_ASSETS_DIRECTORY: "/storage/roms/downloads".
RetroArch [INFO] :: Environ GET_LOG_INTERFACE.
RetroArch [libretro INFO] :: Logger interface initialized...
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: XRGB8888.
Segmentation fault
```

gouchi, 2016-11-25:
I can't reproduce your segmentation fault. Can you try with another game ? On which device are you running Lakka ?

Monanghan, 2016-11-25:
i´m running lakka with:
Intel Pentium Dual Core 1,8Ghz
Memory 4gb
GeForce 210 512MB
Asustek Motherboard
old tft monitor from hyundai
wd 500gig hdd 
xbox control

Monanghan, 2016-11-26:
i think i make a complete reinstall of the system...maybe the problem disapear
when this is not working...i think i give up....

gouchi, 2016-11-26:
You can try to come to our [irc channel](http://webchat.freenode.net/?channels=%23lakkatv) to get more help.

ronilaukkarinen, 2017-01-07:
I'm having the same problem. Segfaults before loading the game.

````
RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 
Built: Dec 21 2016
RetroArch [INFO] :: Version: 1.3.6
RetroArch [INFO] :: Git: 274b40f
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Loading default config.
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Incompatible shader for backend gl, clearing...
RetroArch [INFO] :: Resetting undo buffers.
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bilinear.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/nearest.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/stock.glsl"
RetroArch [INFO] :: Video @ fullscreen
RetroArch [INFO] :: Starting threaded video driver ...
RetroArch [INFO] :: [DRM]: Found 2 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
RetroArch [INFO] :: [DRM]: Connector 0 has 1 modes.
RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 1 connected: no
RetroArch [INFO] :: [DRM]: Connector 1 has 0 modes.
RetroArch [INFO] :: [DRM]: Mode 0: (1024x600) 1024 x 600, 60 Hz
RetroArch [INFO] :: Found GL context: kms
RetroArch [INFO] :: Detecting screen resolution 1024x600.
RetroArch [INFO] :: [EGL]: EGL version: 1.4
RetroArch [INFO] :: [EGL]: Current context: 0xb44077b0.
RetroArch [INFO] :: [KMS]: New FB: 1024x600 (stride: 4096).
RetroArch [INFO] :: [GL]: Vendor: Intel Open Source Technology Center, Renderer: Mesa DRI Intel(R) 945GME .
RetroArch [INFO] :: [GL]: Version: 2.1 Mesa 11.2.0-rc1.
RetroArch [INFO] :: GL: Using resolution 1024x600
RetroArch [INFO] :: [GL]: Default shader backend found: glsl.
RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
RetroArch [INFO] :: Checking GLSL shader support ...
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: [GL]: Using 4 textures.
RetroArch [INFO] :: [GL]: Loaded 1 program(s).
RetroArch [INFO] :: [GL]: Using GL_RGB565 for texture uploads.
RetroArch [INFO] :: Using font rendering backend: freetype.
RetroArch [INFO] :: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
RetroArch [INFO] :: [udev] Adding device /dev/input/event5 as type ID_INPUT_KEYBOARD.
RetroArch [INFO] :: [udev] Adding device /dev/input/event6 as type ID_INPUT_MOUSE.
RetroArch [INFO] :: [udev] Adding device /dev/input/mouse0 as type ID_INPUT_MOUSE.
[... 49 lines cut ...]
RetroArch [INFO] ::    Gamepad (2buttons) (ID: 517)
RetroArch [INFO] ::    Gamepad (4buttons) (ID: 1285)
RetroArch [INFO] ::    Joystick (2axis/2buttons) (ID: 261)
RetroArch [INFO] ::    Joystick (4axis/4buttons) (ID: 1029)
RetroArch [INFO] ::    Joystick (2axis/2buttons + kbd arrows) (ID: 773)
RetroArch [INFO] ::    Joystick (4axis/4buttons + kbd arrows) (ID: 1541)
RetroArch [INFO] ::    Mapper (ID: 1797)
RetroArch [INFO] :: Controller port: 2
RetroArch [INFO] ::    Gamepad (2buttons) (ID: 517)
RetroArch [INFO] ::    Gamepad (4buttons) (ID: 1285)
RetroArch [INFO] ::    Joystick (2axis/2buttons) (ID: 261)
RetroArch [INFO] ::    Joystick (4axis/4buttons) (ID: 1029)
RetroArch [INFO] ::    Mapper (ID: 1797)
RetroArch [INFO] :: Environ SYSTEM_DIRECTORY: "/storage/system".
RetroArch [INFO] :: Environ CORE_ASSETS_DIRECTORY: "/storage/roms/downloads".
RetroArch [INFO] :: Remaps: remap directory: /storage/remappings
RetroArch [INFO] :: Remaps: no game-specific remap found at /storage/remappings/DOSBox/tyrian.rmp.
RetroArch [INFO] :: Remaps: no core-specific remap found at /storage/remappings/DOSBox/DOSBox.rmp.
RetroArch [INFO] :: Environ GET_LOG_INTERFACE.
RetroArch [libretro INFO] :: Logger interface initialized... 
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: XRGB8888.
RetroArch [INFO] :: Content loading skipped. Implementation will load it on its own.
RetroArch [libretro INFO] :: Configuration found at /storage/roms/dos/TYRIAN/tyrian.conf, ignoring core options
Segmentation fault
````

ronilaukkarinen, 2017-01-07:
Worth mention, using eeePC and `Lakka-Generic.i386-devel-20161221183902-r21729-g274b40f-efi`. Fresh install.

gouchi, 2017-01-08:
I reproduce the segfault on i386 but not on x86-64.