libretro/Lakka #355: unpacked file not deleted/cleaned when ROM fails to load

cudencuden, 2016-05-03:
Hi, I am using no-intro romset of GBA and Sega 32X. I left both sets as zip and scanned them fine. Though, when I try to launch games and it fails to load..the game in question gets its content extracted but doesn't get deleted... With other roms (ex: genesis) i can see that the file within the zip gets extracted when i am launching/playing the game..but gets deleted when i close the game.

Kivutar, 2016-05-04:
Where did you took your no-intro roms? archive.org?

cudencuden, 2016-05-04:
yes. using No-Intro-Collection 2016-01-03 Fixed from them

Kivutar, 2016-05-04:
I'm using the same, and I just tested GBA and 32X now on the latest release, and got it working.
Let try the retroarch.cfg fix. But I don't think it can fix that issue.

cudencuden, 2016-05-04:
but either way...shouldn't the extracted file be deleted...regardless if game launched successfully or not? if you delete your gba bios (forcing a fail); does your extracted game files delete itself?

cudencuden, 2016-05-05:
for gba issue, the issue is with core gpsp...gpsp is the default core if i play game from playlist...if i load rom manually and select mGBA core...game loads fine.

Kivutar, 2016-05-05:
I'm using gpsp without problem. Are you using the gpsp from lakka or from the buildbot?

cudencuden, 2016-05-05:
uhmmmm...buildbot? how can I tell? I did used online updater to update all the cores including gpsp

Kivutar, 2016-05-06:
The cores from the buildbot shouldn't be used. They are outdated.

I'm going to simply disable the Core Updater, and anything buildbot related from Lakka.

We can't keep giving access to outdated stuff like that, while our own cores are kept up to date.

> how can i get the cores from lakka?

They are already there, in /usr/lib/libretro

cudencuden, 2016-05-06:
@Kivutar , i'll keep that in mind...i reimaged though. sega 32x and gba roms works now. thanks

closed 2016-05-06 by cudencuden

Kivutar, 2016-05-06:
No problem. Things from the Online Updater, like Core Updater, and Assets, are from the libretro buildbot. And they can override Lakka files. It can become problematic in many cases, so I'm going to hide those.

I will just show "Update Lakka" and "Thumbnails Updater" in the "Online Updater".

Kivutar, 2016-05-06:
BTW, I'm packaging mgba and will enable it on as much platforms as possible.

cudencuden, 2016-05-06:
okay. thank you very much for the help

twinaphex, 2016-05-06:
A better solution would be talking with radius to update the arm cores and everything that is needed for lakka.

Kivutar, 2016-05-06:
That's not a solution to 100% of the problem.

When a user downloads the joypad configs from the updater, then upgrade lakka, the files from the updater still override the ones from lakka.

If this user forgets that he have used the updater, those files stay here, and they complain that the new features like PS3 auto pairing doesn't work.

Same for the assets, they get it from the updater, then forget it. The day we change the icon names in xmb, they start getting wrong icons in their interface and complain to me.

They don't get that lakka already include the cores, compiled with all the optimizations for their platform. They think that they have to get the cores from the updater, because we expose it.

And the cores on the buildbot are not only outdated, they are also incompatible. For example gpsp fails to launch. mednafen-psx launches but have glitches, etc.

So for all these reasons, I'd better keep things really simple, and provide only one way to upgrade.

twinaphex, 2016-05-06:
I merged your PR, but as I stated in the PR, I do want to prevent RA and Lakka to grow too far apart, so we should eventually look at ways to come up with a solution that works for both. Having a whole bunch of HAVE_LAKKA ifdefs all over the place is going to get unmaintainable pretty soon and I want to avoid that.

Kivutar, 2016-05-06:
Ok thanks. I have an idea that might work to remove some HAVE_LAKKA. We can discuss it on IRC.