libretro/Lakka #357: [Request] Select favorites roms to create a short list on each console

MopheusDG, 2016-05-03:
I would be nice to be able to select favorites games from each console. For example if I have 20 games but I have a favorites one, find some way to filter favorites to speed up game searching. Thanks.

cudencuden, 2016-05-04:
make your own playlist

Kivutar, 2016-05-04:
I'd like to make a favorite tab.

We need some refactoring, because for now, the tabs are hardcoded.

Sage222, 2016-10-26:
Great idea. I was thinking the same thing.

sparksterz, 2017-01-03:
So, I'm going through something similar now too. I don't necessarily have an issue with the hard coded tabs so much and I understand the limitation on needing a playlist file... Would it make more sense to have a favorite toggle in each playlist? 

Ex. I see all games in the playlist, when selecting it instead of running it I have an option similar to information to toggle favorite? Or is that view not set up well to write and/or reprocess a playlist?