libretro/Lakka #360: Kodi?

Ali3n0id, 2016-05-06:
Forgive me I couldn't find it in your FAQs but is it possible to access the full kodi interface through Lakka?

cudencuden, 2016-05-07:
http://libretro.com/forums/showthread.php?t=3524&highlight=kodi

Kivutar, 2016-05-07:
You may want to take a look at garbear's RetroPlayer also. Not sure if it is finished yet.

Having KODI, or an alternative, in Lakka would add value to the project for sure. But it's too much work for me for now. With our limited forces, we should focus on providing a perfect gaming experience first.

closed 2016-05-11 by Kivutar

matthuisman, 2016-08-25:
For anyone else looking, I have created nice dual boot images for the Pi to allow for Lakka + KODI.
Either OSMC or LibreELEC.

You can get them here:
www.matthuisman.nz/2016/08/lakka-kodi-dual-boot-images.html

dabit1, 2017-02-22:
Hi @matthuisman! First of all great job! Unfortunatelly im trying to do that in my laptop instead of raspberry. It would be nice if you made a tutorial...

matthuisman, 2017-02-22:
@dabit1 
Sorry, they only work on the Raspberry Pi due to NOOBS only being for the Pi.

philj0st, 2017-05-22:
i think the OP maybe meant to ask whether it is possible to have a menu entry for Kodi and then bind some command that launches Kodi to it. .. not having the libretro team integrate Kodi in Lakka ..