libretro/Lakka #361: HDMI audio not working (even changing the retroarch.cfg)

danbacastro, 2016-05-10:
Hi everybody,
I'm trying to setup Lakka on my notebook. 

I ran the aplay -L command:

null
    Discard all samples (playback) or generate zero samples (capture)
hdmi:CARD=Generic,DEV=0
    HD-Audio Generic, HDMI 0
    HDMI Audio Output
default:CARD=SB
    HDA ATI SB, 92HD81B1X5 Analog
    Default Audio Device
sysdefault:CARD=SB
    HDA ATI SB, 92HD81B1X5 Analog
    Default Audio Device
front:CARD=SB,DEV=0
    HDA ATI SB, 92HD81B1X5 Analog
    Front speakers
surround21:CARD=SB,DEV=0
    HDA ATI SB, 92HD81B1X5 Analog
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=SB,DEV=0
    HDA ATI SB, 92HD81B1X5 Analog
    4.0 Surround output to Front and Rear speakers
surround41:CARD=SB,DEV=0
    HDA ATI SB, 92HD81B1X5 Analog
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=SB,DEV=0
    HDA ATI SB, 92HD81B1X5 Analog
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=SB,DEV=0
    HDA ATI SB, 92HD81B1X5 Analog
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=SB,DEV=0
    HDA ATI SB, 92HD81B1X5 Analog
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers

And changed the audio_device="hdmi:CARD=Generic,DEV=0" in the retroarch.cfg, but no sound comes out from the TV. When I change the audio_device to the others options, the sound comes out from the notebook.

Anybody already solved this problem?

Kivutar, 2016-05-10:
Hello, try to add radeon.audio=1 in the cmdline

danbacastro, 2016-05-10:
Hi Kivutar, thanks for the answer, but how am I supposed to do that? I'm noob on Linux, sorry...

gouchi, 2016-05-10:
You need to add radeon.audio=1 to the boot arguments. You can press Tab in the boot menu to edit the cmdline and your boot arguments. If you want to make permanent try this [documentation](http://wiki.openelec.tv/index.php/Boot_Arguments). 

If it doesn't work, try nightly builds ([i386](http://mirror.lakka.tv/nightly/Generic.i386/Lakka-Generic.i386-devel-20160507183144-r21434-gd23434f.img.gz or `http://mirror.lakka.tv/nightly/Generic.x86_64/Lakka-Generic.x86_64-devel-20160507182037-r21434-gd23434f.img.gz`) because we have included a patch which fix audio hdmi.

Moreover, now you can [switch audio device](http://www.lakka.tv/doc/Audio-settings/#switching-audio-device-from-the-gui) directly in the gui.

danbacastro, 2016-05-10:
I added the radeon.audio=1 argument. Now my cmdline is the following:

root=/dev/ram0 rdinit=/init usbcore.autosuspend=-1 BOOT_IMAGE=/KERNEL radeon.audio=1 boot=LABEL=System disk=LABEL=Storage ssh quiet

Also, i tried to switch audio device in the GUI, but it still not working and the audio continues coming from the notebook. I will try nightly builds i386.

Thank you very much!

danbacastro, 2016-05-11:
I installed nightly build i386 and now the sound is right!

Thanks!

Kivutar, 2016-05-11:
Thanks, I close it.

closed 2016-05-11 by Kivutar