libretro/Lakka #364: Error running mame games

Fransebas, 2016-05-12:
I have the last version of lakka i386-2016 running in my computer and runs everything fine except for mame games I run the game on the console and get the following error

# retroarch -L mame2003_libretro.so ~/ROMs/metalSlug/mslug2.zip

241-p1.bin   NOT FOUND
241-p2.bin   NOT FOUND
241-s1.bin   NOT FOUND
sfix.sfx     NOT FOUND
sp-s2.sp1    NOT FOUND
sm1.sm1      NOT FOUND
241-m1.bin   NOT FOUND
000-lo.lo    NOT FOUND
241-v1.bin   NOT FOUND
241-v2.bin   NOT FOUND
241-c1.bin   NOT FOUND
241-c2.bin   NOT FOUND
241-c3.bin   NOT FOUND
241-c4.bin   NOT FOUND

cudencuden, 2016-05-12:
MAME 0.78 romsets

Fransebas, 2016-05-12:
@cudencuden do I download the "MAME 0.78 romsets"? and if I do, where do I set it on the system folder or in the same folder as the mame 2003?

Kivutar, 2016-05-13:
IIRC, MAME doesn't use the system folder, just the roms folder.

fr500, 2016-05-13:
actually, several mame cores use the system folder, I think 078 included, but not for bioses or anything.

Fransebas, 2016-05-13:
@Kivutar my mistake mame don't use the system folder, anyway, what's wrong whit the already installed mame_2003? do I need to install in cores another version like mame_2015?

Kivutar, 2016-05-14:
Maybe you're missing a parent ROM or something like that. You keep the whole set in your roms folder?

gouchi, 2016-05-27:
You are missing neogeo bios.

Kivutar, 2016-06-01:
Are you running the mame core from Lakka or from the buildbot/updater?

Fransebas, 2016-06-02:
I have the neo geo zip in the folder, I think is a problem with the BIOS mame2013, I think I'll download the mame2015 but I don't know if I should install it in the same folder as the other BIOS that I installed manually or in other place

Kivutar, 2016-06-03:
Don't use the cores from the buildbot, they are incompatible with lakka. Delete everything you have in your cores folder

Kivutar, 2016-07-10:
@Fransebas any news from this?

Kivutar, 2016-08-07:
No news, I close it

closed 2016-08-07 by Kivutar