libretro/Lakka #365: Lakka sending 3D signal to TV

LonelyTVy, 2016-05-13:
I recently installed the latest version of Lakka on my Raspberry Pi 2. I hooked up the Pi 2 to my Sony LCD TV that has 3D capabilities. Once it boots, after a few moments the TV detects a 3D signal from the Raspberry Pi 2 and switches over to 3D mode. I can turn it off on the TV, but the XMB appears stretched and cut off.  When I load a game the image remains the same. My TV won't let me adjust the "zoom" of the image because it is still getting a 3D signal. I have tried disabling overscan and vsync but I get the same issue.

Kivutar, 2016-05-13:
What if you switch to rgui, and what happens when launching a 3D game (n64 or psp) ?

LonelyTVy, 2016-05-13:
Switching to rgui still forced the 3D setting and while running a 3D game it did the same. However, I did manage to get it to not switch over to 3D when I pulled the HDMI cable out of the TV and put it back in when a PSX game was playing. But, if I switched to the menu it switched back over to 3D.

Kivutar, 2016-05-14:
I have to admit that I have no idea how to help you on this... Maybe you could try changing some display settings in the config.txt of your Raspberry Pi. There are

LonelyTVy, 2016-05-14:
> how does the 3D games look on a 3D TV?
Without proper 3D settings the games look like a mess. Because the Raspberry Pi and Lakka are sending out a 3D signal for a non-3D stream it overlaps most of the images. Normally I can fake 3D on my TV for a 2D signal and most of the case they look like 2D with a little depth.