libretro/Lakka #366: Please add a "Donate" button also on lakka.tv

fmaida, 2016-05-13:
Hello,

I just discovered that you accept donations and I've just sent a little contribution, but I didn't found this info by reading your website but instead I found a Paypal donate button on the LibRetro website at http://www.libretro.com/index.php/contributions/

On lakka.tv I didn't found a section that explains how someone could contribute to your project by donating... IMHO you could add a section in the FAQ page where you could state that you accept donations. I think that some people that are unaware of this will be more than happy to send their monetary contributions.

Also, an idea for the future: could you take into consideration the idea of joining a crowdfunding network like Patreon to get recurring donations ? The URL is http://www.patreon.com

Thanks

Kivutar, 2016-05-13:
Yeah you're right. I added one here http://www.lakka.tv/doc/Wishlist/

This page is accessible from here http://www.lakka.tv/doc/Home/

Is it enough?

I don't know yet about patreon.

fmaida, 2016-05-13:
Thank you.

I took advantage of this new link by sending another donation :-)
Hope that it'll cheer up your day and help somehow the further development of Lakka

closed 2016-05-13 by fmaida

Kivutar, 2016-05-14:
Thanks a lot :)