libretro/Lakka #367: Bluetooth on Raspberry Pi 3

kamelopardus, 2016-05-13:
Hi!
I try to use bluetooth on my RPi3 and here's the status I've got

`Lakka:~ # systemctl status bluetooth
● bluetooth.service - Bluetooth service
   Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2016-05-13 11:27:51 UTC; 13min ago
 Main PID: 546 (bluetoothd)
   Status: "Running"
   CGroup: /system.slice/bluetooth.service
           └─546 /usr/lib/bluetooth/bluetoothd

May 13 11:27:51 Lakka systemd[1]: Started Bluetooth service.
May 13 11:27:51 Lakka bluetoothd[546]: Bluetooth daemon 5.37
May 13 11:27:51 Lakka bluetoothd[546]: Starting SDP server
May 13 11:27:51 Lakka bluetoothd[546]: kernel lacks bnep-protocol support
May 13 11:27:51 Lakka bluetoothd[546]: System does not support network plugin
May 13 11:27:51 Lakka bluetoothd[546]: Failed to open RFKILL control device
May 13 11:27:51 Lakka bluetoothd[546]: Bluetooth management interface 1.9 initialized
May 13 11:27:51 Lakka bluetoothd[546]: Failed to obtain handles for "Service Changed" characteristic
May 13 11:30:17 Lakka systemd[1]: Started Bluetooth service.`

As a result, DualShock3 device can't be paired.

Kivutar, 2016-05-14:
To me, this log says that bluetooth is launched.

What is the model of your DS3? The official one from Sony, or one of the Shanwan/Gasia clones?

kamelopardus, 2016-05-14:
Well, yes it's some kind of nonofficial clone. Am i hopeless?

Kivutar, 2016-05-14:
If it's Gasia, it will not work for now. You're not exactly hopeless. But you will have to wait for weeks.

kamelopardus, 2016-05-14:
Yup, dmesg gives me

`[  108.409928] sony 0003:054C:0268.0001: input,hiddev0,hidraw0: USB HID v1.11 Joystick [Gasia Co.,Ltd PS(R) Gamepad] on usb-3f980000.usb-1.3/input0`

No problem, I will wait any time required. Thanks!

Kivutar, 2016-05-26:
I created this issue https://github.com/libretro/Lakka/issues/384

I close this one

closed 2016-05-26 by Kivutar