libretro/Lakka #368: How to add XMB Custom Menus

danbacastro, 2016-05-13:
Hello everybody,

I'm trying to add some icons to a custom menu or to an existing one, but I get a message that the icon files are Read-only file system.

How do I add custom xmb menus? How do I change existing xmb menus?

Kivutar, 2016-05-14:
You can override the assets hierarchy in /storage/assets, it's an overlay to /usr/share/retroarch-assets

danbacastro, 2016-05-15:
Thanks Kivutar!

danbacastro, 2016-05-16:
Sorry, I tried to override, but I don't know how to do it. Could you give me a brief explanation?

Kivutar, 2016-05-16:
Sure, you see the files in /usr/share/retroarch-assets?

There is a folder called xmb, that contains the icon themes

The default icon theme is "monochrome". The monochrome folder contains a png folder, where the icons are.

Let say that you want to change the icon for the SNES, its name is Nintendo - Super Nintendo Entertainment System.png

You need to create an icon with the same path in /storage/assets

So you will place this icons at /storage/assets/xmb/monochrome/png/Nintendo - Super Nintendo Entertainment System.png

danbacastro, 2016-05-16:
Now, it's working!

The retroarch.cfg file points to the assets directory = "/tmp/assets". Is it right?

Kivutar, 2016-05-16:
Yes, this /tmp/assets is the merge of both /usr/share/retroarch-assets and /storage/assets

danbacastro, 2016-05-16:
Wow, now I got it. Thank you very much!

closed 2016-05-16 by danbacastro