libretro/Lakka #369: Menu couldn't display chinese filenames

9chu, 2016-05-14:
It seems those utf-8 character can't be properly decoded.
I also tried to change the default font for menu, but it didn't work.

Kivutar, 2016-05-14:
Hello, yeah RetroArch doesn't support yet UTF-8. You can redeclare this issue on their tracker.

gouchi, 2016-05-14:
Please check #2007

Kivutar, 2016-05-14:
@gouchi are you sure you pasted the right link?

bparker06, 2016-09-17:
Initial UTF8 support for RetroArch was added last month, but currently it only supports the first 256 codepoints, so just Latin-1. This enables languages like French, Spanish, German etc. but the current method of font rendering makes it difficult to support more, so we're still considering options on how to move forward there.

Kivutar, 2016-10-29:
Thanks for the chinese translation @9chu

closed 2016-10-29 by Kivutar