libretro/Lakka #372: Lakka and JXD S192

luciano2000, 2016-05-14:
Hi.
We know that the new JXD Singularity S192 is probably the most powerful emulation machine. It uses the Nvidia K1 processor, with 192 GPU cores !!!
But it uses Android. I'm not a android hater, but I want a exclusive retro machine and I would love to joint this two technologies.

The question is:
It's possible to replace (in NAND?) the original Android OS to a Lakka OS ?

fr500, 2016-05-14:
Not really any more powerful than the Shield tablet that released two years ago.
Regarding the question, hard to tell, most likely it has a locked bootloader and it hasn't even released.

Kivutar, 2016-05-26:
Twinaphex is trying to port Lakka to the TK1. It would certainly help.

But the port is not easy to achieve at all. The important bits are hidden in a ton of bash scripts. The scripts are compatible with Ubuntu only, etc.

So for now, I'm not interested in supporting this kind of devices. It's not open enough.

closed 2016-05-26 by Kivutar