libretro/Lakka #373: Lakka with Berryboot on RPi2

ollef, 2016-05-15:
I can't get recent nightlies of Lakka to work properly with Berryboot. Has anyone else tried this recently? The system boots but "stuff" seems to be missing (e.g. icons in the interface) and the controller doesn't work.

The Systemd logs show that the `tmp-{cores,database,joypads}.mount` units fail. I haven't verified the cause completely but have some suspicions: If I roll back to a nightly before the date of https://github.com/libretro/Lakka/commit/d22bff9ecad28db92928603e63bdc65e81a57ac6 it seems to work again, so there might be a problem with OverlayFS in combination with the way Berryboot works.

Might be worth looking into! :)

Cheers!

Kivutar, 2016-05-15:
Does berryboot override the kernel or something like that?

ollef, 2016-05-15:
It seems that way. `uname -a` inside Lakka yields `Linux berryboot 4.1.7v7-aufs #1 SMP PREEMPT Wed Sep 16 13:19:27 CEST 2015 armv7l GNU/Linux`.

Kivutar, 2016-05-16:
Well, they take a lot of risks of breaking something by overriding our kernel... definitely not a good idea.

R3d-Sky, 2016-05-22:
@Kivutar I'm guessing that maxnet/berryboot#201 already discusses OverlayFS. Quoting from @maxnet's post, OverlayFS support was added in 3.18, and berryboot uses an older kernel to support Allwiner and Amlogic chips. I'm not 100% sure though, so please don't quote me on it.

Kivutar, 2016-05-22:
Another important thing to know, is even if they apply an overlayfs patch on the old kernel they use, it will still be incompatible. Because the old overlayfs patches were using the name "overlayfs" for the module, while the newest one use the name "overlay"

maxnet, 2016-09-10:
>Psuedo-Quoting from the posts, OverlayFS support was added in 3.18, and berryboot uses an older kernel to support Allwiner and Amlogic chips.

On the ARM architecture kernels typically only support one SoC.
So there is a different kernel for each platform, and the one for the Raspberry is not old...
We do use AUFS rather than overlayfs yes, so that files created on the Raspberry are also accessible if you plug the same SD card into another platform that uses an older kernel..

I guess nested overlays, where you are mounting overlayfs on top of the AUFS we already setup are not supported by the kernel.
Reason we are using overlays is to let multiple Linux distribution share a single partition on one SD card.
What is your need to use them?

Kivutar, 2016-09-13:
We use OverlayFS to let users override the content of some directories contained in our SquashFS.

For example, joypad configurations are stored in the SquashFS so they can survive a factory reset. But users can override them thanks to OverlayFS.

Forusim, 2017-06-24:
Is there any way I can use Lakka with Berryboot?