libretro/Lakka #374: Blank screen after booting Lakka image

aquadisco, 2016-05-15:
Trying to install Lakka on a PC.  After copying the img file to a USB (using win32img), i boot up and see the Lakka splash screen.  Selecting install or live results in the same response -- blank screen, no video.  After a minute, the system reboots.

I've attached the output of lshw.  My system supports 64bits, but i tried both builds.  

My graphics card is definitely a few years old so i'd be surprised if it wasn't supported.  Been running different flavors of Ubuntu on this system for the last ~3 years without issues.

[mediacat-hw-details.txt](https://github.com/libretro/Lakka/files/265326/mediacat-hw-details.txt)

Kivutar, 2016-05-16:
Your graphics card is an ATI I guess?

Your blank screen is black or white?

aquadisco, 2016-05-19:
I suppose it's neither black nor white.  It seems as if the video output stops working and the monitor goes into standby.

R3d-Sky, 2016-05-22:
What monitor and port are you using? From the log I guess it's a VGA monitor, but I don't know for sure. Also, can we get a dmesg log?

aquadisco, 2016-05-22:
The board has a DVI and HDMI output.  I"m currently hooked up to the DVI, i've not tried the HDMI but i will do that.

Since the screen goes black (or more probably, the video output shuts off) right after the boot menu, i doubt the kernel has loaded and dmesg is even running.  Regardless, i don't know of any way to get log data off the system since there's no video and i doubt the network I/F is up.  After about 30secs, the system reboots.

Is there a way to setup a serial console (maybe over USB) that i could monitor?

Kivutar, 2016-05-26:
Is ssh running?

aquadisco, 2016-05-27:
Immediately after the splash screen loads and I select either live/install the video turns off in a few seconds after which it just reboots.  Doubt it even starts loading the kernel.

Tried rebuilding from source but change in outcome.

Kivutar, 2016-05-27:
We need to check if the kernel is running or not. You can try to remove the quiet argument in the cmdline. Testing if the box have an IP would help too

aquadisco, 2016-05-27:
Okay, i've made some progress.  My video card has two outputs -- DVI and HDMI.  Up to this point i've been using DVI and been seeing all the behaviors i've described above -- splashsreen, video shutoff, followed by reboot.

So, decided to give HDMI a try not expecting any different result.  Instead, when i tried it, it seemed to work this time.  System booted, video did not turn off, and the strangest part is it got to the Lakka menu and didn't reboot.  I also noticed that the boot time was very short.

So, no i am wondering if Lakka does get loaded when i have the DVI connected but maybe detects there's not monitor and reboots?  Is that possible?  When i had the box connected to HDMI i was in a slightly different configuration with no mouse, Ethernet or gamepad.

Kivutar, 2016-05-27:
> the strangest part is it got to the Lakka menu and didn't reboot. I also noticed that the boot time was very short.

Maybe because you already booted Lakka before, so the partition is already resized and you don't need to resize it more, so no reboot.

> So, no i am wondering if Lakka does get loaded when i have the DVI connected (now that i see how fast it boots) but maybe detects there's no monitor and reboots? Is that possible?

No, I doubt it.

I've never tested DVI so far. If you want to check if Lakka is launched with DVI, check my previous post about ip/ssh/quiet.

gouchi, 2016-05-27:
It seems there is some [option](https://wiki.archlinux.org/index.php/kernel_mode_setting#Forcing_modes_and_EDID) you can add in boot arguments that may help 
**e**: output forced to on
**D**: digital output forced to on (e.g. DVI-I connector) 

Something like 
video=DVI-I-1:e or video=DVI-I-1:D

gouchi, 2016-08-21:
Do you reproduce the issue with [latest nightly](http://mirror.lakka.tv/nightly/Generic.x86_64/Lakka-Generic.x86_64-devel-20160811142524-r21565-gc79651a.img.gz) ?

peterhoang, 2016-09-11:
I have a similar problem. I'm using Lakka in Live mode.  It boots perfectly on one of my older laptops, however, when I try to boot it with main desktop pc, I get no video source (blank screen) after the initial splash.  I use displayport.  I don't have a VGA/DVI adapter handy to test.

gouchi, 2016-09-11:
@peterhoang Try to change [the connector](http://www.lakka.tv/doc/PC) as I explained in the [previous post](https://github.com/libretro/Lakka/issues/374#issuecomment-222160409).

If it still doesn't work, try to change [video_monitor_index](https://github.com/libretro/Lakka/issues/465#issuecomment-240360421).