libretro/Lakka #375: Thumbnail updater not working

danbacastro, 2016-05-17:
Hi everybody, 

I did a Lakka update (through the updater within Lakka) and now, the thumbnail updater is giving an error message saying that failed in dowloading '. Index'. 

How do I solve it?

Rocklima2, 2016-05-18:
Thumbnail updater working perfect fo me, i use rpi 2 and last version of lakka using lakka updater

danbacastro, 2016-05-18:
Here, Lakka updater is working perfect, but the Thumbnail is not. I already change for other Lakka version and the error stills there.

cudencuden, 2016-05-18:
go http://www.lakka.tv/doc/FAQ/ to see how to get log.

Kivutar, 2016-05-18:
Try to purge your retroarch.cfg

    systemctl stop retroarch
    rm .config/retroarch/retroarch.cfg
    systemctl start retroarch

danbacastro, 2016-05-18:
As Kivutar suggested, I purged the retroarch.cfg, but the error continues.

The log output:
RetroArch [INFO] :: === Build =======================================RetroArch [INFO] :: [CPUID]: Vendor: AuthenticAMD
RetroArch [INFO] :: [CPUID]: Features: MMX MMXEXT SSE SSE2 SSE3 SSSE3
MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 Built: May  3 2016
RetroArch [INFO] :: Version: 1.3.4
RetroArch [INFO] :: Git: b2e76de
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: [CPUID]: Vendor: AuthenticAMD
RetroArch [INFO] :: [CPUID]: Features: MMX MMXEXT SSE SSE2 SSE3 SSSE3
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Config: loading config from: /storage/.config/retroarch/retroarch.cfg.
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29975.00 Hz.
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/5xbr-retro.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bead.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/blinky.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/pixellate.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/quilez.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/retro-v2.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/scanline.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/sharp-bilinear.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/stock.glsl"
RetroArch [INFO] :: Video @ 960x720
RetroArch [INFO] :: Starting threaded video driver ...
RetroArch [INFO] :: [DRM]: Found 3 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
RetroArch [INFO] :: [DRM]: Connector 0 has 8 modes.
RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 1 connected: yes
RetroArch [INFO] :: [DRM]: Connector 1 has 26 modes.
RetroArch [INFO] :: [DRM]: Connector 1 assigned to monitor index: #2.
RetroArch [INFO] :: [DRM]: Connector 2 connected: no
RetroArch [INFO] :: [DRM]: Connector 2 has 0 modes.
RetroArch [INFO] :: [DRM]: Mode 0: (1920x1080) 1920 x 1080, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 1: (1920x1080) 1920 x 1080, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 2: (1920x1080i) 1920 x 1080, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 3: (1920x1080i) 1920 x 1080, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 4: (1920x1080) 1920 x 1080, 50 Hz
RetroArch [INFO] :: [DRM]: Mode 5: (1920x1080i) 1920 x 1080, 50 Hz
RetroArch [INFO] :: [DRM]: Mode 6: (1920x1080) 1920 x 1080, 30 Hz
RetroArch [INFO] :: [DRM]: Mode 7: (1920x1080) 1920 x 1080, 30 Hz
RetroArch [INFO] :: [DRM]: Mode 8: (1920x1080) 1920 x 1080, 25 Hz
RetroArch [INFO] :: [DRM]: Mode 9: (1920x1080) 1920 x 1080, 24 Hz
RetroArch [INFO] :: [DRM]: Mode 10: (1920x1080) 1920 x 1080, 24 Hz
RetroArch [INFO] :: [DRM]: Mode 11: (1280x1024) 1280 x 1024, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 12: (1360x768) 1360 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 13: (1280x720) 1280 x 720, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 14: (1280x720) 1280 x 720, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 15: (1280x720) 1280 x 720, 50 Hz
RetroArch [INFO] :: [DRM]: Mode 16: (1024x768) 1024 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 17: (800x600) 800 x 600, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 18: (720x576) 720 x 576, 50 Hz
RetroArch [INFO] :: [DRM]: Mode 19: (720x576i) 720 x 576, 50 Hz
RetroArch [INFO] :: [DRM]: Mode 20: (720x480) 720 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 21: (720x480) 720 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 22: (720x480i) 720 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 23: (720x480i) 720 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 24: (640x480) 640 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 25: (640x480) 640 x 480, 60 Hz
RetroArch [INFO] :: Found GL context: kms
RetroArch [INFO] :: Detecting screen resolution 1920x1080.
RetroArch [INFO] :: [EGL]: EGL version: 1.4
RetroArch [INFO] :: [EGL]: Current context: 0xb43282d8.
RetroArch [INFO] :: [KMS]: New FB: 1920x1080 (stride: 7680).
RetroArch [INFO] :: [GL]: Vendor: X.Org, Renderer: Gallium 0.4 on AMD PALM (DRM 2.43.0, LLVM 3.6.0).
RetroArch [INFO] :: [GL]: Version: 3.0 Mesa 11.2.0-rc1.
[... 37 lines cut ...]
RetroArch [INFO] :: Autodetect: selected configuration: /tmp/joypads/udev/Sony-PlayStation3-DualShock3-Controller-Bluetooth.cfg
RetroArch [INFO] :: [udev]: Pad #0 (/dev/input/event6) supports force feedback.
RetroArch [INFO] :: [udev]: Pad #0 (/dev/input/event6) supports 16 force feedback effects.
RetroArch [INFO] :: Found joypad driver: "udev".
RetroArch [INFO] :: [CPUID]: Vendor: AuthenticAMD
RetroArch [INFO] :: [CPUID]: Features: MMX MMXEXT SSE SSE2 SSE3 SSSE3
RetroArch [INFO] :: ALSA: Using signed 16-bit format.
RetroArch [INFO] :: ALSA: Period size: 384 frames
RetroArch [INFO] :: ALSA: Buffer size: 1536 frames
RetroArch [INFO] :: [CPUID]: Vendor: AuthenticAMD
RetroArch [INFO] :: [CPUID]: Features: MMX MMXEXT SSE SSE2 SSE3 SSSE3
RetroArch [INFO] :: Found menu display driver: "menu_display_gl".
RetroArch [INFO] :: Using font rendering backend: freetype.
RetroArch [INFO] :: Using font rendering backend: freetype.
RetroArch [INFO] :: SRAM will not be saved.
RetroArch [INFO] :: null: [/storage/.config/retroarch/content_history.lpl].
RetroArch [ERROR] :: cannot push NULL or empty core info into the playlist.
[... cut ...]
RetroArch [INFO] :: [KMS]: New FB: 1920x1080 (stride: 7680).
RetroArch [INFO] :: [KMS]: Missed 1 VBlank(s) (Frame: 2, DRM frame: 16777281).
RetroArch [INFO] :: Saving config at path: "/storage/.config/retroarch/retroarch.cfg"
RetroArch [INFO] :: Saved new config to "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Threaded video stats: Frames pushed: 411, Frames dropped: 0.
RetroArch [INFO] :: Monitor FPS estimation is disabled for threaded video.
RetroArch [INFO] :: Monitor FPS estimation is disabled for threaded video.

fluffymadness, 2016-05-19:
I get also a failed downloading index when downloading thumbnails with latest stable build on odroid xu4.

twinaphex, 2016-05-19:
The thumbnail server is down, we exceeded the 3TB bandwidth cap.

We are trying to migrate to mudlord's server and it will require pushing a new version.

fluffymadness, 2016-05-19:
Ah ok. thx for the info.

danbacastro, 2016-05-20:
So, that is it. Thank you!

closed 2016-05-22 by danbacastro