libretro/Lakka #377: PCE-CD Games Not Loading

Unadan, 2016-05-21:
I'm running Lakka X86 64-bit on a Dell Inspiron laptop.  I have the "syscard3.pce" bios file placed in the system folder.  I'm trying to run the PCE iso and bin (unzipped) for the game "Seirei Senshi Spriggan" however when I load content it will not load - there is a short pause but the game does not load.  Any help would be appreciated.

emufrance, 2016-05-22:
Got the same behavior with this game in iso/cue format. Get the proper version (ccd+img+sub+cue), it works with current stable release (20160503153747).

gouchi, 2016-05-22:
I tested on the [latest nightly](http://mirror.lakka.tv/nightly/Generic.x86_64/Lakka-Generic.x86_64-devel-20160511215003-r21438-g88de891.img.gz) available and it works well with iso/cue.

Please check that your cue files point to the correct iso file and try to remove space in filename or change it to . or _

Unadan, 2016-05-22:
Wondering if I need to add ".iso" to the disc image filename to match the reference to the iso file in the cue file?

cudencuden, 2016-05-22:
I usually just load iso file n not the cue file using "load content and detect core"

Unadan, 2016-05-22:
When I try and do this the PCE-CD emulator doesn't come up as a choice - 3D0 and some Sega emulators do however.

gouchi, 2016-05-22:
@Unadan yes you need to add .iso in the cue file. For example,

`FILE "Seirei.Senshi.Spriggan.iso" BINARY`

@cudencuden As far as I know PC Engine/PCE-CD (Mednafen PCE FAST) can't know launch [iso file](https://github.com/libretro/libretro-super/blob/master/dist/info/mednafen_pce_fast_libretro.info#L3) directly ?

Kivutar, 2016-05-24:
Tested on the latest nightly build too, and it still works with a cue. Try to check your syscard3.pce

Unadan, 2016-05-24:
^Thanks.  Wondering if my ROM is no good.  I've tried multiple syscard3.pce files.  How large should this file be?

gouchi, 2016-05-24:
This [one](http://datomatic.nointro.org/index.php?page=show_record&s=12&n=0007) should be alright.

Kivutar, 2016-05-24:
MD5SUM ff1a674273fe3540ccef576376407d1d

Unadan, 2016-05-24:
Which of those two files do I rename to syscard3.pce?  Thanks!

gouchi, 2016-05-24:
Both should be fine the one I gave you is japanese region.

Unadan, 2016-05-24:
Sorry - so just rename [BIOS] Super CD-ROM System (Japan) (v3.0).pce.md5 to syscard3.pce?

gouchi, 2016-05-24:
You need to find the file with [md5sum](http://onlinemd5.com) ff1a674273fe3540ccef576376407d1d or 38179DF8F4AC870017DB21EBCBF53114

The file you are trying to rename (*.md5) is just the md5sum to control if the file you download is correct ;-)

Unadan, 2016-05-24:
Just checked the md5sum for the file I've been trying to use.  It is:  38179DF8F4AC870017DB21EBCBF53114 so looks like I'm not using the correct file.

gouchi, 2016-05-24:
I tried it is working we both file.

Unadan, 2016-05-24:
I'm at a loss.  At least on my system I cannot get this rom to run.

gouchi, 2016-05-24:
Can you get some log ? 
```
systemctl stop retroarch
retroarch --menu --verbose >> log.txt 2>&1
```

Unadan, 2016-05-24:
Trying to login with Putty to my Lakka Inspiron laptop - with Putty I never get prompted for a username/password.  I just get "connection refused" when I type in the ip address of my Lakka box (I suck at this stuff!!).

Kivutar, 2016-05-24:
Then go in the settings > services and enable ssh

Unadan, 2016-05-24:
Enabled under "Lakka Services" and then I rebooted the machine.  Reconnected it to my Windows 10 PC, ran Putty - still same problem - connection refused.  DHCP is enabled on my system.  Thanks for helping me with this.  Thinking at this point the rom from emuparadise must be faulty.

Unadan, 2016-05-24:
Gouchi - I renamed the files as you suggested - now the game loads!!!  Thanks a million~!!

closed 2016-05-25 by Kivutar