libretro/Lakka #378: Failed to compile generic target (output attached)

aquadisco, 2016-05-21:
Following instructions on teh Lakka website, tried building for source (4b62482ae7b869b808b35fd460c688eef1c03bb2).   See attachments for complete error output and lshw output.

(retroarch)[Lakka]: lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.04 LTS
Release:        16.04
Codename:       xenial

**x86_64-openelec-linux-gnu-gcc-4.9.2: internal compiler error: Killed (program lto1)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
lto-wrapper: /home/mjones/src/Lakka/build.Lakka-Generic.x86_64-devel/toolchain/bin/x86_64-openelec-linux-gnu-gcc-4.9.2 returned 4 exit status
/home/mjones/src/Lakka/build.Lakka-Generic.x86_64-devel/toolchain/lib/gcc/x86_64-openelec-linux-gnu/4.9.2/../../../../x86_64-openelec-linux-gnu/bin/ld.gold: fatal error: lto-wrapper failed
collect2: error: ld returned 1 exit status
Makefile:250: recipe for target 'mame2003_libretro.so' failed
make[1]: *** [mame2003_libretro.so] Error 1
make[1]: Leaving directory '/home/mjones/src/Lakka/build.Lakka-Generic.x86_64-devel/mame2003-4145a3b'
Makefile:12: recipe for target 'image' failed
make: *** [image] Error 2**

[mediacat-hw-details.txt](https://github.com/libretro/Lakka/files/275974/mediacat-hw-details.txt)

[nohup.zip](https://github.com/libretro/Lakka/files/275973/nohup.zip)

Kivutar, 2016-05-23:
Maybe your compiler went out of ram or something like this. Try to disable the mame2003 package in packages/lakka/Generic/package.mk

aquadisco, 2016-05-23:
I only have 2GB in this machine so maybe that's it.  I assume i just remove it from PKG_DEPENDS_TARGET?

I'll give that a try and let you know.

aquadisco, 2016-05-24:
Seem's pretty clear i'm running out of memory, even without mame2003.  Going to up the memory and try again in a few days.

Kivutar, 2016-05-24:
Try to use a big swap

aquadisco, 2016-05-25:
a bigger swap space, doh!  It's working now. Thanks for the support

Kivutar, 2016-05-26:
No problem :)

closed 2016-05-26 by Kivutar