libretro/Lakka #380: Problem in the setup menu and zelda game majora's mask

AndersonFirmino, 2016-05-22:
Hello world developers.
First I would like to congratulate them for the emulator.

And also report some bugs in my experience with him.

In my computer I use Linux Mint in its latest version.
And the Retro-arch emulator in the last version 1.3.4

The bug I'm finding is as follows.

I set in the emulator, the directories where are the roms
I changed the wallpaper and set up the controls

I use USB control, USB Twin controller (PS2 gamepad).

And when I open the Zelda, in the middle of the game when I open the menu I can not access the menu items. The game simply skips the items and go to the next menu screen.

I return the settings to see if I can configure the controls, but the emulator crashes and there have to close and restart it.

And the problem persisted, unless I delete the file retroarch.cfg making me have to reconfigure everything again;
I can set it only once, then the problem returns.

thanks for any help. :)

gouchi, 2016-05-25:
You should post your issue on [RetroArch](https://github.com/libretro/RetroArch/issues) project or [Mupen64plus](https://github.com/libretro/mupen64plus-libretro) directly as you don't use Lakka.

Kivutar, 2016-05-26:
Ah yes, I close this one.

closed 2016-05-26 by Kivutar

AndersonFirmino, 2016-06-01:
ops, sorry. And thanks. :)