libretro/Lakka #381: Using lakka toolchain to compile single core

lollo78, 2016-05-22:
@Kivutar 

Hi dear, noob request :D

I noticed that when I compile Lakka, system downloads the modules from source.lakka.tv.
But the packages stored on this repository are not upgraded to the latest origin-master git repo.
For example: mame2003-4145a3b.tar.xz is created on 15-Feb-2016 02:16, but on https://github.com/libretro/mame2003-libretro the last changes are dated 2 days ago. Same for mupen64plus, pcsx_rearmed and so on.

My question is: how can I use your nice and simple toolchain, to compile directly the latest github cores (bypassing your tar.xz)?

To compile a single core I'm using this (example for mupen64plus): 
sudo DISTRO=Lakka PROJECT=RPi2 ARCH=arm scripts/build mupen64plus

Another info: if I don't use sudo, I can't compile some cores (psx_rearmed, mupen64plus...). Sudo is not required for other cores like 4do, scummvm...
It's normal?

Thx a lot

Kivutar, 2016-05-22:
Hello @lollo78 

You can't do that in Lakka, but you can do that by using the latest OpenELEC
They introduced a new syntax in their package.mk to allow that

You can clone their tree, and copy the libretro packages from Lakka. And then adapt them to your need.

Sudo is never required to build a package, so it's not normal.

lollo78, 2016-05-22:
Hi kivutar, thx for reply.
I tried all on Ubuntu 16.04. I don't know why, but without sudo I can't compile some cores. It stuck on tar.xz download from source.lakka.tv (Try again later!).

OK for OpenELEC (you know if this new syntax are included also in LibreELEC?). I'll clone the tree and I'll make some test. 
Only a clarification: what do you mean for "copy the libretro packages from Lakka"?
Thx again

Kivutar, 2016-05-22:
I don't think that LibreELEC have it yet.

By copy I mean

    cp -r Lakka/packages/libretro OpenELEC/packages/

lollo78, 2016-05-22:
Thx, I'll try this solution.
:*

lollo78, 2016-05-22:
Ehm! another noob question :)

I cloned OpenELEC tree:
`git clone https://github.com/OpenELEC/OpenELEC.tv
`
I copied libretro package into Openelec Package:
`cp -r Lakka/packages/libretro/ OpenELEC.tv/packages/
`
I tried to compile mame2003:
`PROJECT=RPi2 ARCH=arm scripts/build mame2003
`
but of course your mame2003/package.mk  point to your source. I have to adapt it, like you said.
But how? Could you point me in the right direction at least for one example?

Your original package.mk
```
PKG_NAME="mame2003"
PKG_VERSION="4145a3b"
PKG_REV="1"
PKG_ARCH="any"
PKG_LICENSE="MAME"
PKG_SITE="https://github.com/libretro/mame2003-libretro.git"
PKG_URL="$LAKKA_MIRROR/$PKG_NAME-$PKG_VERSION.tar.xz"
PKG_DEPENDS_TARGET="toolchain"
PKG_PRIORITY="optional"
PKG_SECTION="libretro"
PKG_SHORTDESC="MAME - Multiple Arcade Machine Emulator"
PKG_LONGDESC="MAME - Multiple Arcade Machine Emulator"

PKG_IS_ADDON="no"
PKG_AUTORECONF="no"

make_target() {
  make ARCH="" CC="$CC" NATIVE_CC="$CC" LD="$CC" -j 1
}

makeinstall_target() {
  mkdir -p $INSTALL/usr/lib/libretro
  cp mame2003_libretro.so $INSTALL/usr/lib/libretro/
}
```

I modified it in:
```
PKG_NAME="mame2003"
PKG_VERSION=""
PKG_REV=""
PKG_ARCH="any"
PKG_LICENSE="MAME"
PKG_SITE="https://github.com/libretro/mame2003-libretro.git"
PKG_URL="https://github.com/libretro/mame2003-libretro/archive/master.zip"
PKG_DEPENDS_TARGET="toolchain"
PKG_PRIORITY="optional"
PKG_SECTION="libretro"
PKG_SHORTDESC="MAME - Multiple Arcade Machine Emulator"
PKG_LONGDESC="MAME - Multiple Arcade Machine Emulator"

PKG_IS_ADDON="no"
PKG_AUTORECONF="no"

make_target() {
  make ARCH="" CC="$CC" NATIVE_CC="$CC" LD="$CC" -j 1
}

makeinstall_target() {
  mkdir -p $INSTALL/usr/lib/libretro
  cp mame2003_libretro.so $INSTALL/usr/lib/libretro/
}
```
Now if I compile with
`PROJECT=RPi2 ARCH=arm scripts/build mame2003
`
it download master.zip inside OpenELEC.tv/sources/mame2003 and create build.OpenELEC-RPi2.arm-8.0-devel folder. Inside of it there is an empty "mame2003-" subfolder. No trace of the compiled .so file.

What I'm wrong? 
I know that this is not a Lakka issue, but I really appreciate your help. 
Thx a lot

R3d-Sky, 2016-05-23:
I believe it is because the build-system expects a .tar based archive, while you download a .zip archive. Try changing master.zip with master.tar.gz and maybe it'll work.

Kivutar, 2016-05-23:
https://github.com/OpenELEC/OpenELEC.tv/blob/master/packages/linux-firmware/dvb-firmware/package.mk#L25-L26

lollo78, 2016-05-23:
Hi kivutar, thx a lot. I modified the package.mk in

```
PKG_NAME="mame2003"
PKG_VERSION="libretro"
PKG_REV=""
PKG_ARCH="any"
PKG_LICENSE="MAME"
PKG_SITE="https://github.com/libretro/mame2003-libretro"
PKG_GIT_URL="https://github.com/libretro/mame2003-libretro.git"
PKG_GIT_BRANCH="master"
PKG_DEPENDS_TARGET="toolchain"
PKG_PRIORITY="optional"
PKG_SECTION="libretro"
PKG_SHORTDESC="MAME - Multiple Arcade Machine Emulator"
PKG_LONGDESC="MAME - Multiple Arcade Machine Emulator"

PKG_IS_ADDON="no"
PKG_AUTORECONF="no"

make_target() {
  make ARCH="" CC="$CC" NATIVE_CC="$CC" LD="$CC" -j 1
}

makeinstall_target() {
  mkdir -p $INSTALL/usr/lib/libretro
  cp mame2003_libretro.so $INSTALL/usr/lib/libretro/
}
```
Now seems that I can download all the necessary package and try to start compiling.
But, I think, that I have issues with the bottom section of the code:make_target and makeinstall_target.

With 
```
make_target() {
  make ARCH="" CC="$CC" NATIVE_CC="$CC" LD="$CC" -j 1
}
```
make stop with this error
```
fatal: Not a valid object name
  UNPACK   mame2003
  BUILD    mame2003 (target)
make: *** No targets specified and no makefile found.  Stop.

```
If I change it with make_target with
```
make_target() {
  : # nothing todo
}

```
seems that it stop to the makeinstall_target  
```
  UNPACK   mame2003
  BUILD    mame2003 (target)
cp: cannot stat 'mame2003_libretro.so': No such file or directory

```
There are no trace for mame2003_libretro.so.
Any other suggestion? :*

Kivutar, 2016-05-23:
Maybe recent versions of mame2003 have the makefile in another place?

lollo78, 2016-05-23:
Thanks to your reply, I investigated better. 
The "make path" of mame2003 is not changed. 

I noticed that If I launch the command **PROJECT=RPi2 ARCH=arm scripts/build mame2003**, in **OpenELEC.tv/source/mame2003** the system create first a **mame2003-libretro.git** folder and after (download completed) this folder disappears and remains a _32 byte_ **mame2003-libretro.tar.xz** file. 
The issue is that **OpenELEC.tv/build.OpenELEC-RPi2.arm-8.0-devel/mame2003-libretro** is always empty. This cause my issue of "make_target not found".
How to fix that?

Continuing the test... 
I manually downloaded **mame2003-libretro-master.zip** from https://github.com/libretro/mame2003-libretro/archive/master.zip, unzipped it in **OpenELEC.tv/build.OpenELEC-RPi2.arm-8.0-devel/mame2003-libretro** and launch again **PROJECT=RPi2 ARCH=arm scripts/build mame2003**.
Now compiling starts, but it stuck on

```
In file included from src/inptport.h:4:0,
                 from src/osdepend.h:5,
                 from src/unzip.h:5,
                 from src/mame_unzip.c:1:
src/memory.h: In function 'cpu_readop16':
src/memory.h:911:35: warning: cast increases required alignment of target type [-Wcast-align]
 #define cpu_readop16_unsafe(A)  (*(data16_t *)&OP_ROM[(A) & mem_amask])
                                   ^
src/memory.h:920:106: note: in expansion of macro 'cpu_readop16_unsafe'
 INLINE data16_t cpu_readop16(offs_t A)  { if (address_is_unsafe(A)) { activecpu_set_op_base(A); } return cpu_readop16_unsafe(A); }
                                                                                                          ^~~~~~~~~~~~~~~~~~~
src/memory.h: In function 'cpu_readop32':
src/memory.h:912:35: warning: cast increases required alignment of target type [-Wcast-align]
 #define cpu_readop32_unsafe(A)  (*(data32_t *)&OP_ROM[(A) & mem_amask])
                                   ^
src/memory.h:921:106: note: in expansion of macro 'cpu_readop32_unsafe'
 INLINE data32_t cpu_readop32(offs_t A)  { if (address_is_unsafe(A)) { activecpu_set_op_base(A); } return cpu_readop32_unsafe(A); }
                                                                                                          ^~~~~~~~~~~~~~~~~~~
src/memory.h: In function 'cpu_readop_arg16':
src/memory.h:914:38: warning: cast increases required alignment of target type [-Wcast-align]
 #define cpu_readop_arg16_unsafe(A) (*(data16_t *)&OP_RAM[(A) & mem_amask])
                                      ^
src/memory.h:923:109: note: in expansion of macro 'cpu_readop_arg16_unsafe'
 INLINE data16_t cpu_readop_arg16(offs_t A) { if (address_is_unsafe(A)) { activecpu_set_op_base(A); } return cpu_readop_arg16_unsafe(A); }
                                                                                                             ^~~~~~~~~~~~~~~~~~~~~~~
src/memory.h: In function 'cpu_readop_arg32':
src/memory.h:915:38: warning: cast increases required alignment of target type [-Wcast-align]
 #define cpu_readop_arg32_unsafe(A) (*(data32_t *)&OP_RAM[(A) & mem_amask])
                                      ^
src/memory.h:924:109: note: in expansion of macro 'cpu_readop_arg32_unsafe'
 INLINE data32_t cpu_readop_arg32(offs_t A) { if (address_is_unsafe(A)) { activecpu_set_op_base(A); } return cpu_readop_arg32_unsafe(A); }
                                                                                                             ^~~~~~~~~~~~~~~~~~~~~~~
src/mame_unzip.c: At top level:
src/mame_unzip.c:8:18: fatal error: zlib.h: No such file or directory

                  ^
compilation terminated.
Makefile:307: recipe for target 'src/mame_unzip.o' failed
make: *** [src/mame_unzip.o] Error 1

```
:(

escalade, 2016-05-25:
The error message is quite clear, you are missing zlib.h. If you do a "make image" before compiling the core, you should have everything you need in place.

The reason your github download was empty is because you didn't set a correct PKG_VERSION, which should be set to the 7-digit commit that you want. For example "77c470d".

There's also the "LibreELEC way", by using the github archive function like this: https://github.com/escalade/LibreELEC.tv/blob/emu/packages/escalade/emulation/mame2003-libretro/package.mk

I prefer this method because the download will be smaller. The extraction directory will some times differ from the expected one and will have to be moved in place like here: https://github.com/escalade/LibreELEC.tv/blob/emu/packages/escalade/emulation/craft-libretro/package.mk

Kivutar, 2016-05-25:
Nice, I wasn't aware of that. I will use that now I think. And also borrow some of your packages.

escalade, 2016-05-25:
Go ahead, I borrowed yours :)

lollo78, 2016-05-25:
Lovely escalade!
I just cloned your LibreELEC.tv git repo and now I'm doing a **PROJECT=RPi2 ARCH=arm make image**.
Sorry for the noob question, but (please) could you tell me the right build command to compile a single libretro core (for example mame2003) with your release? Before to launch a full "make image" I tried with **PROJECT=RPi2 ARCH=arm scripts/build mame2003** but in this way it not found mame2003 package.mk.

Another question. I would like to use this system to compile single libretro cores to test and integrate them in my homebrew Kodi addon. 
I installed LibreELEC 7.0.1 on my Rpi2 and RPi3 for a daily use, and for this reason I would like to stay on stable release. I see that your LibreELEC fork is based on LE 8.0. If I compile the core (or retroarch) directly from your git I have to expect some dependencies issues?

Thanks a lot

escalade, 2016-05-25:
Your command is correct but my package is named mame2003-libretro, look in packages/escalade/emulation.

LE7 and LE8 is more or less compatible, but there are bound to be a few differences so I would indeed expect some dependency issues.

lollo78, 2016-05-25:
> mame2003-libretro, look in packages/escalade/emulation.

Thanks a lot for pointing me in the right direction! Now I'm trying with PROJECT=RPi2 ARCH=arm scripts/build mame2003-libretro and it's started like expected. I'm waiting for it to finish ;)

But I don't now why "PROJECT=RPi2 ARCH=arm make image" command ended with issue:
```
In file included from ../include/xfs.h:37:0,
                 from crc32.c:37:
../include/xfs/linux.h:20:23: fatal error: uuid/uuid.h: No such file or directory
compilation terminated.
Makefile:124: recipe for target 'crc32selftest' failed
gmake[3]: *** [crc32selftest] Error 1
gmake[3]: *** Waiting for unfinished jobs....
include/buildrules:35: recipe for target 'libxfs' failed
gmake[2]: *** [libxfs] Error 2
Makefile:70: recipe for target 'default' failed
make[1]: *** [default] Error 2
make[1]: Leaving directory '/home/lollo78/LibreELEC.tv/build.LibreELEC-RPi2.arm-8.0-devel/xfsprogs-dev-53cefc4'
Makefile:12: set di istruzioni per l'obiettivo "image" non riuscito
make: *** [image] Errore 1

```
I'm using Ubuntu 16.04
:sob: 

> 
LE7 and LE8 is more or less compatible, but there are bound to be a few differences so I would indeed expect some dependency issues.

OK, I continue to test with your build, also because I have to understand how to compile it and how to solve my compilation issues (but I do not know which way to turn). If there will be dependencies issues using the core compiled from your build, I'll try to clone official LibreELEC 7.0.1 git and copy/paste your package/escalade/emulation.

escalade, 2016-05-25:
Either install uuid-dev or set XFS_SUPPORT=no in distributions/LibreELEC/options. It's supposed to find uuid.h in the toolchain, but my package isn't handling it correctly will investigate.

lollo78, 2016-05-25:
@Kivutar 
Sorry if I take advantage of your Issues section. If you want, I move the discussion on the escalade git. But I think that, although if this is not a problem, it's an interesting discussion for the noobs like me. :D 

@escalade 
OK, I installed uuid-dev. To retry the compilation I have to delete first all the build.LibreELEC-RPi2.arm-8.0-devel?
Please, see again my post above. I edited it adding a retroarch compilation issue exactly when you answered (or uuid.dev could solve also this?). :*

escalade, 2016-05-25:
Or better yet, use my thread on the LE forums. You can just do "make image" again. As for your error, it's not related to Retroarch but the Python package (https://github.com/LibreELEC/LibreELEC.tv/commit/53eca15b39207c09c876c222211255b87dea9ff6#commitcomment-17621210). If you're going to be compiling my tree, I suggest you set up a build environment with Arch Linux which is what I'm using.

lollo78, 2016-05-25:
If possible, I would remain on Ubuntu. :relaxed: 
With uuid-dev "PROJECT=RPi2 ARCH=arm make image" quit again, but this time with the same error of retroarch ("Assumed value of MB_LEN_MAX wrong" in function wctomb):

```
In file included from /usr/include/stdlib.h:958:0,
                 from libcurses/fileio.c:41:
/usr/include/x86_64-linux-gnu/bits/stdlib.h: In function 'wctomb':
/usr/include/x86_64-linux-gnu/bits/stdlib.h:90:3: error: #error "Assumed value of MB_LEN_MAX wrong"
 # error "Assumed value of MB_LEN_MAX wrong"
   ^
/home/lollo78/LibreELEC.tv/build.LibreELEC-RPi2.arm-8.0-devel/toolchain/bin/host-gcc  -I. -I./libterminfo -DHAVE_WCHAR -I./libcurses -O2 -Wall -pipe -I/home/lollo78/LibreELEC.tv/build.LibreELEC-RPi2.arm-8.0-devel/toolchain/include -Wno-format-security -fPIC -c -o libcurses/instr.o libcurses/instr.c
GNUmakefile:399: recipe for target 'libcurses/fileio.o' failed
make[1]: *** [libcurses/fileio.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/lollo78/LibreELEC.tv/build.LibreELEC-RPi2.arm-8.0-devel/netbsd-curses-47b256a/.x86_64-linux-gnu'
Makefile:12: set di istruzioni per l'obiettivo "image" non riuscito
make: *** [image] Errore 2

```
To solve it I have to install another Python version?
Mine is:
```
lollo78@lollo78-P55A-UD4:~$ python --version
Python 2.7.11+
```
It's not clear for me the advice in LibreELEC/LibreELEC.tv@53eca15#commitcomment-17621210.
You speek about "getcwd" error. It's related to my issue? In my build.LibreELEC-RPi2.arm-8.0-devel there are no Kodi folder to delete.

escalade, 2016-05-25:
I was referring to the PR, not specifically the last comment. JStyle21 has the same error as you on Gentoo. Like I said, if you want to build my tree use Arch, or be prepared to tackle these issues yourself. In any case, this is not the place to discuss this as we are spamming others subscribed to this issue, please close it if your answer has been answered.

lollo78, 2016-05-25:
You are right. It is not the exact location.
I'll try to install Arch and I'll come to spam on your git.
Thx a lot for you precious time and all the help. :*

I leave this thread open. If I understand correctly, @kivutar have intension to impement this functionality also in Lakka. So, I'll try this again in Lakka, when he is ready. If not, please Kivutar close it.
Thx to all!

lollo78, 2016-05-26:
Just to inform you that using official LibreELEC git repository (https://wiki.libreelec.tv/index.php?title=Compile) and "package/escalade" folder grabbed from "https://github.com/escalade/LibreELEC.tv/" I can compile all the cores on Ubuntu 16.04 LTS without issues.
Changing PKG_VERSION= argument, inside the package.mk file, with the desired commits, I can test all that I want. Finally I have a simple to use environment, to crosscompile libretro cores. It's wonderfull. I'm crying of joy...

All the compiled cores works fine, only retroarch doesn't work.
The compiled binary has this dependencie issues on my system (official LibreELEC 7.0.1):
```
OpenELEC-Rpi3:~/.kodi/addons/emulator.lakka/bin # ldd retroarch | grep no
	libavcodec.so.57 => not found
	libavformat.so.57 => not found
	libavutil.so.55 => not found
	libswscale.so.4 => not found
	libswresample.so.2 => not found
```
Maybe I have to "play a little" with the if cicle inside 
https://github.com/escalade/LibreELEC.tv/blob/emu/packages/escalade/emulation/retroarch/package.mk
@Kivutar: there's also a way to modify this file to compile the retroarch version personalized for  Lakka?

Kivutar, 2016-05-26:
@lollo78 lakka uses the regular retroarch, no personalization, except maybe one or two small patches

lollo78, 2016-05-26:
Thanks for the clarification. I always thought about more complex personalization. 

Are you thinking to implement the escalade package in Lakka?
They are so cool! And, if you want, you could add also the other libretro cores not included by escalade (tyrquake, stella, genesis...) to create a really simple and completed environment for beta testers :)

Kivutar, 2016-05-27:
Yeah I'm going to switch to URLs of this form https://github.com/escalade/LibreELEC.tv/blob/emu/packages/escalade/emulation/craft-libretro/package.mk#L27

escalade, 2016-05-27:
@lollo78 

Thanks, but you should give credit where it's due. My packages are mostly ripped straight from Lakka, only updated the version. I just added Emulationstation, Amiga emulators and some missing cores.

As for your library issue, it seems it's been compiled against your system ffmpeg instead of the ones in your toolchain. I ran into the same issue, which is why I added "--host=$TARGET_NAME" to the configure arguments, which seems to do the trick here.

lollo78, 2016-05-27:
Thanks to you, Thanks to Kivutar, thanks to Twinaphex, tanks and credit to all the guys who works so hardly to create this magic!!! :*

About ffmpeg (thanks to suggest me that this could be the issue), I do some research. 
The libraries above are related to ffmpeg 3.x, but LibreELEC 7.0.1 (installed on my RPix) use ffmpeg version 2.8.6:
```
OpenELEC-Rpi3:~/.kodi/addons/emulator.lakka/bin # ffmpeg -version
ffmpeg version 2.8.6 Copyright (c) 2000-2016 the FFmpeg developers
built with gcc 5.3.0 (GCC)

``` 

Maybe, when I done 
`git clone https://github.com/LibreELEC/LibreELEC.tv.git`
it cloned the new LibreELEC 8.0 beta, instead of stable branch 7.0.1.
Than, all that I compiled is based on 8.0 beta. 

If this is possible, please, there is a way to set my actual LibreELEC git folder on 7.0.1 without download and rebuild again all the files?
If I have to delete it and restart again, please could you tell me how to clone from git exactly the stable branch 7.0.x?

Thanks a lot, really!

closed 2016-06-17 by Kivutar