libretro/Lakka #382: FB Alpha Scanning missing files and not working

kbazzor, 2016-05-23:
Im running the PC version, fresh install, found all the rom's recommended in the documentation, get to FBA and the scan seems to be missing a bunch of files, the ones i was specifically after where simpsons and teenage mutant ninja turtles which both didnt show up

I can't seem to get the core it self to run either, none of the roms that do show up work loads the lakka image, screen 'phizzes' out a bit and nothing happens (have put the neogeo.zip in the rom directory and no luck either), could it be because in the playlist is has detect in the spot where the core should be?

Thanks

cudencuden, 2016-05-23:
which fba rom set do you have?

kbazzor, 2016-05-23:
FBA_v0.2.97.38 the one that was reccomended

Kivutar, 2016-05-24:
I tried again today on an RPi2 and it was working.

Can you try to get an error log please? And can you tell us more about your hardware, especially the graphics card.

kbazzor, 2016-05-25:
**Its an intel pentium quad core NUC (N3700) using iGPU (Intel HD Graphics)**

**Intel CPU Specs**
http://ark.intel.com/products/87261/Intel-Pentium-Processor-N3700-2M-Cache-up-to-2_40-GHz

**Intel NUC Specs**
http://www.intel.com/content/www/us/en/nuc/nuc-kit-nuc5ppyh.html

To do some troubleshooting of my own i tried to source alternate single roms to see if the original package i had was the issue but this was not the case

I can try to load it up on my raspi 2 as well with the same rom set if needed to double check though

kbazzor, 2016-05-25:
May 25 10:26:01 Lakka kernel: hid-generic 0003:AA55:0101.0002: input,hidraw1: USB HID v1.10 Keyboard [XGaming X-Arcade] on usb-0000:00:14.0-2/input1
May 25 10:25:57 Lakka kernel: BRK [0x02bec000, 0x02becfff] PGTABLE
May 25 10:25:57 Lakka lsb_release[182]: Lakka (unofficial) - Version: devel-20160503153747-r21429-g9e4f0ee
-- Logs begin at Wed 2016-05-25 10:25:57 UTC, end at Wed 2016-05-25 10:25:58 UTC. --
May 25 10:25:57 Lakka systemd-journald[213]: Runtime journal (/run/log/journal/) is currently using 512.0K.
May 25 10:26:01 Lakka kernel: hid-generic 0003:AA55:0101.0002: input,hidraw1: USB HID v1.10 Keyboard [XGaming X-Arcade] on usb-0000:00:14.0-2/input1
May 25 10:26:01 Lakka kernel: usb 1-3: new full-speed USB device number 3 using xhci_hcd
May 25 10:26:01 Lakka kernel: logitech-djreceiver 0003:046D:C52B.0005: hiddev0,hidraw2: USB HID v1.11 Device [Logitech USB Receiver] on usb-0000:00:14.0-3/input2
May 25 10:26:02 Lakka kernel: input: Logitech K830 as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.2/0003:046D:C52B.0005/0003:046D:4032.0006/input/input16
May 25 10:26:02 Lakka kernel: logitech-hidpp-device 0003:046D:4032.0006: input,hidraw3: USB HID v1.11 Keyboard [Logitech K830] on usb-0000:00:14.0-3:1
May 25 10:26:02 Lakka avahi-daemon[310]: Joining mDNS multicast group on interface eth0.IPv4 with address 192.168.0.56.
May 25 10:26:02 Lakka avahi-daemon[310]: New relevant interface eth0.IPv4 for mDNS.
May 25 10:26:02 Lakka avahi-daemon[310]: Registering new address record for 192.168.0.56 on eth0.IPv4.
May 25 10:26:02 Lakka connmand[307]: eth0 {add} address 192.168.0.56/24 label eth0 family 2
May 25 10:26:02 Lakka connmand[307]: eth0 {add} route 192.168.0.0 gw 0.0.0.0 scope 253 <LINK>
May 25 10:26:02 Lakka connmand[307]: eth0 {add} route 192.168.0.1 gw 0.0.0.0 scope 253 <LINK>
May 25 10:26:02 Lakka connmand[307]: eth0 {add} route 198.142.0.51 gw 192.168.0.1 scope 0 <UNIVERSE>
May 25 10:26:02 Lakka connmand[307]: eth0 {add} route 211.29.132.12 gw 192.168.0.1 scope 0 <UNIVERSE>
May 25 10:26:02 Lakka kernel: usb 1-5: new high-speed USB device number 4 using xhci_hcd
May 25 10:26:02 Lakka connmand[307]: eth0 {add} route 198.142.235.14 gw 192.168.0.1 scope 0 <UNIVERSE>
May 25 10:26:02 Lakka connmand[307]: eth0 {add} route 0.0.0.0 gw 192.168.0.1 scope 0 <UNIVERSE>
May 25 10:26:02 Lakka systemd[1]: retroarch.service: Service hold-off time over, scheduling restart.
May 25 10:26:02 Lakka connmand[307]: eth0 {del} route fe80:: gw :: scope 0 <UNIVERSE>
May 25 10:26:02 Lakka systemd[1]: Stopped Retroarch.
May 25 10:26:02 Lakka connmand[307]: eth0 {del} route ff00:: gw :: scope 0 <UNIVERSE>
May 25 10:26:02 Lakka systemd[1]: Started Retroarch.
May 25 10:26:02 Lakka kernel: hub 1-5:1.0: USB hub found
May 25 10:26:02 Lakka kernel: hub 1-5:1.0: 4 ports detected
May 25 10:26:02 Lakka connmand[307]: ntp: time slew -35996.482186 s
May 25 00:26:05 Lakka systemd[1]: Time has been changed
May 25 00:26:06 Lakka kernel: usb 1-5.1: new full-speed USB device number 5 using xhci_hcd
May 25 00:26:06 Lakka kernel: Bluetooth: Core ver 2.21
May 25 00:26:06 Lakka kernel: NET: Registered protocol family 31
May 25 00:26:06 Lakka kernel: Bluetooth: HCI device and connection manager initialized
May 25 00:26:06 Lakka kernel: Bluetooth: HCI socket layer initialized
May 25 00:26:06 Lakka kernel: Bluetooth: L2CAP socket layer initialized
May 25 00:26:06 Lakka kernel: Bluetooth: SCO socket layer initialized
May 25 00:26:06 Lakka kernel: usbcore: registered new interface driver btusb
May 25 00:26:06 Lakka kernel: Bluetooth: hci0: read Intel version: 370810011003110e20
May 25 00:26:06 Lakka kernel: Bluetooth: hci0: Intel device is already patched. patch num: 20
May 25 00:26:06 Lakka systemd[1]: Reached target Bluetooth.
May 25 00:26:07 Lakka kernel: retroarch[462]: segfault at 0 ip           (null) sp 00007f1087911cf8 error 14 in retroarch[400000+195000]
May 25 00:26:07 Lakka systemd[1]: [[0;1;39mretroarch.service: Main process exited, code=killed, status=11/SEGV[[0m
May 25 00:26:07 Lakka systemd[1]: [[0;1;39mretroarch.service: Unit entered failed state.[[0m
May 25 00:26:07 Lakka systemd[1]: [[0;1;39mretroarch.service: Failed with result 'signal'.[[0m
May 25 00:26:10 Lakka systemd[1]: retroarch.service: Service hold-off time over, scheduling restart.
May 25 00:26:10 Lakka systemd[1]: Stopped Retroarch.
May 25 00:26:10 Lakka systemd[1]: Started Retroarch.
May 25 00:26:24 Lakka kernel: [[0;1;39mrandom: nonblocking pool is initialized[[0m
May 25 00:26:24 Lakka sshd[341]: Server listening on 0.0.0.0 port 22.
May 25 00:26:24 Lakka sshd[341]: Server listening on :: port 22.
May 25 00:26:52 Lakka systemd[1]: retroarch.service: Service hold-off time over, scheduling restart.
May 25 00:26:52 Lakka systemd[1]: Stopped Retroarch.
May 25 00:26:52 Lakka systemd[1]: Started Retroarch.
May 25 00:27:04 Lakka crond[259]: [[0;1;31mtime disparity of -598 minutes detected[[0m
May 25 00:28:26 Lakka sshd[494]: Accepted password for root from 192.168.0.52 port 28891 ssh2
May 25 00:29:17 Lakka systemd[1]: retroarch.service: Service hold-off time over, scheduling restart.
May 25 00:29:17 Lakka systemd[1]: Stopped Retroarch.
May 25 00:29:17 Lakka systemd[1]: Started Retroarch.
May 25 00:30:35 Lakka systemd[1]: retroarch.service: Service hold-off time over, scheduling restart.
May 25 00:30:35 Lakka systemd[1]: Stopped Retroarch.
May 25 00:30:35 Lakka systemd[1]: Started Retroarch.
May 25 00:30:46 Lakka systemd[1]: retroarch.service: Service hold-off time over, scheduling restart.
May 25 00:30:46 Lakka systemd[1]: Stopped Retroarch.
May 25 00:30:46 Lakka systemd[1]: Started Retroarch.

Is there another debug mode i can use  to give more details? Anything after the ssh login is me trying to launch roms multiple times

Kivutar, 2016-05-25:
Ok, you can do this

    systemctl stop retroarch

and launch retroarch manually

    retroarch -v

and launch a rom. You will get a log

kbazzor, 2016-05-25:
RetroArch [INFO] :: === Build =======================================RetroArch [INFO] :: [CPUID]: Vendor: GenuineIntel
RetroArch [INFO] :: [CPUID]: Features: MMX MMXEXT SSE SSE2 SSE3 SSSE3 SSE4 SSE4.2 AES
MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 SSE4 SSE4.2 AES Built: May  3 2016
RetroArch [INFO] :: Version: 1.3.4
RetroArch [INFO] :: Git: b2e76de
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: [CPUID]: Vendor: GenuineIntel
RetroArch [INFO] :: [CPUID]: Features: MMX MMXEXT SSE SSE2 SSE3 SSSE3 SSE4 SSE4.2 AES
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Config: loading config from: /storage/.config/retroarch/retroarch.cfg.
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29975.00 Hz.
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/5xbr-retro.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bead.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/blinky.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/pixellate.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/quilez.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/retro-v2.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/scanline.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/sharp-bilinear.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/stock.glsl"
RetroArch [INFO] :: Video @ 960x720
RetroArch [INFO] :: Starting threaded video driver ...
RetroArch [INFO] :: [DRM]: Found 4 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: no
RetroArch [INFO] :: [DRM]: Connector 0 has 0 modes.
RetroArch [INFO] :: [DRM]: Connector 1 connected: no
RetroArch [INFO] :: [DRM]: Connector 1 has 0 modes.
RetroArch [INFO] :: [DRM]: Connector 2 connected: yes
RetroArch [INFO] :: [DRM]: Connector 2 has 21 modes.
RetroArch [INFO] :: [DRM]: Connector 2 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 3 connected: no
RetroArch [INFO] :: [DRM]: Connector 3 has 0 modes.
RetroArch [INFO] :: [DRM]: Mode 0: (1440x900) 1440 x 900, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 1: (1400x1050) 1400 x 1050, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 2: (1400x1050) 1400 x 1050, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 3: (1280x1024) 1280 x 1024, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 4: (1280x1024) 1280 x 1024, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 5: (1440x900) 1440 x 900, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 6: (1280x960) 1280 x 960, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 7: (1152x864) 1152 x 864, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 8: (1024x768) 1024 x 768, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 9: (1024x768) 1024 x 768, 70 Hz
RetroArch [INFO] :: [DRM]: Mode 10: (1024x768) 1024 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 11: (832x624) 832 x 624, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 12: (800x600) 800 x 600, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 13: (800x600) 800 x 600, 72 Hz
RetroArch [INFO] :: [DRM]: Mode 14: (800x600) 800 x 600, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 15: (800x600) 800 x 600, 56 Hz
RetroArch [INFO] :: [DRM]: Mode 16: (640x480) 640 x 480, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 17: (640x480) 640 x 480, 73 Hz
RetroArch [INFO] :: [DRM]: Mode 18: (640x480) 640 x 480, 67 Hz
RetroArch [INFO] :: [DRM]: Mode 19: (640x480) 640 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 20: (720x400) 720 x 400, 70 Hz
RetroArch [INFO] :: Found GL context: kms
RetroArch [INFO] :: Detecting screen resolution 1440x900.
RetroArch [INFO] :: [EGL]: EGL version: 1.4
RetroArch [INFO] :: [EGL]: Current context: 0x7f3590122ec0.
RetroArch [INFO] :: [KMS]: New FB: 1440x900 (stride: 6144).
RetroArch [INFO] :: [GL]: Vendor: Intel Open Source Technology Center, Renderer: Mesa DRI Intel(R) HD Graphics (Cherryview) .
RetroArch [INFO] :: [GL]: Version: 3.0 Mesa 11.2.0-rc1.
RetroArch [INFO] :: Querying GL extension: ARB_ES2_compatibility => exists
RetroArch [INFO] :: Querying GL extension: ARB_sync => exists
RetroArch [INFO] :: Querying GL extension: ARB_texture_float => exists
RetroArch [INFO] :: Querying GL extension: EXT_texture_sRGB => exists
RetroArch [INFO] :: Querying GL extension: ARB_framebuffer_sRGB => exists
RetroArch [INFO] :: GL: Using resolution 1440x900
RetroArch [INFO] :: [GL]: Default shader backend found: glsl.
[... 58 lines cut ...]
RetroArch [INFO] :: arg #6: -c
RetroArch [INFO] :: arg #7: /storage/.config/retroarch/retroarch.cfg
RetroArch [INFO] :: arg #8: -L
RetroArch [INFO] :: arg #9: DETECT
RetroArch [INFO] :: Threaded video stats: Frames pushed: 1023, Frames dropped: 0.
RetroArch [INFO] :: Monitor FPS estimation is disabled for threaded video.
" is neither a file nor directory. Ignoring.ECT
RetroArch [INFO] :: Redirecting save file to "/storage/savefiles/sonicwi3.srm".
RetroArch [INFO] :: Redirecting savestate to "/storage/savestates/sonicwi3.state".
RetroArch [INFO] :: === Build =======================================RetroArch [INFO] :: [CPUID]: Vendor: GenuineIntel
RetroArch [INFO] :: [CPUID]: Features: MMX MMXEXT SSE SSE2 SSE3 SSSE3 SSE4 SSE4.2 AES
MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 SSE4 SSE4.2 AES Built: May  3 2016
RetroArch [INFO] :: Version: 1.3.4
RetroArch [INFO] :: Git: b2e76de
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: [CPUID]: Vendor: GenuineIntel
[... cut ...]
"etroArch [ERROR] :: Failed to open libretro core: "DETECT
: cannot open shared object file: No such file or directory
RetroArch [ERROR] :: Fatal error received in: "load_dynamic()"
RetroArch [ERROR] :: Failed to start audio driver. Will continue without audio.
RetroArch [INFO] :: Saving config at path: "/storage/.config/retroarch/retroarch.cfg"
RetroArch [INFO] :: Saved new config to "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Monitor FPS estimation is disabled for threaded video.

Kivutar, 2016-05-25:
Ok, it looks like a playlist bug. Let try to boot the rom manually, from Load Content Detect Core

kbazzor, 2016-05-25:
Picked 3 different cores and nothing, the last one showed up for a few seconds though

And the middle one look to have gotten to the bios stage but has issues

 [INFO] :: === Build =======================================RetroArch [INFO] :: [CPUID]: Vendor: GenuineIntel
RetroArch [INFO] :: [CPUID]: Features: MMX MMXEXT SSE SSE2 SSE3 SSSE3 SSE4 SSE4.2 AES
MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 SSE4 SSE4.2 AES Built: May  3 2016
RetroArch [INFO] :: Version: 1.3.4
RetroArch [INFO] :: Git: b2e76de
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: [CPUID]: Vendor: GenuineIntel
RetroArch [INFO] :: [CPUID]: Features: MMX MMXEXT SSE SSE2 SSE3 SSSE3 SSE4 SSE4.2 AES
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Config: loading config from: /storage/.config/retroarch/retroarch.cfg.
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29975.00 Hz.
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/5xbr-retro.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bead.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/blinky.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/pixellate.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/quilez.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/retro-v2.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/scanline.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/sharp-bilinear.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/stock.glsl"
RetroArch [INFO] :: Video @ 960x720
RetroArch [INFO] :: Starting threaded video driver ...
RetroArch [INFO] :: [DRM]: Found 4 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: no
RetroArch [INFO] :: [DRM]: Connector 0 has 0 modes.
RetroArch [INFO] :: [DRM]: Connector 1 connected: no
RetroArch [INFO] :: [DRM]: Connector 1 has 0 modes.
RetroArch [INFO] :: [DRM]: Connector 2 connected: yes
RetroArch [INFO] :: [DRM]: Connector 2 has 21 modes.
RetroArch [INFO] :: [DRM]: Connector 2 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 3 connected: no
RetroArch [INFO] :: [DRM]: Connector 3 has 0 modes.
RetroArch [INFO] :: [DRM]: Mode 0: (1440x900) 1440 x 900, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 1: (1400x1050) 1400 x 1050, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 2: (1400x1050) 1400 x 1050, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 3: (1280x1024) 1280 x 1024, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 4: (1280x1024) 1280 x 1024, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 5: (1440x900) 1440 x 900, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 6: (1280x960) 1280 x 960, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 7: (1152x864) 1152 x 864, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 8: (1024x768) 1024 x 768, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 9: (1024x768) 1024 x 768, 70 Hz
RetroArch [INFO] :: [DRM]: Mode 10: (1024x768) 1024 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 11: (832x624) 832 x 624, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 12: (800x600) 800 x 600, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 13: (800x600) 800 x 600, 72 Hz
RetroArch [INFO] :: [DRM]: Mode 14: (800x600) 800 x 600, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 15: (800x600) 800 x 600, 56 Hz
RetroArch [INFO] :: [DRM]: Mode 16: (640x480) 640 x 480, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 17: (640x480) 640 x 480, 73 Hz
RetroArch [INFO] :: [DRM]: Mode 18: (640x480) 640 x 480, 67 Hz
RetroArch [INFO] :: [DRM]: Mode 19: (640x480) 640 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 20: (720x400) 720 x 400, 70 Hz
RetroArch [INFO] :: Found GL context: kms
RetroArch [INFO] :: Detecting screen resolution 1440x900.
RetroArch [INFO] :: [EGL]: EGL version: 1.4
RetroArch [INFO] :: [EGL]: Current context: 0x7f44ec122ec0.
RetroArch [INFO] :: [KMS]: New FB: 1440x900 (stride: 6144).
RetroArch [INFO] :: [GL]: Vendor: Intel Open Source Technology Center, Renderer: Mesa DRI Intel(R) HD Graphics (Cherryview) .
RetroArch [INFO] :: [GL]: Version: 3.0 Mesa 11.2.0-rc1.
RetroArch [INFO] :: Querying GL extension: ARB_ES2_compatibility => exists
RetroArch [INFO] :: Querying GL extension: ARB_sync => exists
RetroArch [INFO] :: Querying GL extension: ARB_texture_float => exists
[... 1153 lines cut ...]
RetroArch [INFO] :: [udev] Adding device /dev/input/event11 as type ID_INPUT_KEYBOARD.
RetroArch [INFO] :: [udev] Adding device /dev/input/event12 as type ID_INPUT_KEYBOARD.
RetroArch [INFO] :: [udev] Adding device /dev/input/event13 as type ID_INPUT_KEYBOARD.
RetroArch [INFO] :: [udev] Adding device /dev/input/mouse1 as type ID_INPUT_KEYBOARD.
RetroArch [INFO] :: [udev] Adding device /dev/input/event5 as type ID_INPUT_KEYBOARD.
RetroArch [INFO] :: [udev] Adding device /dev/input/mouse0 as type ID_INPUT_KEYBOARD.
RetroArch [INFO] :: [udev] Adding device /dev/input/event13 as type ID_INPUT_MOUSE.
RetroArch [INFO] :: [udev] Adding device /dev/input/mouse1 as type ID_INPUT_MOUSE.
RetroArch [INFO] :: [udev] Adding device /dev/input/event5 as type ID_INPUT_MOUSE.
RetroArch [INFO] :: [udev] Adding device /dev/input/mouse0 as type ID_INPUT_MOUSE.
RetroArch [INFO] :: Found joypad driver: "udev".
RetroArch [INFO] :: [CPUID]: Vendor: GenuineIntel
RetroArch [INFO] :: [CPUID]: Features: MMX MMXEXT SSE SSE2 SSE3 SSSE3 SSE4 SSE4.2 AES
RetroArch [INFO] :: ALSA: Using signed 16-bit format.
[... cut ...]
RetroArch [INFO] :: [CPUID]: Features: MMX MMXEXT SSE SSE2 SSE3 SSSE3 SSE4 SSE4.2 AES
RetroArch [INFO] :: Using font rendering backend: freetype.
RetroArch [INFO] :: SRAM will not be saved.
RetroArch [INFO] :: null: [/storage/.config/retroarch/content_history.lpl].
RetroArch [INFO] :: [KMS]: New FB: 1440x900 (stride: 6144).
RetroArch [INFO] :: [KMS]: Missed 82 VBlank(s) (Frame: 5750, DRM frame: 32156).
RetroArch [INFO] :: [KMS]: New FB: 1440x900 (stride: 6144).
RetroArch [INFO] :: [KMS]: Missed 4 VBlank(s) (Frame: 5755, DRM frame: 32161).

Kivutar, 2016-05-25:
Wait, these are the cores from the buildbot. They are unsupported, most of the time incompatible with Lakka.

Please delete all the files in your core directory, and just try with the regular FBA from Lakka.

kbazzor, 2016-05-26:
I used the lakka img for x64 straight off the website and thats how it was setup, now i can get it to work, but only but running it direct through the core, playlist still doesnt work

Kivutar, 2016-05-26:
Ok, so there is still the playlist bug. Can you post your LPL somewhere and paste a link here? The FBA LPL should be in your playlists folder. Thanks.

kbazzor, 2016-05-26:
Dropbox
https://www.dropbox.com/s/uvxaxgdxntxatp5/FB%20Alpha%20-%20Arcade%20Games.lpl?dl=0

Kivutar, 2016-07-11:
It may be due to this https://github.com/libretro/libretro-database/commit/15241134c20bd1d0c833fb01e219e521b0b2c8eb

I'm trying with a v0.2.97.38 merged romset

Kivutar, 2016-10-17:
I'm still able to scan my romset. Is this bug still present on master?

smurfxx, 2016-10-22:
I have the same problem, is this fixed with latest lakka nightly?

kbazzor, 2016-10-22:
i dont remember what build im on but its working for me now

Kivutar, 2016-10-22:
It's working in master at least. I close it.

closed 2016-10-22 by Kivutar

smurfxx, 2016-10-22:
I updated to latest version of Lakka, purged retroarch.cfg, deleted FB Alpha - Arcade Games.lpl, scanned again the folder with my roms but some roms are missing.
I tested my romset with Final Burn Alpha on windows and every rom is recognized. Do I have to do some additional thing?

Kivutar, 2016-10-23:
The recommanded romset is FB Alpha 0.2.97.38.7z (Split)

smurfxx, 2016-10-23:
You're right, I tried with a new romset and now it's perfect!

rick-pit, 2017-05-14:
I have some problem of that kind too, after download the recommended set with FBA some games doesnt work. i see the same error:

[FBA] Searching ROM at index XXX with CRC 0xXXXXXXXXX and name XXXXXX.rom => Not Found

I discart rom corruption in many ways... after that i try open mannualy in mame and that work there. i think is good enough for me but then some roms of the recommended set are not working in FBA? what you think guys? the rom i test is xmvsf.zip. i open that find and i check there is a txt file there. maybe that could do somthing with the index emulator try to check?

Kivutar, 2017-05-14:
Hello @rick-pit .
Things have changed since the last time, and the recommanded set is now 0.2.97.40.

This is the xmvsf.zip entry in our database https://github.com/libretro/libretro-database/blob/master/dat/FB%20Alpha%20-%20Arcade%20Games.dat#L30062

It's the first time I hear about some txt files included in an arcade rom, and it's certainly the cause of the problem.

We are using the CRC of the zip itself to check if a rom matches the database entry. If it contains an additional file like this, the CRC changes.