libretro/Lakka #385: Enabling bluetooth broken on XU3

Kivutar, 2016-05-26:
On my XU3, having .cache/services/bluetooth.conf isn't enough to get bluetooth started on boot.

I can do a systemctl start bluetooth and then it works.

Rodrigomezs, 2016-06-24:
I have same problem, when i restart lakka, i see .cache/services/bluetooth.conf  is not saved and controllers dont pair. I continue pairing by ssh every time i want to play via bluetooth.

epicbrew, 2017-05-29:
Same issue here. Enabling via "systemctl enable bluetooth" doesn't seem to permanently enable the service for some reason. But manually starting the service works fine. I'm using a USB bluetooth adaptor. I notice that a "systemctl status bluetooth" yields:

    Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; vendor preset: disabled)

Wondering if that "vendor preset: disabled" has anything to do with it...