libretro/Lakka #389: Unable to install on UEFI-only computer

maxengel, 2016-05-30:
I have a Cherry Trail-based Kangaroo PC that only supports UEFI. I have tried the recommended installation paths on Linux, Windows, and Mac, but the target machine never sees the created USB drive. It can see other created bootable media, but it seems that the .img file doesn't work on EFI-only machines.

Also, I am trying the 64 bit version for PC's.

Kivutar, 2016-05-30:
Hello, I saw a thread on the forums, about somebody compiling Lakka with UEFI support. Is that you?

maxengel, 2016-06-02:
That wasn't me. I've successfully compiled Lakka for 32 bit and 64 bit systems, but neither boots. FWIW, I'm compiling in Ubuntu 16.04 and using dd to copy the .img to a USB drive.

I tried joining the forums, but never got a confirmation email...

pmlopes, 2016-06-07:
FYI: I've just downloaded an image from a user in the forum:

http://libretro.com/forums/showthread.php?t=5920&highlight=uefi

And I can confirm that it boots and runs on my intel compute stick, the requirement is to set the UEFI to 32bit (Windows 8.1/10) and disable secure boot.

Maybe it would be nice to have these builds also listed on the website.

Kivutar, 2016-06-07:
I will start by adding those builds in the nightly script.

pmlopes, 2016-07-28:
Any updates on this?

closed 2016-07-29 by Kivutar

maxengel, 2016-11-14:
I am unable to boot from the EFI issues on my Atom-based Kangaroo PC. The drive is a selectable boot device and has been formatted as FAT with GPT, but when I select it as the boot device, it simply has a single underscore frozen in the top-left.

Kivutar, 2016-11-15:
Try this http://mirror.lakka.tv/nightly/Generic.i386/Lakka-Generic.i386-devel-20161106232131-r21640-ge218f0f-efi.img.gz

maxengel, 2017-01-22:
Unfortunately that doesn't work. Does it make a difference if I try installation when connected to a PC monitor vs. a TV? right now I am trying on my monitor.

conradportelli, 2017-01-25:
I'm in the same boat man I have a kangaroo as well . Same issues and to answer ur question tv or monitor does not matter . I find it so hard to understand why so many ego based images can be booted but not this

maxengel, 2017-01-25:
It is good to know I'm not alone but really too bad it won't install on the Kangaroo. Have people with Intel NUC's had success?

conradportelli, 2017-01-25:
I think the nucs have legacy options  and work with either uefi or legacy

maxengel, 2017-01-25:
So no UEFI-only systems work with Lakka?

conradportelli, 2017-01-25:
The only one I heard of is the lattepanda which can boot the libreelec versions of lakka .. I tried those and still nothing

maxengel, 2017-01-25:
Interesting... I'm wondering what the most powerful sub-$100 computer that can run Lakka is. I was hoping the Kangaroo would work.

Also, is this feature still under consideration?

conradportelli, 2017-01-25:
I'm sure it will work sooner or later I'm not gonna give up

pmlopes, 2017-01-25:
@maxengel intel compute stick 1 did boot lakka for me with the efi image. This stick has only UEFI support. I haven't tested any new images since the commit after my request above.

sammcj, 2017-02-14:
Indeed it seems the Lakka image / installer does not have UEFI boot capabilities, quite a shame as the three tablet like devices I have are all locked to EFI boot mode only.

cmcginty, 2017-04-25:
Trying to book from a Chromebox and I think I'm in the same boat.

commit b569ae08 (2016-07-29): Automatically build UEFI images. Fixes #389