libretro/Lakka #390: PSX ROM Scanner not picking up ROMs

akusei, 2016-05-31:
I added about 33 ROMs to the Lakka roms directory and started a scan. The scan only picked up about 10 of those ROMs. I made sure that every single .cue and .bin was named correctly, based on the information in the [libretro psx database file](https://github.com/libretro/libretro-database/blob/master/dat/Sony%20-%20PlayStation.dat). I will use "Ape Escape" as an example since that is one of the ROMs that was not found with the scanner.

I named the Ape Escape .cue file "Ape Escape (USA).cue" and the .bin file "Ape Escape (USA).bin". I also made sure that the reference to the bin file inside the .cue file was correct. I then did an md5 hash and a sha1 hash of the .bin file, this is what I got:

MD5: de4e7ab78c08bd03712e83c14d4cf642
SHA1: 466cce4bcd6992f57227abd270323bcdad2fb7fc

I also checked the .bin serial and it is "SCUS-94423". All of this information matches the information from [redump](http://redump.org/disc/3086/) exactly, even the provided cuesheet from redump is identical to mine.

With everything matching the dat from github and the information/hashes/checksums/names from redump, this ROM should have been found by the scanner but it wasn't.

Oh, and the ROM was in the directory scanned, not in a sub directory.

Kivutar, 2016-05-31:
Thanks, certainly a bug in the scanner. The serial is stored in many placed in the bin, and sometimes the syntax differs a little. How did you check the serial?

akusei, 2016-05-31:
I just search for the common prefix with a hex editor, like SCUS or SLUS if I know it's a USA version. Normally the serial is stored at offset 0x9340 but in the case of Ape Escape it's been replaced by all spaces which I find odd since the md5 matches redump exactly so it doesn't seem like someone did that on purpose but 0x9340 is where most of them should be.

Ape Escape was found at a few different offset but some of those have inserted characters like semi colons and periods. Here are the offsets and values I found in Ape Escape

Address | Value
------------ | -------------
0000CBDF | SCUS_944.23;1
0000D374 | SCUS_944.23;1
000B7ADA | SCUS-94423
000B7C1A | SCUS-94423

Kivutar, 2016-05-31:
Ok. The function that extracts the serial is there https://github.com/libretro/RetroArch/blob/master/tasks/task_database_cue.c#L143

It is way more complex than the one I had implemented initaly. @aliaspider do you have any idea of why these images doesn't scan?

aliaspider, 2016-05-31:
ill have to check it again, 
the older code looked blindly for the serial where this one reads the iso file system to find the SYSTEM.CNF file and extract the serial from it, so of course it is more complex :P

akusei, 2016-06-01:
I don't have a debug environment so I just went through the function manually and plugged in the values for a ROM that works and a ROM that doesn't. I found where the problem is but I'm unsure of how to fix it because it may be a problem with the ROM itself and I don't know the filesystem structure so you guys will have to be the judge.

The working ROM results in SYSTEM.CNF being located in the 23rd sector at file offset 0xD375 and the data located there is correct (showing some of previous file since it is part of the problem):

`BOOT = cdrom:\SLUS_009.66;1`
`
The broken ROM results in SYSTEM.CNF being located at the same place in the 23rd sector at file offset 0xD375 and this is the data at that offset:

`BOOT=cdrom:\SCUS_944.23;1`

Notice that the offsets are the same but the previous sector has **BOOT=cdrom** in the broken one and **BOOT = cdrom** in the working ROM. So with the offsets being the same, the broken ROM ends up trying to parse **CUS_944.23;1** resulting in the serial **CUS_-44.3;**.

Again, I'm not sure if this is a result of an error in the ROM or a calculation in the code but the numbers seem to be correct in the ROM and the function does what's expected, the removed spaces around the = is making things unaligned.

This seems to be the case for all the ROMs I have that did not get picked up by the scanner, they all have missing spaces around the = in `BOOT = cdrom`.

akusei, 2016-06-01:
How is this an enhancement? It is definitely a bug in the function. The function at line 201 assumes that the text `BOOT = cdrom:` is exactly 13 characters and hard codes that value into the seek:

`filestream_seek(fp, 13 + skip + cd_sector * frame_size, SEEK_SET);`

The start of the file is correct in the ROM, the code is just assuming a fixed length for data inside SYSTEM.CNF. If anything, the code should do this:

```
filestream_seek(fp, skip + cd_sector * frame_size, SEEK_SET);
filestream_read(fp, buffer, 256);

tmp = (uint8_t*)strchr((const char*)buffer, '\\');
tmp++;
```

Error checking and correcting for if \ is not found was omitted in the above example because it's just an example. Either way, the issue is definitely a bug in the function and not an enhancement.

aliaspider, 2016-06-02:
yeah strchr that should work, you need to set buffer[255] to 0 to prevent overflow though, since the function might be reading an invalid file.
and SYSTEM.CNF is reliable, or you know any official software with that doesn't have the serial there ?

akusei, 2016-06-02:
You're right, all that should be needed is to remove the hard coded 13 and doing a strrchr will still work. The only reason I suggested strchr instead is because _typically_ the first entry in the SYSTEM.CNF file is the boot entry which is what you want so strchr will have better performance than strrchr since it will have less characters to check before hitting '\'.

I'm also pretty sure that '\' will always be there since the boot entry is a path to a file on the disk. Being that it's a path on the disk, I'm thinking even though strchr would have better performance, strrchr would be what you still want since the path could contain multiple '\' characters (i.e. cdrom:\folder1\folder2\file.ext).

SYSTEM.CNF should still be reliable enough since it seems most all commercial ROMs seem to follow the same naming convention for the boot file but that's still not a 100% certainty since the developer is responsible for naming that file and it's not actually a serial field. To be 100% accurate, it would be better to take an MD5 hash of the ROM and compare that instead of the serial, although that would be much slower.

I will create a small program made from the source in the function to run over all the 250 or so ROMs I have and see if any one of those doesn't have the serial as the file name for the boot entry.

akusei, 2016-06-02:
Out of the 250 images I tested only 2% deviated from the current assumption that BOOT will always have the same format. Below are the following formats that I have seen in those 2%.

```
cdrom:\\XXXX_###.##
cdrom:XXXX_###.##
cdrom:\XXXX_##.###
cdrom:\XXXX_###.##
cdrom:\XXXXXX\XXXX_###.##
```

Notice that some have a single '\', some have multiple '\' in a row, some have no '\', some are in different directories resulting in multiple spaced out '\' and some don't follow the XXX_###.## naming convention and instead place the '.' in a different location. My sample size was very small compared to the total number of PSX images out there. I do have more that I can test, it will just take some more time to get them uncompressed. Either way, it looks like the parsing of SYSTEM.CNF could use some changes.

However, in ALL cases the BOOT entry started with `cdrom:` and ended with `CRLF` keep in mind that not all of them ended with `;1`. I'll add more info once I'm able to run every image I have through the test.

aliaspider, 2016-06-03:
"MD5 hash of the ROM" you're kidding right :P ?
thanks for gathering the data, that is certainly a very big help, it also remind me why I went with the fixed 13 offset since it would have worked for all those cases, except the ones with the missing spaces of course. 
I'll try to think of something, maybe just look for "cdrom:" and then strrchr from there.

akusei, 2016-06-03:
I finished scanning the remainder of the images I have. I scanned 1278 images and of those images 335 (26%) had serials that could not be found by the existing function. I've attached all of the BOOT entries from all 1278 images' system.cnf files. Also, in 1 image, the sector that contained the pointer to the file table that contains system.cnf was located in the 166th sector not the 16th which I found really odd and it was a valid licensed image.

@aliaspider I wasn't serious about implementing MD5, I said it just to put another option out there but I know that it would be entirely impossible due to performance reasons.

[serials.txt](https://github.com/libretro/Lakka/files/297027/serials.txt)

aliaspider, 2016-06-03:
pushed a fix : https://github.com/libretro/RetroArch/commit/87788f0ffb39beb3ccda14cda6dc07ac2d07b399

can you test it ?

akusei, 2016-06-03:
Ran the test again with the updated code and it still has just a couple issues. It failed to find the serial where the following case is true:

- BOOT parameter is not the first line (making \n appear before the argument)

The above case summarizes the 3 things I found in a few system.cnf files

1. The first entry in system.cnf is not always BOOT
2. Some system.cnf files start with CRLF
3. Some system.cnf files have -- comments in them, making boot not on the first line

aliaspider, 2016-06-03:
pushed another fix : https://github.com/libretro/RetroArch/commit/e23b39d2b5407fa6af8fd5d70de608ec3ee6772d
can you try again ?

akusei, 2016-06-03:
Works great. Out of the 1278 images, 100% of the serials were extracted; with the exception of that 1 image that has the file table pointer in the 166th sector instead of the 16th but that seems really odd to me in the first place. I'd consider this fixed.

akusei, 2016-06-03:
Is the serial being testing against the serial from the libretro dat file or the serial from redump? If it's comparing against the dat file then there's 1 problem. The dat file has a serial of SLUS-P01206 and SLUS-P01346 and the scanner will ignore the P and make then SLUS-01206 and SLUS-01346, this will cause the image to not be added to the playlist because they won't match

aliaspider, 2016-06-04:
the scanner ignores the P ? you sure ? this commit should have fixed that : 
https://github.com/libretro/RetroArch/commit/748a913a6d92ed7046b2c32136625ba5b6dd1510

akusei, 2016-06-04:
You're right, my mistake sorry. When I copied the code, I added an extra tmp++ by accident after appending the '-'. All is good.

closed 2016-06-04 by aliaspider