libretro/Lakka #392: DOSBox not working with Cubieboard2

MadMax33309, 2016-06-02:
Hello,

i tried to start DOSBox (via Load Core and Start Core) but only a blackscreen appears and then back to the main menu.

I use Cubieboard2 with a Xbox 360 controller.
I have the newst version of Lakka installed.

Kivutar, 2016-06-02:
OK, delete everything in your **cores** folder. Those cores are incompatible with Lakka.

Download and flash this image http://static.kivutar.me/Lakka-a20.arm-devel-20160602142349-r21459-gb403498.img.gz

It contains a packaged DOSBox core. I was able to launch the core with success on my Cubietruck.

If you don't want to erase your current SD card, uou can also wait for the next update.

MadMax33309, 2016-06-02:
Works now, but the updater is broken.
Can only update Lakka and Thumbnails.
I will use the old version for now. Ty for your effort.

Kivutar, 2016-06-02:
The updater is not broken. I removed it.

There is no need for it.

All the assets/joypads/shaders/cores/etc are already upgraded when you upgrade Lakka. The only thing you can do with it is break your install. So I disabled it.

closed 2016-06-02 by Kivutar

MadMax33309, 2016-06-02:
Ok I understand.
Ty for your work.

Kivutar, 2016-06-02:
NP. I updated scummvm and it works now on arm. It will land in the upgrades tonight.

MadMax33309, 2016-06-02:
wow that was fast, nice done ty.

dragonkid, 2016-06-11:
I have update to the newest version of nightly build, but the dosbox can still not run on my raspberry pi3.

Does that fix only work on Cubieboard2?

Kivutar, 2016-06-11:
No the fix is effective everywhere. And I tried it on my RPi3 too.

You have to load the core manually, then launch it. You will end up on a DOS like interface, where you can type commands to launch a game.

dragonkid, 2016-06-11:
I have tried what you said. I met the same problem like MadMax33309.

Which version of lakka did you tried on RPi3?

Kivutar, 2016-06-11:
I tried the last update from the online updater