libretro/Lakka #393: LattePanda cannot boot from USB PC image

MadMax33309, 2016-06-02:
Hello,

I download the image for PC and flash it on a USB drive.
LattePanda (http://www.lattepanda.com/) can't boot it from because cannot detect the USB drive.
If I understand it right, the name of the USB drive have to be "WINPE" :
http://www.lattepanda.com/docs/#How_to_re-install_LattePanda_Windows_system?
It is possible to give us a option the name the USB drive?
Because maybe it will works with the right name to boot from a usb on the LattePanda.

Kivutar, 2016-06-02:
Interesting device.

Can you check if it has a BIOS or at least an UEFI with BIOS emulation?

MadMax33309, 2016-06-02:
Yeah it has a BIOS but the USB driver still be not detected.
Hm after flash it on USB, windows can only give me the erase the USB drive. I cannot find a option die rename the first partition.

Kivutar, 2016-06-02:
Can you boot Plop? https://www.plop.at/en/bootmanagers.html

MadMax33309, 2016-06-02:
Nope sorry.
Plop look like don't support win10. I tried to install it but don't worked.
And I don't have a external CD drive.
I even tried it with the SD card slot...

Kivutar, 2016-06-02:
Or maybe they let only Windows boot? Have you tried any other OS like an Ubuntu?

MadMax33309, 2016-06-02:
Nope but I speak already with someone and he tried to install but with the same issues.
I believe they blocked it. I saw a line in the EFI shell with something "blocked device" and it was the USB drive. My know how about EFI shell is very bad so I didn't attempt to do something.
Can you creat a image with the label "WINPE"? 
Because LattePanda can boot from a USB (see here: http://www.lattepanda.com/docs/#How_to_re-install_LattePanda_Windows_system?)

MadMax33309, 2016-06-02:
And I found you can update the BIOS via USB. If I have time, I will update the BIOS. Maybe this will help?
It have to be FAT32. I cannot see what format the USB drive have after flashing the image on it.

Kivutar, 2016-06-02:
Let try to find a tool on windows to rename the first partition to WINPE

MadMax33309, 2016-06-02:
I found this tool: https://rufus.akeo.ie/?locale=en_US
But after I load your image, i am unable to rename the drive.
Your image block maybe this.
I tried all kinds of rename function of windows but noone of them work.

MadMax33309, 2016-06-02:
Ok WINPE have to be the name of the USB drive.
I flash with the Plop CD image a USB drive and now the LattePanda detect the usb drive BUT cannot boot from it...
I don't know why.
But I think the boot device need a UEFI shell... because at the start of the boot line from USB stand "UEFI:"

MadMax33309, 2016-06-07:
Hey,
I did a little bit research and find out the boot device need a UEFI/EFI on it.
Linux can be installed on it in this way. Can you make a image with a EFI boot loader?
I am not really sure if it still important that the name of the USB has to be WINPE or a EFI boot loader is enough.

Kivutar, 2016-06-07:
Ok, let see this thread https://github.com/libretro/Lakka/issues/389
There is an EFI image on the forums. I'm working on automating this.

MadMax33309, 2016-06-07:
I tried the EFI image.
Good news is the LattePanda still detect it (so the name is not important) but the bad news is, I can't still not boot it from it. After a blackscreen the LattePanda reset itself and try it again but don't work.

MadMax33309, 2016-06-08:
Good news.
I use your new EFI image and it works!
Can boot, install and boot Lakka. Nice.
But unfortunately we have now another problem...
http://i.imgur.com/MkUNtyt.jpg
Any ideas? Controller works, so I can navigate through the menu.

closed 2016-06-08 by MadMax33309

5Up3r2, 2017-01-02:
hi MadMax33309,
Its an intermittent problem. After i reset it it got fixed
Cheers