libretro/Lakka #394: Xbox One (Wired) Controller Flashes then wont turn on

aidanayala, 2016-06-05:
Hi,

Plug in my xbox one controller. It flashes for about 30 seconds then turns off.

Lakka display the message "Microsoft Xbox pad configured port 0" and also display the disconnect message. 

However it does not seem to "keep it alive" or get any input.

The pad works on my laptop and my PC.

Lakka is running on a custom build. Gigabyte Mobo. 8GB ram, AMD CPU with integrated GPU.

I also have no sound but this is a separate issue.

gouchi, 2016-06-05:
@Kivutar is it possible to update the [patch](https://github.com/libretro/Lakka/blob/lakka/packages/linux/patches/4.4.2/linux-063-xpad-fix_xbox360_wireless.patch) with [latest update](https://github.com/paroj/xpad) as the patch is from Sun, 20 Dec 2015 ?

elitedevx, 2016-06-12:
Just confirming that this bug exists in current nightly build.

It appears that this isn't limited to just the Xbox One controller as the same issue occurs with a PS3 dual shock controller connected via USB. udev does detect both controllers, however there's no input being registered on the device.

Kivutar, 2016-06-12:
I have a PS3 Dualshock, and pressing the home button once is required to get inputs.

elitedevx, 2016-06-12:
I'm sorry, seems I was a victim of a faulty USB cable. However, I'm still confirming that the Xbox One controller is not working in current build.

Kivutar, 2016-06-12:
Ok, I will buy one with the money from the donations, and I will try to fix it. It is a very common controller so we need to support it.

Kivutar, 2016-06-29:
dmesg:

```
[  124.680014] usb 1-1.2: new full-speed USB device number 4 using dwc_otg
[  124.776873] usb 1-1.2: New USB device found, idVendor=045e, idProduct=02d1
[  124.776897] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  124.776910] usb 1-1.2: Product: Controller
[  124.776922] usb 1-1.2: Manufacturer: Microsoft
[  124.776933] usb 1-1.2: SerialNumber: 7EED87729711
[  124.800585] input: Microsoft X-Box One pad as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/input/input0
[  124.801307] usbcore: registered new interface driver xpad
```

`/dev/input/js0` created but nothing added when button pressed.

Works on RPi3, but not on Generic. Also reported to be broken on C1 and a20.

Kivutar, 2016-07-10:
This is certainly fixed by the last commits about xbox one. I tried mine and it works.

closed 2016-07-10 by Kivutar

vbmithr, 2016-11-15:
My Xbox one controller does not work with the latest release of Lakka.

JorgeHawkins, 2016-11-27:
My XBox One controller is not working in the latest nightly build. 
Also Wi-Fi works intermittently. When I try to download thumbnails or updates, they disconnect and fail to download. 

Specs:
Video Card: NVidia Quadro 2000 GDDR5 1GB
Motherboard: BIOSTAR G41D3C
Processor: Core 2 Quad Q8200 @ 2.33GHz
RAM: 8 GB
HDD: 2 x 512 GB

Kivutar, 2016-11-27:
OK I try it

Kivutar, 2016-11-27:
Mine works, I will try to check the joypad config repo