libretro/Lakka #395: Two-player arcade controller kit's joysticks not reporting as separate sticks

kevinfishburne, 2016-06-06:
I'm using Lakka in my arcade cabinet, which uses a 16-button, two-joystick kit and USB controller. Previously it worked fine on the same hardware in MAME under Mint 17. In Lakka everything works fine except the joysticks report as the same two axes when mapping the directions for player one and player two. So moving player one moves player two, and vice-versa. I tried clearing the digital stick mappings, mapping the sticks as analog, enabling the option to convert analog to digital, and using the left joystick as the left analog and right joystick as the right analog but it exhibits the same behavior. Retroarch treats the two joysticks as the same joystick no matter what I do.

The USB controller board for the button/joystick kit reports in Retroarch as "THT Arcade console 2P USB Player (#0)".

Kivutar, 2016-06-06:
Ok, I think I will need the help of @twinaphex on this one. IIRC, he also own a similar controller. Could you redeclare this bug in the retroarch issue tracker too?

kevinfishburne, 2016-06-07:
I've resubmitted it under Retroarch here: https://github.com/libretro/RetroArch/issues/3084

kevinfishburne, 2016-06-20:
Bug may affect non-arcade hardware. Try on one gamepad with twin sticks, using the left stick to map player one's directions and the right stick to map player two's directions. Mappings for axes report +/- 0 and 1, but not 2 and 3 so they become locked.

Kivutar, 2016-07-13:
@kevinfishburne I forgot to ask, are you using autoconfig or manual binding here?

rixter110, 2016-07-13:
I am new to this but had similar issues on raspberry pi with same controller. By modifying the boot/cmdline.txt file so the system recognizes the two players using the xinmo usb encoder so the settings will be passed to the kernel.  Add the following to /boot/cmdline.txt (just add a space to the existing entry so its all on one line)
usbhid.quirks=0x16c0:0x05e1:0x040

The first two numbers the vendor/product (16c0 and 05e1).

On Lakka with intel Nuc
Not able to write directly to syslinux.cfg, "write only file system" ..not sure if this is because its running in live mode. Unable to test to see if I can write to nano /flash/extlinux.conf because I dont have it installed but that might work.

Kivutar, 2016-07-13:
You can remount it as read/write http://www.lakka.tv/doc/The-bootloader/

Kivutar, 2016-10-17:
OK. Try to disable Input Autoconfig. Then, bind manually both pads and tell me if it works.

elric91, 2017-05-08:
Workaround confirmed working on Lakka 2 :
- log through SSH (enable SSH with the GUI, log with root/root)
- enable write on the partition : mount -o remount,rw /flash
- add the quirks "usbhid.quirks=0x16c0:0x05e1:0x040" at the end of the file /flash/extlinux.conf
- save & reboot

et voilà :)

Kivutar, 2017-05-09:
Can somebody test that it works with this image please? http://static.kivutar.me/Lakka-Generic.x86_64-8.0-devel-20170509120246-r26121-g0d61319.img.gz