libretro/Lakka #396: XboxDrv

KingPatte, 2016-06-07:
Is it maybe possible to put the XboxDrv in the Build instead of Xpad?

The Controller shuts after a little Time down.

Kivutar, 2016-06-07:
Wired or wireless controller?

KingPatte, 2016-06-07:
Its the wireless Controller and it does not show an Playerindicator.

Kivutar, 2016-06-07:
Ok, what image of lakka are you using?

KingPatte, 2016-06-07:
I'm using the last one

Kivutar, 2016-06-07:
For what hardware? The kernel differs.

KingPatte, 2016-06-07:
I have the Raspberry Pi 3

Kivutar, 2016-06-28:
Xbox360 or Xbox One?

KingPatte, 2016-06-29:
The XboxDrv is for the Xbox 360 Wireless Controller.

gyates100895, 2016-08-19:
If you did add xboxdrv, i could easily fix the analog control stick problem with mayflash controllers. Ive already made the systemd units to basically manage the program, its really a matter of making the xboxdrv configs.

Antiriad, 2017-01-18:
I get the exact same problem (Xbox 360 wireless controller) it shuts after a little time down, even during gameplay.
IMHO switching to xboxdrv will fix also the [annoying blinking light](https://www.youtube.com/watch?v=Uvkm9CMZh40). 

Thank you

neosquid, 2017-03-10:
Any news on this? Just got the 240p composite thing working on CRT with PI. 

I've now started using Lakka seriously, and now the only problem is the frequent wireless 360 controller disconnects described in this thread that happen around every 2 to 3 mins, making gaming impossible.

Thanks.

Kivutar, 2017-03-10:
I was never able to reproduce this bug. I have two 360 controllers at home and none of them is disconnecting.

neosquid, 2017-03-11:
Apologies, the controller batteries were running out! No disconnects with fresh batteries installed.