libretro/Lakka #397: Screen issue with the new EFI image on the LattePanda

MadMax33309, 2016-06-08:
Hey,

I installed the new EFi 64bit image on a USB drive for the LattePanda.
But it have screen isuues
http://i.imgur.com/MkUNtyt.jpg
I can navigate through the menu and can access via SSH.
It seems to run games perfectly (abside of the screen issue ofc^^).

Kivutar, 2016-06-08:
Ok, so now Lakka boots on the device?

On windows 10, can you force the resolution at 1080p, then reboot to Lakka?

MadMax33309, 2016-06-08:
Weird, I reinstalled it and the screen issue was gone but after a reboot it was there again...
Win10 was already with a resolution at 1080p.
Something else I can try?

MadMax33309, 2016-06-08:
Okay, if I first boot win10 then restart within win10, lakka boot without screen issue but sound over HDMI don't work...
After that if I restart the LattePanda and boot lakka the screen issue are coming back

MadMax33309, 2016-06-08:
And after the restart within win10, the window folder copy itself in the ROM folder(btw the system folder with the EFI is in the ROM folder too).

Kivutar, 2016-06-09:
Those two folders are not copied, they are mounted. Like when you plug an USB key.

MadMax33309, 2016-06-09:
Alright, any possibilities to prevent that?
And any idea what exactly is the reason for the screen issue?
I mean I can do always first boot win and restart into lakka but after that I still have no sound(over HDMI and Jack)

Kivutar, 2016-06-09:
To prevent that, you can pass udevil=0 to the bootloader. But you will loose the ability to automount USD keys.

I think the screen issues is due to a mismatch between the bootloader resolution and the screen resolution. Forcing a resolution on the bootloader would certainly fix the problem. Read about the video= argument at the bottom of this page https://wiki.archlinux.org/index.php/kernel_mode_setting

MadMax33309, 2016-06-09:
This are my connectors:
DP-1: disconnected
DP-2: disconnected
DSI-1: connected
HDMI-A-1: disconnected
HDMI-A-2: connected
and regardless what I do nothing change. I cannot change the resolution or force to turn on/off a connector.
I connected via SSH with Putty.
Did I something wrong?

Kivutar, 2016-06-09:
I think that the thing you tried "video=HDMI-A-2:1920x1080@60" is good.

So maybe the problem is something else.

@gouchi any idea about this?

MadMax33309, 2016-06-09:
I think video= have no effect at all.
I have a two monitor setup and on one TFT was my PC and on the other was the LattePanda.
I type this "video=HDMI-A-2:1024x768" and nothing happend on the screen from the LattePanda.
I assume something has to be happend or?

MadMax33309, 2016-06-09:
This is really weird.
If I boot win10 or the BIOS/Setup and then boot lakka, no screen issue at all.
And no sound at all.

MadMax33309, 2016-06-09:
Ok I believe I found the problem.
I changed a setting in the BIOS and now the screen issue are gone.
Something with the VBIOS but unfortunately I cannot check it anymore because it seems the videocard driver load after the boot now and if I go into the BIOS I have only a blackscreen...

So now only the non-sound issue to fix. Should I open a new issue and close this or should I let this issue open?

MadMax33309, 2016-06-09:
Found it.
I change the GOP driver setup.
"Enable GOP driver will unload VBIOS
Disable it will load VBIOS"
I disable it and the problem was fix
I believe this was the reason why it works after I boot win10 or the Setup/BIOS, becaus the driver was already reloaded I guess.
I think you have to change regard to the GOP driver something

Kivutar, 2016-06-09:
add udevil=0 in the cmdline

MadMax33309, 2016-06-09:
Where can I find the cmdline in the new image? Under /flash/ is not there. Or should I create it?

Kivutar, 2016-06-10:
Under /flash, do you see a extlinux.conf?

MadMax33309, 2016-06-10:
Nope.
After ls:
EFI                        System Volume Information  syslinux.cfg
KERNEL                     ldlinux.c32
SYSTEM                     ldlinux.sys

Kivutar, 2016-06-10:
Ok what is in your syslinux.cfg?

MadMax33309, 2016-06-10:
DEFAULT linux
PROMPT 0

LABEL linux
 KERNEL /KERNEL
 APPEND boot=LABEL=System disk=LABEL=Storage  ssh quiet
vt.global_cursor_default=0 loglevel=2

Kivutar, 2016-06-10:
APPEND boot=LABEL=System disk=LABEL=Storage ssh quiet udevil=0

Kivutar, 2016-06-10:
You will need to remont flash as read write before:

    mount -o remount,rw /flash

MadMax33309, 2016-06-10:
Ok.
What is with this part
> vt.global_cursor_default=0 loglevel=2
Should I remove it?

Kivutar, 2016-06-10:
That part is to hide the blinking cursor of the console, and to make the kernel more silent. You can keep it, unless you want to debug something.

MadMax33309, 2016-06-10:
Hm doesn't work...
Still mount Windows and System.
Or do I have to dismount both manually?

> DEFAULT linux
> PROMPT 0
> 
> LABEL linux
>  KERNEL /KERNEL
>  APPEND boot=LABEL=System disk=LABEL=Storage  ssh quiet udevil=0
> vt.global_cursor_default=0 loglevel=2

Kivutar, 2016-06-12:
I don't get why it doesn't work. Can you check if the udevil service is still running?

MadMax33309, 2016-06-14:
I don't know hot to check if its still running...
I checked top but don't saw anthing with udevil

Kivutar, 2016-06-14:
systemctl status udevil

should do it

MadMax33309, 2016-06-14:
> ● udevil.service
>    Loaded: not-found (Reason: No such file or directory)
>    Active: inactive (dead)

Kivutar, 2016-06-14:
Sorry, systemctl status udevil-mount

MadMax33309, 2016-06-14:
> ● udevil-mount.service
>    Loaded: not-found (Reason: No such file or directory)
>    Active: inactive (dead)

Kivutar, 2016-06-14:
Ok, wrong again, just type systemctl to see the list of services?

MadMax33309, 2016-06-14:
```

sys-devices-pci0000:00-0000:00:02.0-backlight-acpi_video0.device loaded active plugged   /sys/devices/pci0000:00/0000
sys-devices-pci0000:00-0000:00:02.0-backlight-acpi_video1.device loaded active plugged   /sys/devices/pci0000:00/0000
sys-devices-pci0000:00-0000:00:02.0-backlight-acpi_video2.device loaded active plugged   /sys/devices/pci0000:00/0000
sys-devices-pci0000:00-0000:00:02.0-backlight-acpi_video3.device loaded active plugged   /sys/devices/pci0000:00/0000
sys-devices-pci0000:00-0000:00:02.0-backlight-acpi_video4.device loaded active plugged   /sys/devices/pci0000:00/0000
sys-devices-pci0000:00-0000:00:02.0-backlight-acpi_video5.device loaded active plugged   /sys/devices/pci0000:00/0000
sys-devices-pci0000:00-0000:00:02.0-backlight-acpi_video6.device loaded active plugged   /sys/devices/pci0000:00/0000
sys-devices-pci0000:00-0000:00:02.0-backlight-acpi_video7.device loaded active plugged   /sys/devices/pci0000:00/0000
sys-devices-pci0000:00-0000:00:14.0-usb1-1\x2d3-1\x2d3.1-1\x2d3.1:1.0-tty-ttyACM0.device loaded active plugged   Leonardo (CDC ACM, HID)
sys-devices-pci0000:00-0000:00:14.0-usb1-1\x2d4-1\x2d4:1.0-net-eth0.device loaded active plugged   USB_10_100_LAN
sys-devices-pci0000:00-0000:00:14.0-usb2-2\x2d1-2\x2d1:1.0-host0-target0:0:0-0:0:0:0-block-sda-sda1.device loaded active plugged   Ultra_Fit System
sys-devices-pci0000:00-0000:00:14.0-usb2-2\x2d1-2\x2d1:1.0-host0-target0:0:0-0:0:0:0-block-sda-sda2.device loaded active plugged   Ultra_Fit Storage
sys-devices-pci0000:00-0000:00:14.0-usb2-2\x2d1-2\x2d1:1.0-host0-target0:0:0-0:0:0:0-block-sda.device loaded active plugged   Ultra_Fit
sys-devices-pci0000:00-80860F14:00-mmc_host-mmc0-mmc0:0001-block-mmcblk0-mmcblk0boot0.device loaded active plugged   /sys/devices/pci0000:00/8086
sys-devices-pci0000:00-80860F14:00-mmc_host-mmc0-mmc0:0001-block-mmcblk0-mmcblk0boot1.device loaded active plugged   /sys/devices/pci0000:00/8086
sys-devices-pci0000:00-80860F14:00-mmc_host-mmc0-mmc0:0001-block-mmcblk0-mmcblk0p1.device loaded active plugged   /sys/devices/pci0000:00/8086
sys-devices-pci0000:00-80860F14:00-mmc_host-mmc0-mmc0:0001-block-mmcblk0-mmcblk0p2.device loaded active plugged   /sys/devices/pci0000:00/8086
sys-devices-pci0000:00-80860F14:00-mmc_host-mmc0-mmc0:0001-block-mmcblk0-mmcblk0p3.device loaded active plugged   /sys/devices/pci0000:00/8086
sys-devices-pci0000:00-80860F14:00-mmc_host-mmc0-mmc0:0001-block-mmcblk0-mmcblk0p4.device loaded active plugged   /sys/devices/pci0000:00/8086
sys-devices-pci0000:00-80860F14:00-mmc_host-mmc0-mmc0:0001-block-mmcblk0.device loaded active plugged   /sys/devices/pci0000:00/8086
sys-devices-platform-serial8250-tty-ttyS1.device loaded active plugged   /sys/devices/platform/serial
sys-devices-platform-serial8250-tty-ttyS2.device loaded active plugged   /sys/devices/platform/serial
sys-devices-platform-serial8250-tty-ttyS3.device loaded active plugged   /sys/devices/platform/serial
sys-devices-pnp0-00:01-tty-ttyS0.device loaded active plugged   /sys/devices/pnp0/00:01/tty/
sys-devices-virtual-block-loop0.device loaded active plugged   /sys/devices/virtual/block/l
sys-devices-virtual-block-ram0.device loaded active plugged   /sys/devices/virtual/block/r
sys-devices-virtual-block-ram1.device loaded active plugged   /sys/devices/virtual/block/r
sys-devices-virtual-block-ram10.device loaded active plugged   /sys/devices/virtual/block/r
sys-devices-virtual-block-ram11.device loaded active plugged   /sys/devices/virtual/block/r
sys-devices-virtual-block-ram12.device loaded active plugged   /sys/devices/virtual/block/r
sys-devices-virtual-block-ram13.device loaded active plugged   /sys/devices/virtual/block/r
sys-devices-virtual-block-ram14.device loaded active plugged   /sys/devices/virtual/block/r
sys-devices-virtual-block-ram15.device loaded active plugged   /sys/devices/virtual/block/r
sys-devices-virtual-block-ram2.device loaded active plugged   /sys/devices/virtual/block/r
sys-devices-virtual-block-ram3.device loaded active plugged   /sys/devices/virtual/block/r
sys-devices-virtual-block-ram4.device loaded active plugged   /sys/devices/virtual/block/r
sys-devices-virtual-block-ram5.device loaded active plugged   /sys/devices/virtual/block/r
sys-devices-virtual-block-ram6.device loaded active plugged   /sys/devices/virtual/block/r
sys-devices-virtual-block-ram7.device loaded active plugged   /sys/devices/virtual/block/r
sys-devices-virtual-block-ram8.device loaded active plugged   /sys/devices/virtual/block/r
sys-devices-virtual-block-ram9.device loaded active plugged   /sys/devices/virtual/block/r
sys-module-fuse.device      loaded active plugged   /sys/module/fuse
sys-subsystem-net-devices-eth0.device loaded active plugged   USB_10_100_LAN
-.mount                     loaded active mounted   /
dev-mqueue.mount            loaded active mounted   POSIX Message Queue File Sys
flash.mount                 loaded active mounted   /flash
storage-roms-SYSTEM.mount   loaded active mounted   /storage/roms/SYSTEM
storage-roms-Windows.mount  loaded active mounted   /storage/roms/Windows
storage.mount               loaded active mounted   /storage
sys-fs-fuse-connections.mount loaded active mounted   FUSE Control File System
sys-kernel-debug.mount      loaded active mounted   Debug File System
tmp-assets.mount            loaded active mounted   Assets directory
tmp-cores.mount             loaded active mounted   Cores directory
tmp-database.mount          loaded active mounted   Database directory
[... cut ...]
systemd-tmpfiles-clean.timer                                              loaded active waiting   Daily Cleanup of Temporary Directories

LOAD   = Reflects whether the unit definition was properly loaded.
ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
SUB    = The low-level unit activation state, values depend on unit type.

[[0;1;39m116 loaded units listed.[[0m Pass --all to see loaded but inactive units, too.
To show all installed unit files use 'systemctl list-unit-files'.
```

Kivutar, 2016-06-14:
Ok, I see

    storage-roms-SYSTEM.mount   loaded active mounted   /storage/roms/SYSTEM
    storage-roms-Windows.mount  loaded active mounted   /storage/roms/Windows

That are certainly created by

    udevil-mount@-dev-sda1.service                                            loaded active exited    Udevil mount service

So udevil is still loaded. I don't get why. Adding udevil=0 works well for me on RPi, and I tested in on PC some time ago too....
    udevil-mount@-dev-sda2.service                                            loaded active exited    Udevil mount service

MadMax33309, 2016-06-14:
maybe syslinux.cfg isn't the right file?
Or maybe I did something wrong?

MadMax33309, 2016-07-18:
I read something about that intel have new drivers for the GPU of the Atom x5 CPUs.
Not sure if this helpful:
https://01.org/linuxgraphics/downloads

gouchi, 2016-07-18:
Is it possible to give the output of 
`lspci -nn | grep VGA`
and
`dmesg`

You need to access [Lakka command line interface](http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface/).

Thank you.

MadMax33309, 2016-07-18:
> 00:02.0 VGA compatible controller [0300]: Intel Corporation Device [8086:22b0] (rev 22)

> [    0.000000] BIOS-e820: [mem 0x000000005ba76000-0x000000005ba76fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000005ba77000-0x000000005ba79fff] usable
[    0.000000] BIOS-e820: [mem 0x000000005ba7a000-0x000000005ba7afff] reserved
[    0.000000] BIOS-e820: [mem 0x000000005ba7b000-0x000000005bffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000e3ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fea00000-0x00000000feafffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed03000-0x00000000fed03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed06000-0x00000000fed06fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed08000-0x00000000fed09fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1cfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fedbffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffc00000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000017fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.40 by American Megatrends
[    0.000000] efi:  ESRT=0x5b150000  ACPI=0x5b15b000  ACPI 2.0=0x5b15b000  SMBIOS=0x5b90df98
[    0.000000] efi: requested map not found.
[    0.000000] esrt: ESRT header is not in the memory map.
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: To be filled by O.E.M. To be filled by O.E.M./Cherry Trail CR, BIOS S70C_001 03/17/2016
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x180000 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 000000000 mask FC0000000 write-back
[    0.000000]   1 base 040000000 mask FE0000000 write-back
[    0.000000]   2 base 05F000000 mask FFF000000 uncachable
[    0.000000]   3 base 05E800000 mask FFF800000 uncachable
[    0.000000]   4 base 05E400000 mask FFFC00000 uncachable
[    0.000000]   5 base 100000000 mask F80000000 write-back
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- WT
[    0.000000] e820: update [mem 0x5e400000-0xffffffff] usable ==> reserved
[    0.000000] e820: last_pfn = 0x5c000 max_arch_pfn = 0x400000000
[    0.000000] Scanning 1 areas for low memory corruption
[    0.000000] Base memory trampoline at [ffff880000098000] 98000 size 24576
[    0.000000] BRK [0x0320b000, 0x0320bfff] PGTABLE
[    0.000000] BRK [0x0320c000, 0x0320cfff] PGTABLE
[    0.000000] BRK [0x0320d000, 0x0320dfff] PGTABLE
[    0.000000] BRK [0x0320e000, 0x0320efff] PGTABLE
[    0.000000] BRK [0x0320f000, 0x0320ffff] PGTABLE
[    0.000000] BRK [0x03210000, 0x03210fff] PGTABLE
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x000000005B15B000 000024 (v02 ALASKA)
[    0.000000] ACPI: XSDT 0x000000005B15B0A8 0000D4 (v01 ALASKA A M I    01072009 AMI  00010013)
[    0.000000] ACPI: FACP 0x000000005B179D48 00010C (v05 ALASKA A M I    01072009 AMI  00010013)
[    0.000000] ACPI: DSDT 0x000000005B15B210 01EB34 (v02 ALASKA A M I    01072009 INTL 20120913)
[    0.000000] ACPI: FACS 0x000000005B751F80 000040
[    0.000000] ACPI: APIC 0x000000005B179E58 000084 (v03 ALASKA A M I    01072009 AMI  00010013)
[    0.000000] ACPI: FPDT 0x000000005B179EE0 000044 (v01 ALASKA A M I    01072009 AMI  00010013)
[    0.000000] ACPI: FIDT 0x000000005B179F28 00009C (v01 ALASKA A M I    01072009 AMI  00010013)
[    0.000000] ACPI: MSDM 0x000000005B179FC8 000055 (v03 ALASKA A M I    01072009 AMI  00010013)
[    0.000000] ACPI: MCFG 0x000000005B17A020 00003C (v01 ALASKA A M I    01072009 MSFT 00000097)
[    0.000000] ACPI: SSDT 0x000000005B17A060 004020 (v01 DptfTb DptfTab  00001000 INTL 20120913)
[    0.000000] ACPI: SSDT 0x000000005B17E080 000654 (v01 CpuDpf CpuDptf  00001000 INTL 20120913)
[    0.000000] ACPI: SSDT 0x000000005B17E6D8 000058 (v01 LowPM  LowPwrM  00001000 INTL 20120913)
[    0.000000] ACPI: UEFI 0x000000005B17E730 000042 (v01 ALASKA A M I    00000000      00000000)
[    0.000000] ACPI: HPET 0x000000005B17E778 000038 (v01 ALASKA A M I    01072009 AMI. 00000005)
[    0.000000] ACPI: SSDT 0x000000005B17E7B0 000763 (v01 PmRef  CpuPm    00003000 INTL 20120913)
[    0.000000] ACPI: SSDT 0x000000005B17EF18 000290 (v01 PmRef  Cpu0Tst  00003000 INTL 20120913)
[... 785 lines cut ...]
[   11.172180] intel_sst_acpi 808622A8:00: Got drv data max stream 25
[   11.177875] rt5645 i2c-10EC5645:00: failed to initialize gpiod
[   11.204115] rt5645: probe of i2c-10EC5645:00 failed with error -38
[   11.204432] had: ******** HAD DRIVER loading.. Ver: 0.01.003
[   11.206349] [drm] mid_hdmi_audio_register: Scheduling HDMI audio work queue
[   11.234376] dw_dmac INTL9C60:00: DesignWare DMA Controller, 8 channels
[   11.261049] dw_dmac INTL9C60:01: DesignWare DMA Controller, 8 channels
[... cut ...]
[   11.631388] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[   11.643037] fuse init (API version 7.24)
[   11.658404] r8152 1-4:1.0 eth0: v1.08.3
[   11.659359] cht-bsw-rt5645 cht-bsw-rt5645: ASoC: CODEC DAI rt5645-aif1 not registered
[   11.659371] cht-bsw-rt5645 cht-bsw-rt5645: snd_soc_register_card failed -517
[   11.711958] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   13.289814] had: snd_intelhad_open: HDMI cable plugged-out
[   14.162633] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready

gouchi, 2016-07-21:
There are no fatal errors.

@Kivutar is it possible to create an efi image with the new release image ?

gouchi, 2016-07-31:
@MadMax33309 Can you try to make a test with the [latest nightly build](http://sources.lakka.tv/nightly/Generic.x86_64/Lakka-Generic.x86_64-devel-20160731121452-r21547-g170e484-efi.img.gz).

MadMax33309, 2016-07-31:
Nope is still there and N64 is super unstable now.
Tried to start SSB and Pod Racer and both crash almost instant the system and freeze it.
Sound still work.

I just update it or should I install it from the ground up new?

gouchi, 2016-08-01:
Try from the ground up for now to avoid side effect from update.

Can you provide the output of 
`journalctl | grep drm`

And [log](http://www.lakka.tv/doc/FAQ/#how-do-i-debug-retroarch) from retroarch.

Thank you.

MadMax33309, 2016-08-01:
The situation became even worst.
The screen issue is stronger.
Lakka don't connect with my network anymore(so no log or output).
N64 is still super unstable and sometimes after rebooting I got even a blackscreen

gouchi, 2016-08-01:
You can connect by [direct access](http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface).

MadMax33309, 2016-08-01:
Ok.
Nothing happen with "journalctl | grep drm"
And here the log from retroarch:

> RetroArch [INFO] :: === Build =======================================
> Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 SSE4 SSE4.2 AES 
> Built: Jul 31 2016
> RetroArch [INFO] :: Version: 1.3.6
> RetroArch [INFO] :: Git: 170e484
> RetroArch [INFO] :: =================================================
> RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
> RetroArch [INFO] :: Config: loading config from: /storage/.config/retroarch/retroarch.cfg.
> RetroArch [INFO] :: Resetting undo buffers.
> RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
> RetroArch [INFO] :: Version of libretro API: 1
> RetroArch [INFO] :: Compiled against API: 1
> RetroArch [INFO] :: Set audio input rate to: 29975.00 Hz.
> RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bilinear.glslp"
> RetroArch [INFO] :: Found shader "/usr/share/common-shaders/nearest.glslp"
> RetroArch [INFO] :: Found shader "/usr/share/common-shaders/stock.glsl"
> RetroArch [INFO] :: Video @ 960x720
> RetroArch [INFO] :: Starting threaded video driver ...
> RetroArch [INFO] :: [DRM]: Found 5 connectors.
> RetroArch [INFO] :: [DRM]: Connector 0 connected: no
> RetroArch [INFO] :: [DRM]: Connector 0 has 0 modes.
> RetroArch [INFO] :: [DRM]: Connector 1 connected: no
> RetroArch [INFO] :: [DRM]: Connector 1 has 0 modes.
> RetroArch [INFO] :: [DRM]: Connector 2 connected: yes
> RetroArch [INFO] :: [DRM]: Connector 2 has 23 modes.
> RetroArch [INFO] :: [DRM]: Connector 2 assigned to monitor index: #1.
> RetroArch [INFO] :: [DRM]: Connector 3 connected: no
> RetroArch [INFO] :: [DRM]: Connector 3 has 0 modes.
> RetroArch [INFO] :: [DRM]: Connector 4 connected: yes
> RetroArch [INFO] :: [DRM]: Connector 4 has 1 modes.
> RetroArch [INFO] :: [DRM]: Connector 4 assigned to monitor index: #2.
> RetroArch [INFO] :: [DRM]: Mode 0: (1920x1080) 1920 x 1080, 60 Hz
> RetroArch [INFO] :: [DRM]: Mode 1: (1920x1080) 1920 x 1080, 60 Hz
> RetroArch [INFO] :: [DRM]: Mode 2: (1920x1080i) 1920 x 1080, 60 Hz
> RetroArch [INFO] :: [DRM]: Mode 3: (1920x1080i) 1920 x 1080, 60 Hz
> RetroArch [INFO] :: [DRM]: Mode 4: (1920x1080) 1920 x 1080, 50 Hz
> RetroArch [INFO] :: [DRM]: Mode 5: (1920x1080i) 1920 x 1080, 50 Hz
> RetroArch [INFO] :: [DRM]: Mode 6: (1680x1050) 1680 x 1050, 60 Hz
> RetroArch [INFO] :: [DRM]: Mode 7: (1400x1050) 1400 x 1050, 60 Hz
> RetroArch [INFO] :: [DRM]: Mode 8: (1600x900) 1600 x 900, 60 Hz
> RetroArch [INFO] :: [DRM]: Mode 9: (1280x1024) 1280 x 1024, 60 Hz
> RetroArch [INFO] :: [DRM]: Mode 10: (1440x900) 1440 x 900, 60 Hz
> RetroArch [INFO] :: [DRM]: Mode 11: (1280x800) 1280 x 800, 60 Hz
> RetroArch [INFO] :: [DRM]: Mode 12: (1152x864) 1152 x 864, 60 Hz
> RetroArch [INFO] :: [DRM]: Mode 13: (1280x720) 1280 x 720, 60 Hz
> RetroArch [INFO] :: [DRM]: Mode 14: (1280x720) 1280 x 720, 60 Hz
> RetroArch [INFO] :: [DRM]: Mode 15: (1280x720) 1280 x 720, 50 Hz
> RetroArch [INFO] :: [DRM]: Mode 16: (1024x768) 1024 x 768, 60 Hz
> RetroArch [INFO] :: [DRM]: Mode 17: (800x600) 800 x 600, 60 Hz
> RetroArch [INFO] :: [DRM]: Mode 18: (720x576) 720 x 576, 50 Hz
> RetroArch [INFO] :: [DRM]: Mode 19: (720x480) 720 x 480, 60 Hz
> RetroArch [INFO] :: [DRM]: Mode 20: (720x480) 720 x 480, 60 Hz
> RetroArch [INFO] :: [DRM]: Mode 21: (640x480) 640 x 480, 60 Hz
> RetroArch [INFO] :: [DRM]: Mode 22: (640x480) 640 x 480, 60 Hz
> libGL: Can't open configuration file /storage/.drirc: No such file or directory.
> RetroArch [INFO] :: Found GL context: kms
> RetroArch [INFO] :: Detecting screen resolution 1920x1080.
> RetroArch [INFO] :: [EGL]: EGL version: 1.4
> libGL: Can't open configuration file /storage/.drirc: No such file or directory.
> RetroArch [INFO] :: [EGL]: Current context: 0x7f52ec121600.
> RetroArch [INFO] :: [KMS]: New FB: 1920x1080 (stride: 7680).
> RetroArch [INFO] :: [GL]: Vendor: Intel Open Source Technology Center, Renderer: Mesa DRI Intel(R) HD Graphics (Cherryview) .
> RetroArch [INFO] :: [GL]: Version: 3.0 Mesa 11.2.0-rc1.
> RetroArch [INFO] :: Querying GL extension: ARB_ES2_compatibility => exists
> RetroArch [INFO] :: Querying GL extension: ARB_sync => exists
> RetroArch [INFO] :: Querying GL extension: ARB_texture_float => exists
[... 32 lines cut ...]
> RetroArch [INFO] :: [udev] Adding device /dev/input/event5 as type ID_INPUT_MOUSE.
> RetroArch [INFO] :: [udev] Adding device /dev/input/mouse1 as type ID_INPUT_MOUSE.
> RetroArch [INFO] :: [udev] Adding device /dev/input/event6 as type ID_INPUT_MOUSE.
> RetroArch [INFO] :: [udev] Adding device /dev/input/mouse2 as type ID_INPUT_MOUSE.
> RetroArch [INFO] :: [udev]: Plugged pad: Microsoft X-Box 360 pad (1118:654) on port #0.
> RetroArch [INFO] :: Autodetect: 125 profiles found
> RetroArch [INFO] :: Autodetect: selected configuration: /tmp/joypads/udev/Microsoft_X-Box_360_pad.cfg
> RetroArch [INFO] :: [udev]: Pad #0 (/dev/input/event7) supports force feedback.
> RetroArch [INFO] :: [udev]: Pad #0 (/dev/input/event7) supports 16 force feedback effects.
> RetroArch [INFO] :: Found joypad driver: "udev".
> RetroArch [INFO] :: ALSA: Using signed 16-bit format.
> RetroArch [INFO] :: ALSA: Period size: 384 frames
> RetroArch [INFO] :: ALSA: Buffer size: 1536 frames
> RetroArch [INFO] :: Found menu display driver: "menu_display_gl".
> RetroArch [INFO] :: Querying GL extension: ARB_framebuffer_object => exists
> RetroArch [INFO] :: Using font rendering backend: freetype.
> function is no-op
> RetroArch [INFO] :: Using font rendering backend: freetype.
> function is no-op
> RetroArch [INFO] :: SRAM will not be saved.
> RetroArch [INFO] :: null: [/storage/.config/retroarch/content_history.lpl].
> RetroArch [INFO] :: [KMS]: New FB: 1920x1080 (stride: 7680).
> RetroArch [INFO] :: [KMS]: New FB: 1920x1080 (stride: 7680).
> function is no-op
> RetroArch [INFO] :: [KMS]: Missed 2 VBlank(s) (Frame: 10, DRM frame: 211484).

gouchi, 2016-08-02:
Can you add this boot parameters (press tab to edit) :

drm.debug=0x1e log_buf_len=1M

`journalctl | grep drm` should give you information now ;-)

gouchi, 2016-08-21:
Can you try retry with the [latest nightly](http://mirror.lakka.tv/nightly/Generic.x86_64/Lakka-Generic.x86_64-devel-20160811143244-r21565-gc79651a-efi.img.gz) ?

Thank you.

MadMax33309, 2016-08-22:
@gouchi 
The new image dont work.
After the Lakka Logo I have only a black screen.

gouchi, 2016-08-22:
Can you try to modify [retroarch configuration](http://www.lakka.tv/doc/Configuring-Lakka) ?

Try to change [video_monitor_index](https://github.com/libretro/RetroArch/blob/master/retroarch.cfg#L149) in the configuration with `video_monitor_index = 2`

Nano shortcuts are CTRL+O (Save), CTRL+W (Search), CTRL+X (Quit)

Kivutar, 2017-02-05:
No reply since august, I close the issue

closed 2017-02-05 by Kivutar