libretro/Lakka #398: No sound on the LattePanda

MadMax33309, 2016-06-09:
Hi,

I though it is better to open a new issue for this problem.
On the LattePanda is no sound.
I used this line "cat /proc/asound/cards" and this is the result:
--- no soundcards ---
Win10 have sound so I think a driver is missing?

gouchi, 2016-06-09:
It seems it is not yet supported out of the box but there is a [patch](http://pastebin.com/raw/kZyiZMR5) for kernel 4.5.

[Source](http://www.cnx-software.com/2016/04/19/hdmi-audio-in-linux-is-now-working-on-cherry-trail-mini-pcs-openelec-7-0-image-released/)

MadMax33309, 2016-06-09:
I have sounds but it more like farts sounds ^^
Maybe I did something wrong?

MadMax33309, 2016-06-09:
Ok it work. I changed something in the BIOS and now I have sound

gouchi, 2016-06-09:
@MadMax33309  What did you change to make it work ?

MadMax33309, 2016-06-09:
The LPE Audio Support have to be on PCI mode and it was on ACPI mode.

Kivutar, 2016-06-10:
Ok, if Lakka runs well enough on this LattePanda, we may want to aquire one and support it officially. So if you have some free time, please create a page with all the specific BIOS settings on our wiki.

MadMax33309, 2016-06-10:
@Kivutar 
Lakka runs great on the LattePanda besides the two issues.
N64 runs pretty decent, FBA and SNES runs perfectly.
Tell me if I should try out other things.

**Topic wiki:**
I am not really sure where and my english is not that good I guess, so I will write it down here so you or someone else can put it on the wiki.

**Audio doesn't work properly**
Enter BIOS Setup
Go to Chipset -> South Bridge -> Audio Configuration
Change LPE Audio Support to LPE Audio PCI mode

**Screen issue**
Two options:
1(Safe option): 
Boot BIOS Setup or Win10 and then restart into Lakka.
Note:
From BIOS Setup you can boot Lakka direktly.

2(Not recommended. ON YOUR OWN RISK!): 
Enter BIOS Setup
Go to Chipset -> North Bridge -> Intel IGD Configuration
Change GOP Driver to disabled
Note: 
After this change you are unable to boot Win10 or enter BIOS Setup, because the GOP driver is important for enabling the videocard.
If you have your Lakka on USB drive, remove the USB drive and start the LattePanda.
After a while it **"should"** reset the BIOS setting to default and you **"should"** be able to use your LattePanda normally again. **No guarantee!!!**
Note²:
After shutting down the system, Lakka doesn't start properly, you can only see the boot logo but you can access via SSH(if enabled).

**My opinion on Option 2:**
It is only a short term solution. After shutting down the system and turning it on again, Lakka doesn't boot into the menu anymore. But if you only reboot(instead of turning off) Lakka will boot normally. I don't know why...

Kivutar, 2016-06-14:
So if I understand well, the audio fix is OK and safe, but the video fix doesn't last?

MadMax33309, 2016-06-14:
Audio fix work pretty well with the new kernel.
The video issue is very weird. Sometimes I did nothing and everything run perfectly.
And sometimes even with the boot trick it doesn't worked at all.
Is very inconsistent and seem to be random.

MadMax33309, 2016-06-17:
Any update so far regard for a new image with sound support?

Kivutar, 2016-06-17:
What if you force the resolution in the cmdline ?

MadMax33309, 2016-06-17:
You mean this?
https://github.com/libretro/Lakka/issues/397#issuecomment-224875613
Should I just type this ""video=HDMI-A-2:1920x1080@60" in the cmdline?

MadMax33309, 2016-06-17:
I type it in the cmdline and have still screen issues after a reboot

MadMax33309, 2016-07-04:
I saw a new update for Lakka.
Did you try to fix the sound or screen issue?
If yes, both are still not fixed.

gouchi, 2016-07-08:
@MadMax33309 Can you try this [image](https://www.sendspace.com/file/ssafj8) if the sound works ?

@Kivutar I think we will have to integrate this [patch](https://github.com/LibreELEC/LibreELEC.tv/blob/master/projects/Generic/patches/linux/linux-030-BYT-CHT-SOC-audio-support.patch) with CONFIG_SUPPORT_HDMI=y for kernel config.

MadMax33309, 2016-07-08:
@gouchi 
It the same image like last time or?
Yeah like I said, it works if I change LPE Audio Support to PCI mode in the BIOS.

gouchi, 2016-07-08:
@MadMax33309 Yes same image so the sound is working reliably ?

MadMax33309, 2016-07-08:
@gouchi 
Yeah as long as PCI mode is enable.
But I tried only a few songs.

MadMax33309, 2016-07-18:
@gouchi @Kivutar 
Sound work now plus I dont have to change the LPE Audio Support.
But the sreen issue is still there.
I read something about that intel have new drivers for the GPU of the Atom x5 CPUs.
Not sure if this helpful:
https://01.org/linuxgraphics/downloads

Kivutar, 2016-07-18:
Ah good, can we close this issue then and discuss the screen issue on the dedicated thread?

MadMax33309, 2016-07-18:
Yes Sir!

closed 2016-07-18 by MadMax33309