libretro/Lakka #399: XMB is trademarked

fenwick67, 2016-06-09:
First off - love this project.  It's so slick.

But... the term "XMB" is trademarked by Sony, so it might be a good idea to remove that name from the interface.  I don't think anyone will care if that's what it's called in the code etc. but Sony might get mad if you're calling your interface that.

Kivutar, 2016-06-10:
Yeah, I was thinking about that too... but to be honest, Lakka is so inspired by sony stuff that if they want to get mad at us, they have many reasons. So if we go this way we might just need to remove the whole GUI and create our own.

fenwick67, 2016-06-10:
I disagree, there's a big difference between using a trademarked term and making a similar ui (IANAL).  Imagine if Samsung called touchwiz "iOS".  It's also not really clear to users what xmb means... might be better to just say "Lakka UI" or something.

twinaphex, 2016-06-10:
guess we could just call it 'rmb' if that makes the bloodhounds go away.