libretro/Lakka #400: Retroarch used only one CPU core

MadMax33309, 2016-06-10:
Hey,

I don't know if this really a bug/issue but I noticed that Retroarch on the LattePanda with the EFI image only run on one CPU core.
For SNES and FBA absolute no problem at all but if you play Super Smash Bros on the N64 with four characters you will noticed a FPS drops and it because Lakka only used 1/4 of the CPU power.

If it possibly to enable more cores for Retroarch?

Kivutar, 2016-06-10:
Some cores are not multi threaded. Can you try with PPSSPP (and enable the threaded options in the core options once a game is launched) ?

twinaphex, 2016-06-10:
You cannot simply  scale workload like that for cores that heavily rely on single threading.

Also, not every emulator benefits from multithreading, not every program in general fro that fact.

MadMax33309, 2016-06-10:
@Kivutar 
PSP run but top show that don't go over 23% usage.
Which option do you mean exactly?
Because I am not sure what I supposed to do.
I only found this "Threaded input hack" (after game launched and I pressed home button and go to option). Do you mean this?

Kivutar, 2016-06-10:
Yeah that was it. And audio and video are threaded by default in our configuration. I think that's the best we can do for now.

MadMax33309, 2016-06-10:
Yeah I though that.
Alright, what a shame.

closed 2016-06-10 by MadMax33309