libretro/Lakka #403: Recognizing new DualShock 4 controllers

phuque99, 2016-06-14:
Late 2015 DualShock 4 controllers are detected as a generic "Wireless Controller":

`Lakka:~ # dmesg|tail -3
[    9.570970] sony 0005:054C:05C4.0001: Using modified Dualshock 4 Bluetooth report descriptor
[    9.573653] input: Wireless Controller as /devices/platform/soc/3f201000.uart/tty/ttyAMA0/hci0/hci0:11/0005:054C:05C4.0001/input/input0
[    9.575114] sony 0005:054C:05C4.0001: input,hidraw0: BLUETOOTH HID v1.00 Gamepad [Wireless Controller] on b8:27:eb:ce:ec:4e
`

Copying Sony-PlayStation4-DualShock4-Controller.cfg to "/storage/joypads/udev/Wireless Controller.cfg" and changing the input_device entry to "Wireless Controller" allows this until to be used normally.

Can you add the right detection code for Lakka to recognize these new models as DualShock 4 controllers for auto-configuration?

Kivutar, 2016-06-14:
Maybe with a kernel patch yes.

phuque99, 2016-06-14:
Thanks for the quick response. Let me know if you need more information other than that dmesg output.

Kivutar, 2016-06-17:
Ok, I acquired a recent DS4 controller, and was able to reproduce that bug.

Additional informations:

 * The controller works well in wired mode
 * In bluetooth mode, it does appear as /dev/input/js0
 * Only the trackpad is recognized
 * Named "Wireless Controller", so the vid:pid may be missing in the kernel

I take a look at the kernel now.

Kivutar, 2016-07-09:
With this image for PC http://static.kivutar.me/Lakka-Generic.x86_64-devel-20160709104308-r21505-gb28260e.img.gz

and using the reset button on my controller, I have been able to pair and play with my recent DS4 that was not working before.

closed 2016-07-30 by Kivutar