libretro/Lakka #405: (Pi3) RetroArch crashes when opening content on Windows CIFS

brianblakely, 2016-06-16:
I began to talk about this issue in #293, but this is a different root problem, so here is a new ticket.

#### Problem

RetroArch crashes when attempting to load zipped content hosted on a Windows CIFS. Opening the archive as a folder shows no contents within.  Playlist scanning shows no results.

#### Permissions

Permissions are wide open.  I can create files, execute shell scripts, and even `cat content.zip` to see garbled binary text.

### Cache Directory

It is set to `/tmp`.

#### .7z

7zipping the content still crashes when loading the archive directly from the core, but **does** reveal the content when viewing as a folder.

#### unzip

Running `unzip content.zip` returns the error:

`cannot find or open content.zip, content.zip.zip, or content.zip.ZIP`

#### Non-zipped

Raw content that is left unzipped loads from the Windows CIFS just fine.

#### macOS

Connecting Lakka to a macOS CIFS works flawlessly.

Running RetroArch on macOS and connecting to the Windows CIFS works flawlessly.

brianblakely, 2016-06-25:
I mounted this location (with the same mount settings) on a Raspbian Pi3 and, at the very least, `unzip` is working (I wasn't able to build RetroArch).

I checked the hexdump of the archives and they are verified plain ZIP files.  Also ran `file` via macOS Terminal; output: `Zip archive data, at least v2.0 to extract`.

hgomez, 2016-10-19:
Same problem with NFS