libretro/Lakka #407: Stuck on Flower Logo

davidkc, 2016-06-17:
I am having troubles with the latest image installation, it gets stuck at the logo screen, the HDD led is blinking so it seems to be doing something, I have tried x86 and x64 builds, nightly builds, Live USB and HDD installation, all with the same result, these are my specs:

Dell Inspiron 1545:

GM45 Chipset
Intel Pentium Dual Core T4200
4 GB DDR2 SDRAM
Intel GMA 4500MHD graphics

Please help

Kivutar, 2016-07-11:
Can you try this build please? http://static.kivutar.me/Lakka-Generic.x86_64-devel-20160710153555-r21519-gd0ce46d.img.gz

davidkc, 2016-07-11:
Hi Kivutar, sadly is still not working

gouchi, 2016-07-13:
Is it possible to get some [log](http://www.lakka.tv/doc/FAQ/#how-do-i-debug-retroarch) about retroarch. You need to [access command line interface](http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface/).

Moreover, can you get the output of :
` lspci -nn | grep VGA`

Thank you for your tests.

theDripp, 2016-07-16:
I have the same issue with the added twist that just yesterday the darn thing ran fine. My PC specs are:
* Intel(R) Pentium(R) 4 CPU 3.20GHz
* 3 GB DDR2 SDRAM
* Intel Corporation 82945G/GZ Integrated Graphics
* Intel(R) 82801G (ICH7 Family) chipset
The log says:
```
RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 
Built: Jun  2 2016
RetroArch [INFO] :: Version: 1.3.4
RetroArch [INFO] :: Git: f8b1f05
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Config: loading config from: /storage/.config/retroarch/retroarch.cfg.
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29975.00 Hz.
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bilinear.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/nearest.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/stock.glsl"
RetroArch [INFO] :: Video @ 960x720
RetroArch [INFO] :: Starting threaded video driver ...
RetroArch [INFO] :: [DRM]: Found 2 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
RetroArch [INFO] :: [DRM]: Connector 0 has 12 modes.
RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 1 connected: no
RetroArch [INFO] :: [DRM]: Connector 1 has 0 modes.
RetroArch [INFO] :: [DRM]: Mode 0: (1920x1080) 1920 x 1080, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 1: (1280x800) 1280 x 800, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 2: (1280x720) 1280 x 720, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 3: (1024x768) 1024 x 768, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 4: (1024x768) 1024 x 768, 70 Hz
RetroArch [INFO] :: [DRM]: Mode 5: (1024x768) 1024 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 6: (800x600) 800 x 600, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 7: (800x600) 800 x 600, 72 Hz
RetroArch [INFO] :: [DRM]: Mode 8: (800x600) 800 x 600, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 9: (640x480) 640 x 480, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 10: (640x480) 640 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 11: (720x400) 720 x 400, 70 Hz
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
RetroArch [INFO] :: Found GL context: kms
RetroArch [INFO] :: Detecting screen resolution 1920x1080.
RetroArch [INFO] :: [EGL]: EGL version: 1.4
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
RetroArch [INFO] :: [EGL]: Current context: 0x7fdac400e250.
RetroArch [INFO] :: [KMS]: New FB: 1920x1080 (stride: 8192).
RetroArch [INFO] :: [GL]: Vendor: Intel Open Source Technology Center, Renderer: Mesa DRI Intel(R) 945G .
RetroArch [INFO] :: [GL]: Version: 2.1 Mesa 11.2.0-rc1.
RetroArch [INFO] :: Querying GL extension: ARB_ES2_compatibility => exists
RetroArch [INFO] :: Querying GL extension: ARB_texture_non_power_of_two => exists
RetroArch [INFO] :: Querying GL extension: ARB_fragment_program => exists
RetroArch [INFO] :: Querying GL extension: ARB_sync => exists
RetroArch [INFO] :: Querying GL extension: ARB_texture_float => doesn't exist
RetroArch [INFO] :: Querying GL extension: EXT_texture_sRGB => exists
RetroArch [INFO] :: Querying GL extension: ARB_framebuffer_sRGB => doesn't exist
RetroArch [INFO] :: GL: Using resolution 1920x1080
RetroArch [INFO] :: [GL]: Default shader backend found: glsl.
RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
RetroArch [INFO] :: Checking GLSL shader support ...
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: [GL]: Using 4 textures.
RetroArch [INFO] :: [GL]: Loaded 1 program(s).
RetroArch [INFO] :: [GL]: Using GL_RGB565 for texture uploads.
RetroArch [INFO] :: Using font rendering backend: freetype.
RetroArch [INFO] :: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
RetroArch [ERROR] :: Currently active window is not an X11 window. Cannot use this driver.
RetroArch [ERROR] :: Cannot initialize input driver. Exiting ...
RetroArch [ERROR] :: Fatal error received in: "init_video_input()"
```
The command `lspci -nn | grep VGA` spat out:
```
00:02.0 VGA compatible controller [0300]: Intel Corporation 82945G/GZ Integrated Graphics Controller [8086:2772] (rev 02)
```
I can try the build @Kivutar mentioned, but I'll have to backup my roms first.

gouchi, 2016-07-18:
> RetroArch [ERROR] :: Currently active window is not an X11 window. Cannot use this driver.  ??

Can you check if you reproduce the issue with the [new release](http://mirror.lakka.tv/nightly/Generic.x86_64/Lakka-Generic.x86_64-devel-20160717204331-r21527-g4656d16.img.gz) ? 

Thank you.

theDripp, 2016-07-18:
@gouchi Sure thing, I'll report back later today.

theDripp, 2016-07-19:
Looks like it's working again. I tried a normal snes (Super Mario World) rom, a FX chip snes game (Starfox) rom and a psx game (Castlevania: Symphony of the Night). This is the log in case it's needed:
```
RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 
Built: Jul 17 2016
RetroArch [INFO] :: Version: 1.3.4
RetroArch [INFO] :: Git: 4656d16
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Config: loading config from: /storage/.config/retroarch/retroarch.cfg.
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29975.00 Hz.
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bilinear.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/nearest.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/stock.glsl"
RetroArch [INFO] :: Video @ 960x720
RetroArch [INFO] :: Starting threaded video driver ...
RetroArch [INFO] :: [DRM]: Found 2 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
RetroArch [INFO] :: [DRM]: Connector 0 has 12 modes.
RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 1 connected: no
RetroArch [INFO] :: [DRM]: Connector 1 has 0 modes.
RetroArch [INFO] :: [DRM]: Mode 0: (1920x1080) 1920 x 1080, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 1: (1280x800) 1280 x 800, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 2: (1280x720) 1280 x 720, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 3: (1024x768) 1024 x 768, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 4: (1024x768) 1024 x 768, 70 Hz
RetroArch [INFO] :: [DRM]: Mode 5: (1024x768) 1024 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 6: (800x600) 800 x 600, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 7: (800x600) 800 x 600, 72 Hz
RetroArch [INFO] :: [DRM]: Mode 8: (800x600) 800 x 600, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 9: (640x480) 640 x 480, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 10: (640x480) 640 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 11: (720x400) 720 x 400, 70 Hz
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
RetroArch [INFO] :: Found GL context: kms
RetroArch [INFO] :: Detecting screen resolution 1920x1080.
RetroArch [INFO] :: [EGL]: EGL version: 1.4
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
RetroArch [INFO] :: [EGL]: Current context: 0x7f8b8000dbd0.
RetroArch [INFO] :: [KMS]: New FB: 1920x1080 (stride: 8192).
RetroArch [INFO] :: [GL]: Vendor: Intel Open Source Technology Center, Renderer: Mesa DRI Intel(R) 945G .
RetroArch [INFO] :: [GL]: Version: 2.1 Mesa 11.2.0-rc1.
RetroArch [INFO] :: Querying GL extension: ARB_ES2_compatibility => exists
RetroArch [INFO] :: Querying GL extension: ARB_texture_non_power_of_two => exists
RetroArch [INFO] :: Querying GL extension: ARB_fragment_program => exists
RetroArch [INFO] :: Querying GL extension: ARB_sync => exists
RetroArch [INFO] :: Querying GL extension: ARB_texture_float => doesn't exist
RetroArch [INFO] :: Querying GL extension: EXT_texture_sRGB => exists
RetroArch [INFO] :: Querying GL extension: ARB_framebuffer_sRGB => doesn't exist
RetroArch [INFO] :: GL: Using resolution 1920x1080
RetroArch [INFO] :: [GL]: Default shader backend found: glsl.
RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
RetroArch [INFO] :: Checking GLSL shader support ...
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: [GL]: Using 4 textures.
RetroArch [INFO] :: [GL]: Loaded 1 program(s).
RetroArch [INFO] :: [GL]: Using GL_RGB565 for texture uploads.
RetroArch [INFO] :: Using font rendering backend: freetype.
RetroArch [INFO] :: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
RetroArch [INFO] :: [udev] Adding device /dev/input/event2 as type ID_INPUT_KEYBOARD.
RetroArch [INFO] :: Found joypad driver: "udev".
RetroArch [INFO] :: ALSA: Using signed 16-bit format.
RetroArch [INFO] :: ALSA: Period size: 1024 frames
RetroArch [INFO] :: ALSA: Buffer size: 2048 frames
RetroArch [INFO] :: Found menu display driver: "menu_display_gl".
RetroArch [INFO] :: Querying GL extension: ARB_framebuffer_object => exists
RetroArch [INFO] :: Using font rendering backend: freetype.
function is no-op
RetroArch [INFO] :: Using font rendering backend: freetype.
RetroArch [INFO] :: SRAM will not be saved.
RetroArch [INFO] :: null: [/storage/.config/retroarch/content_history.lpl].
RetroArch [INFO] :: [KMS]: New FB: 1920x1080 (stride: 8192).
RetroArch [INFO] :: [KMS]: New FB: 1920x1080 (stride: 8192).
RetroArch [INFO] :: [KMS]: Missed 1 VBlank(s) (Frame: 2, DRM frame: 256174).
function is no-op
RetroArch [INFO] :: [KMS]: Missed 2 VBlank(s) (Frame: 8, DRM frame: 256180).
```

gouchi, 2016-07-21:
Closing the issue as it is working with the new release.

closed 2016-07-21 by gouchi