libretro/Lakka #408: No way to set/disable/adjust overscan/underscan without X11, xrandr or binary NVIDIA driver

kevinfishburne, 2016-06-20:
At this point the only thing stopping me from using Lakka instead of RetroArch on a normal distro is this problem, as I'm using a CRT television in an arcade cabinet I built. May be a limitation of DRM/KMS/nouveau...I have no idea.

Kivutar, 2016-06-20:
OK. I found this on the web, maybe it can help us understand the problem better https://bugs.freedesktop.org/show_bug.cgi?id=47846

A few workarounds: 

 * Setting an overscan directly in RetroArch using the custom viewport options in the video settings.
 * Compiling your own version of Lakka with x11. Requires 

There is also something there if you search underscan https://01.org/linuxgraphics/gfx-docs/drm/drm-kms-properties.html

hizzlekizzle, 2016-06-20:
Can you post a photo of the issue? Is an excessive amount getting cut off or is it showing things it shouldn't?

gouchi, 2016-06-20:
To modify [kms properties](https://01.org/linuxgraphics/gfx-docs/drm/drm-kms-properties.html) we need to include [modetest](https://wiki.linaro.org/xinliang/libdrm/modetest).

Kivutar, 2016-06-20:
Commenting this line should do it https://github.com/libretro/Lakka/blob/lakka/packages/graphics/libdrm/package.mk#L59

kevinfishburne, 2016-06-21:
@hizzlekizzle Nouveau is enabling "overscan compensation" (underscanning) by default. This would be useful if using a CRT television for desktop computing where it's important that the entire desktop be displayed. Old console games were designed for CRT televisions which naturally overscanned past their bezel, so nouveau's behavior (at least for CRT televisions) is not useful for emulation. Nouveau's outputting to "TV-1" over S-Video. Considering no one uses CRT televisions for desktop computing anymore a more sensible default would be full overscan or to make it adjustable via KMS.

In the screenshot (the lighting is poor) you can see black around RetroArch's menu.
[image]

kevinfishburne, 2016-06-21:
@Kivutar I'm going to try my best to compile Lakka with X11 and test xrandr. I've seen the drm-kms-properties page but don't know the syntax for experimenting with the underscan options. Right now my KMS line looks like this:

APPEND boot=LABEL=System disk=LABEL=Storage  ssh quiet vt.global_cursor_default=0 loglevel=2 nouveau.tv_norm=hd480i video=TV-1:720x480@60

Using the custom viewport options in RetroArch would be a good workaround for recuding overscan, but nouveau is by default scaling down/shrinking the display output so it wouldn't work to increase overscan.

@gouchi So commenting out line 59 will expose the KMS properties listed on the drm-kms-properties page? On that page under "DRM" there is a "TV" group with an "overscan" property accepting values between 0 and 100. nvidia-settings allows controlling overscan via the .nvidia-settings-rc file and accepts values between 0 and 25 (which I've successfully tested), so this might be the one.

Kivutar, 2016-08-12:
Have you tried setting a custom viewport?